absalom-muc/MHI-AC-Ctrl #1: Detailed Component List

lifeandfate1, 2020-01-12:
Hi, I'm currently attempting to construct 3 of the MHI-AC adapter's for my AC units.

I have been attempting to find all of the components that you used constructing these boards but I'm struggling to fully read the information from the schematic files you have included and I don't want to make any mistakes.

Is there any chance that you could add a section with a detailed list of all the little individual components that where used to make these boards including capacitors, connectors, etc.

Thank you so much for this project, it was exactly what I was looking for!

absalom-muc, 2020-01-12:
Please find a list of the components exported by Eagle:
Part     Value               Package           Library      Position (mil)        Orientation
C1       22µ/50V             E5-10,5           rcl          (225 850)             R90
C2       100n                C025-024X044      rcl          (1900 925)            R180
C3       22p                 C025-024X044      rcl          (575 1025)            R0
JP1                          JP1               jumper       (825 1025)            R270
LS1      LEVEL-SHIFTER-4CH   LEVEL-SHIFTER-4CH diy-modules  (1500 575)            R90
U$1      WEMOS-D1-MINI       WEMOS-D1-MINI     diy-modules  (1625 575)            R90
U$4      JST-PH_5-PIN_HEADER JST_B5B-PH_HEADER srg          (150 350)             R90
U$5      JST-PH_5-PIN_HEADER JST_B5B-PH_HEADER srg          (775 350)             R90
U1       TSR_1-2450          TSR-1             _traco_tsr-1 (675 700)             R0

absalom-muc, 2020-01-18:
Also added to [Hardware.md](/absalom-muc/MHI-AC-Ctrl/blob/master/Hardware.md)

closed 2020-01-18 by absalom-muc

hvorragend, 2020-04-29:
Sorry, I have some questions:

(1)
Are C2 and C3 multilayer Ceramic Capacitors? With 25 Volt?
I can't see them here: https://raw.githubusercontent.com/absalom-muc/MHI-AC-Ctrl/master/images/Assembled-PCB.jpg

(2)
I think that this level shifter should fit:
https://de.aliexpress.com/item/32715353192.html

(3) 
Do you know a "china solution" for the TSR-1 2450?

Thank you.

absalom-muc, 2020-04-29:
(1) Choose the cheapest capacitors you find for C2 and C3, voltage >=15V is fine
(2) looks good
(3) no unfortunately I don't know a "china solution". If you are looking for the datasheet, it is located [here](https://www.reichelt.de/index.html?ACTION=7&LA=3&OPEN=0&INDEX=0&FILENAME=D400%2FTSR1_EN.pdf)