absalom-muc/MHI-AC-Ctrl #2: No issue, just a question

coyote1, 2020-01-19:
Hey,
it's nice to see that someone has taken on the project again.
At the moment I am still using rjdekker's version, which works so far, but I would also like to test your solution with the Wemos Mini.
Where do you order your board? In Germany?
I also found something in China, but I would need a Gerber file instead of the Eagle file. Can you maybe also provide a Gerber file?

absalom-muc, 2020-01-19:
I ordered the PCB in Germany. It was simple for me because they accept an Eagle file and they are fast (but of course more expensive). I created now the [Gerber files ](https://github.com/absalom-muc/MHI-AC-Ctrl/files/4082769/CAMOutputs.zip), but since I've no experience with this, I strongly recommend that you have a close look on the files before you issue an order based on these files.
Good luck!

coyote1, 2020-01-19:
Thank you.
Do you have a link, where you ordered your circuit boards

absalom-muc, 2020-01-19:
[https://www.eurocircuits.de/](https://www.eurocircuits.de/)

absalom-muc, 2020-01-20:
Please consider the updated [Hardware.md](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/Hardware.md) description! I added some useful information related to the connection of the AC and the remote control PCB.

closed 2020-01-25 by absalom-muc