absalom-muc/MHI-AC-Ctrl #3: MQTT Broker Authentication Issues

lifeandfate1, 2020-01-22:
Hi

I'm trying to connect the board to my Mosquito MQTT broker in Home Assistant but struggling due to the broker requiring a username and password.

I have tried looking through the pubsubclient MQTT Authentication example to find a means to add a username and password field in the v1.2 MHI-AC-Ctrl.ino but am having no luck in adding any options for this.

Hopefully I'm not missing anything simple that you have already included.

If not is there any chance to update the ino file to version 1.3 that include an optional field that can be filled in if there is a username and password requirement for the MQTT broker?

Any help here would be amazing!

absalom-muc, 2020-01-22:
Currently I don't use MQTT authentication, so I can't test it. But according my understanding of the [MQTT API description](https://pubsubclient.knolleary.net/api.html) you have only to replace the connect command in line 83
`if (MQTTclient.connect (hostname, MQTT_PREFIX "connected", 0, true, "false")) {`
by
`if (MQTTclient.connect (hostname, "Broker-Username", "Broker-Password", MQTT_PREFIX "connected", 0, true, "false")) {`
Please try it and give us feedback if it works.

lifeandfate1, 2020-01-22:
Last night I think I stumbled across the solution that you mentioned as well by adding "Broker-Username" and "Broker-Password" in that section.

I also added definitions to them at the start among the other MQTT definitions as below:
```
#define MQTT_USER "***********"
#define MQTT_PASSWORD "***********"
```

Then I changed your section of code to read:

`if (MQTTclient.connect (hostname, MQTT_USER, MQTT_PASSWORD, MQTT_PREFIX "connected", 0, true, "false")) {`

That seemed to do the trick and it can authenticate now.

Thank you so much for suggestion and hopefully this helps someone if they also use MQTT authentication with this project!

closed 2020-01-22 by lifeandfate1

absalom-muc, 2020-01-24:
Glad to hear that! Thanks for your feedback.