absalom-muc/MHI-AC-Ctrl #5: Soft WDT Reset of Lonlin WEMOS D1 Mini with MHI-AC-Ctrl Sketch

lifeandfate1, 2020-01-31:
Hi again,

I have been slowly loosing my mind trying to fix a looping soft reset of the Watch Dog Timer on the D1 Mini.

I have tested the device with other sketches including MHI-AC-SPI and found that there is no physical issue with the devices itself as these sketches work without issue.

I'm just wondering if anyone else has had a similar issue or if there is a simple solution I haven't tried.

Currently this is the output of the ESP exception decoder that I'm using to debug the sketch.

`60:MHI-AC-Ctrl starting`
`...........OTA Ready`
`IP address: 192.168.2.55`
 `connected to IoT, IP address: 192.168.2.55`
`MQTTreconnect`
`Attempting MQTT connection...connected`
`Soft WDT reset`

`>>>stack>>>`

`ctx: cont
sp: 3ffffd90 end: 3fffffc0 offset: 01b0
3fffff40:  00000001 3ffe875c 00000000 00000000  
3fffff50:  00000000 00000001 000000ff 000000ff  
3fffff60:  00000000 00000000 00000001 00000008  
3fffff70:  000000ff 000000ff 000000ff 000000ff  
3fffff80:  000000ff 00000000 3ffe84dc 4020d710  
3fffff90:  4020e0d8 3702a8c0 feefeffe 3ffeeb14  
3fffffa0:  3fffdad0 00000000 3ffeead4 4020bf64  
3fffffb0:  feefeffe feefeffe 3ffe851c 40100f29`
`<<<stack<<<`
 
 `ets Jan  8 2013,rst cause:2, boot mode:(3,6)`
`load 0x4010f000, len 1392, room 16 
tail 0
chksum 0xd0
csum 0xd0
v3d128e5c
~ld`

Decoding of the stack results shows the following errors

`0x4020d710:`
`std::_Function_handler ::_M_invoke(std::_Any_data const&, char*, unsigned char*, unsigned int)` 
`at c:\users\lifef\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2/functional line 2069`

`0x4020bf64:` 
`loop_wrapper() at` `C:\Users\lifef\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266\core_esp8266_main.cpp line 180`

If any more information may be useful in figuring out what is going wrong here is required I can happy provide.

Any help here to diagnose what is going wrong would be really helpful.

absalom-muc, 2020-01-31:
Hi,
Please check that SCK is connected. There are some program lines waiting for a SCK toggle, e.g. 
`while (digitalRead(SCK)) {} // wait for falling edge`
If there is no SCK toggle then the WDT is triggered.
Good luck!
absalom

lifeandfate1, 2020-01-31:
Right I think I get it.

So you mean that I need to have the device (D1 Mini, and other parts) connected to my AC unit as there are some lines in the sketch that will cause a WDT as it is expecting a reading of some form from the head unit?

If that makes sense :)

absalom-muc, 2020-02-04:
Yes, SCK has to toogle, else the program is caught in an endless loop and can't call 
`delay(0);`
to serve the watchdog.

absalom-muc, 2020-03-01:
I assume after connecting SCK it worked, so closed. In the latest version v1.3 a frequency measurement is implemented to check thecorrect connection of the board to the air conditioner.

closed 2020-03-01 by absalom-muc