absalom-muc/MHI-AC-Ctrl #6: ZSX signature ?

metarutaiga, 2020-02-01:
My ZSX signature first byte is 0x6D, not 0x6C.

so I suggest the code
if ((rx_SPIframe[SB0] == 0x6c)
could be
if (((rx_SPIframe[SB0] & 0xfe) == 0x6c)

Thanks.

absalom-muc, 2020-02-04:
Have you already tested that all commands are working in the same way as for the ZS series?
What is the exact name of the AC? Is it SRK xx ZSX-S for the indoor unit and SRC xx ZSX-S for the outdoor unit?

metarutaiga, 2020-02-05:
Yes, it's ZSX series. The all command are almost working.
But some function like auto-fan and 6 level vanes and wide-vanes need to check who to set.

absalom-muc, 2020-03-01:
Implemented in v1.3

closed 2020-03-01 by absalom-muc

oliv3r, 2025-01-17:
@absalom-muc I wonder if we have any sense if this difference in siganture comes from the fact whether it uses 22 or 33 byte frames? I don't think we've seen a lot of signature bytes from users to make a sensible statement here?

absalom-muc, 2025-01-25:
yes, this is correct.