absalom-muc/MHI-AC-Ctrl #7: Cannot get it working

gamblor999, 2020-02-10:
I have been trying to get this working with both rjdekkars library and now your library with no success. Below is a picture of my connections which looks the same as your picture (excuse the blurryness) as aircon is right next to wall so was difficult to hook up)

I’ve tried swapping the scl and MOSI but no success I can see mqtt is connecting again and again to my broker which suggests it’s getting caught in the while loop and resetting. This would suggest nothing coming through on the scl but I don’t get it my picture looks the same as yours. I’m using level shifter from spark fun and polu voltage regulator as per rjdekkar with a nodemcu over the Wemos. I’ve tried diffrent cpu frequencies. Any help would be greatly appreciated. My unit is an srk

[image]

gamblor999, 2020-02-10:
[image]
[image]

[image]

gamblor999, 2020-02-10:
Also
When I check the voltages on the aircon port I get the following

Pin1 = 12v
Pin2= 4.7v
Pin3=4.7v
Pin4=2.78v
Pin5 =ground

absalom-muc, 2020-02-10:
Spontaneously I see no mistake. Only in the pictures you write "D7 & D5 (SCL and MISO) swapped", I assume that there is only a typo because it should be SCL and MOSI (instead of MISO), but D7 & D5 is correct.
btw: For your testing I recommend that you connect power supply and SCL only, to prevent unintended driving of an aircon output.

absalom-muc, 2020-02-10:
Can you see that the mqtt is connecting again and again to your broker also when your ESP board is not connected via USB with your computer? With this question I want to ensure that the power supply with your DCDC is working and that the ESP board is not supplied via USB.

gamblor999, 2020-02-10:
Yes when it’s connected to USB on pc it resets due to the while loop digital read as expected and in the serial I just see it connecting to mqtt and then resetting, when connected to the 12v through pololu regulator to 5v I see the same thing on the broker it connects to mqtt and then resets. On your last message I disconnected everything Except scl and power and tried each of the possible 3 pins as scl between ground and 12v and the same thing happened for all of them. Is it possible that maybe there is an srk model that does not work...  I even bought all the parts to rjdekker last version (arduino pro mini, esp and adapter ) followed the code and still no success with that either. No one replies to comments on that one as he no longer maintains it so really appreciate your reply

gamblor999, 2020-02-10:
Out of curiosity do you get the same voltages on each of the 5 pins from the aircon as per my previous message, the one that is 2.78v seems odd was wondering if you have one that voltage also

absalom-muc, 2020-02-10:
Imho a CNS connector is a clear indication for a SPI, but of course the protocol could be different. Nevertheless the SPI CLK should toggle.
Currently I can't check the voltage levels, but your list
Pin1 = 12v
Pin2= 4.7v (SCL)
Pin3=4.7v (MOSI)
Pin4=2.78v (MISO)
Pin5 =ground
seems to be reasonable.
- SCL and MOSI are AC outputs and have mostly a value of 5V, so a value just below 5V seems to be fine.
- MISO is an AC input, every level between 0 - 5V is o.k. (without the knowledge if there are AC internal pull up/down resistors)

Can you measure the SCL and MOSI voltage behind the DCDC-converter? Should be about 3V independent if SCL/MOSI are connected to the ESP board

gamblor999, 2020-02-10:
the voltages behind the level converter that go to the nodemcu were:
Scl= 3.03
MOSI = 3
Miso = 3.22

absalom-muc, 2020-02-10:
The voltage levels look fine. Now we should measure the SCK frequency, expected is approx. 32kHz. but we have to consider that SCK is only sometimes toggling as described in the timing clause of the SPI docu. Can you measure the frequency? With a standard frequency meter I would expect a value of 20*20*8Hz=3.2kHz 
If you have no frequency meter we can use most probably the ESP for the measurement.

gamblor999, 2020-02-11:
i have never measured frequency but found a method using the ardunio uno which i tried (readings vary a bit so not sure)
https://github.com/PaulStoffregen/FreqCount
https://www.pjrc.com/teensy/td_libs_FreqCount.html

[image]
[image]

gamblor999, 2020-02-11:
for the above measurements below are the averages for each pin

PIN 2 (SCL) =   5717.17

PIN 3 (MOSI) = 2927.41

PIN 4 (MISI) = 944.79

absalom-muc, 2020-02-11:
To be honest, none of these values make sense for me. Unfortunatelly I can't use the lib you used with my WEMOS board. Therefore I wrote my own very simple frequeny measurement program:

```
// MHI-AC-frequency-simpel_v1.0
// simple frequency measurement for SCK input

#define SCK 14
void setup() {
  Serial.begin(115200);
  pinMode(SCK, INPUT);
}

void loop() {
  uint cnt=0;
  unsigned long starttimeMillis = millis();
  uint SCK_state = digitalRead(SCK);
  while(millis() - starttimeMillis < 1000){
    if(SCK_state != digitalRead(SCK)){
      cnt++;
      SCK_state = !SCK_state;
    }
  }
  Serial.printf("frequency=%i\n", cnt/2);
}
```

Results are between 3040 - 3200. That is the range I expected. Can you try this program?

gamblor999, 2020-02-12:
your code worked really well!   below are the results for pin 2 SCL

[image]

absalom-muc, 2020-02-12:
Very good. Then let us check if you have the same problem as metarutaiga described [here](https://github.com/absalom-muc/MHI-AC-Ctrl/issues/6) that the signature of your AC is different. Please use the original MHI-AC-Ctrl sketch and replace in row 370
`if ((rx_SPIframe[SB0] == 0x6c) & (rx_SPIframe[SB1] == 0x80) & (rx_SPIframe[SB2] == 0x04) & (rx_SPIframe[CBH] == highByte(rx_checksum)) & (rx_SPIframe[CBL] == lowByte(rx_checksum)))`
by
`if ((rx_SPIframe[SB1] == 0x80) & (rx_SPIframe[SB2] == 0x04) & (rx_SPIframe[CBH] == highByte(rx_checksum)) & (rx_SPIframe[CBL] == lowByte(rx_checksum)))`

So we don't longer check SB0 for detecting of a valid frame.

gamblor999, 2020-02-12:
in below serial it connects ok,  but from the broker  (windows) it never is able to subscribe to a topic  ( am i doing something very wrong here).  

[image]

absalom-muc, 2020-02-12:
What I can see from the screenshot MHI-AC-Ctrl is connected to the broker, so on this side all is working. But I can't tell you why mosquitto_sub is not connecting to the broker (could be a wrong broker configuration, a firewall topic, a virus scanner etc.). The refused connection has nothing to do with MHI-AC-Ctrl, therefore I can't help you with this problem.

gamblor999, 2020-02-12:
Really appreciate all the help you have given to get me to this point.   I have added inbound rules to firewall, temporarily disabled bitdefender and forwarded ports on the router/modem,   I will install mosquitto on my rpi and give it a shot from there perhaps something else in windows i have overlooked.  I will let you know if i do get this going.

one last question:  as a last resort do you know of any online/cloud mqtt brokers i can use that will eliminate any problems in my personal setup/hardware/configuration ?  i have read one of the other closed issues on this thread to add a username and password which seems easy enough.

coyote1, 2020-02-13:
I don't want to open a new issue. Yesterday, i soldered all the components together and did a first test. Looks good so far, all values are displayed in "ioBroker" MQTT Server.
Now my question, no "set /" topics are created, I then created them by myself and then the control of the device also works.

gamblor999, 2020-02-13:
Good sir absalom-muc,  i tip my hat to you!!!   All is working!    I did try changing row 370 back to the original code (including SB0) to see if it would still work but nothing shows up in mqtt execpt connected topic.  

For anyone else having issues my setup is as follows:

Model = SRK20ZSA-W
1. Modified version of absalom-muc' code where line 370 has been changed as per the above.
2. Flashed to NodeMCU (not sure if it matters but CPU frequency was 80Mhz
3. AIRCON PIN SETUP
Pin1 = 12V
Pin2 = SCL     -> level converter  -> D5
Pin3 = MOSI  -> level converter -> D7
Pin4 = MISO - > level converter .> D6
Pin5 = GROUND
4. Mqtt broker was setup with openhab on the raspberry pi (anyone with similar setup first 8minutes of this video helped - https://www.youtube.com/watch?v=R-SrZvKHXdA)
5. Tested the mqtt setup using MQtt.fx (as per video).

Thanks for taking the time and for all your help absalom-muc,  I live in Australia and its summer so being able to automate the cooling of the house will make such a difference.

[image]

absalom-muc, 2020-02-13:
@gamblor999: Great to hear and thank you for the feedback! I have one request: I'm interested to know the signature of your AC. Could you add to row 369
`Serial.println(rx_SPIframe[SB0]);`
and check the results? Thank you!

absalom-muc, 2020-02-13:
@coyote1: The "set/" topics (commands to the AC) have to be published by your client which should control your AC. Publishing e.g. the topic MHI-AC-Ctrl/set/Power with value "On" is the right way. If you have still problems with it please create a new Issue thread

coyote1, 2020-02-13:
OK, thanks for your answer. Works fine so far, can control 3 Aircons now. Good Work 👍

absalom-muc, 2020-02-13:
Congratulation coyote1 - good to hear!

gamblor999, 2020-02-14:
> @gamblor999: Great to hear and thank you for the feedback! I have one request: I'm interested to know the signature of your AC. Could you add to row 369
> `Serial.println(rx_SPIframe[SB0]);`
> and check the results? Thank you!

will get this done tomorrow night and report back! thanks again absalom-muc

gamblor999, 2020-02-18:
Apologies for delay, i had it all soldered and taped up.   new parts came in the mail that i could prototype again to serial debug.
I tried your original code again and it worked with no modification ( originally it appeared i was only getting the subscription for connected,  but this was not the case this time round as all topics are working so have left the code as per your original). 

I did run the following as requested:
Serial.println(rx_SPIframe[SB0]);    

The serial output displayed   109

I have updated/corrected my comments further up for future readers of this issue,  i will now close this off.

closed 2020-02-18 by gamblor999

absalom-muc, 2020-02-18:
Thank you gamblor999, enjoy the summer!

aaronactive, 2024-04-16:
Sorry to bring this up again, but is there any help here to fix this on the original [rjdekker](https://github.com/rjdekker)
/
[MHI2MQTT](https://github.com/rjdekker/MHI2MQTT) code? I have the same model SRK20ZSA-W and presumably this is the same issue?