absalom-muc/MHI-AC-Ctrl #8: PCB or assembled PCB

hvorragend, 2020-03-01:
Hi absalom-muc,
ich weiß leider nicht, wie ich dich anders erreichen kann. Bist du in irgendwelchen Boards aktiv, worüber ich dich anschreiben kann?

Ich wollte dich fragen, ob du noch eine Platine oder vielleicht sogar eine (evtl. zwei) zusammengebaute Platinen zur Verfügung hast? Notfalls löte ich mir diese auch selbst zurecht. 

Kannst du mir helfen? Ich will nicht zwei Platinen in China bestellen. 

Danke dir. Und sorry für diesen unüblichen Weg. 

Viele Grüße

absalom-muc, 2020-03-01:
Hallo hvorragend,
das ist schon der richtige Weg. Leider habe ich keine (zusammengebauten) Platinen übrig. 
Viele Grüße
absalom

hvorragend, 2020-03-01:
Danke für die Rückmeldung. Also auch keine reinen unbestückten PCBs?

absalom-muc, 2020-03-01:
Nein, weder bestückt, noch unbestückt.

hvorragend, 2020-03-01:
Danke dennoch.

hvorragend, 2020-04-26:
Nur zur Info:
Ich habe 10 Boards bestellt. Vielen Dank für deine Dateien. 

https://forum.fhem.de/index.php/topic,110493.msg1046911.html#msg1046911

>

absalom-muc, 2020-04-26:
Super Aktion - Danke für die Info!
Und wenn du Unterstützung brauchst, du weißt ja wo du mich findest. Passt bitte nur bei der Verbindung Klimaanlage zum Board auf. Auf den Dreher im Kabel ist schon der ein oder andere _hereingefallen_. Ist [hier](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/Hardware.md#connector) beschrieben.

hvorragend, 2020-04-26:
Danke. Ich verfolge das Projekt hier und habe die Hinweise bereits gelesen. 

Es ist dein Projekt und deine Arbeit, von der wir hier profitieren. Wenn du magst, dann sende ich dir eine Platine zu. Natürlich kostenlos.

absalom-muc, 2020-04-28:
Das ist sehr nett von dir! Vielen Dank, aber solange mir nichts abbrennt, bin ich ausreichend versorgt.

ligangyi, 2020-05-06:
Just wanted to let you know, I bought my PCBs from JLCPCB by submitting your set of exported gerbers. They came out fine and are now working in my ACs. 

I paid USD$9.94 and was shipped to me within 2 weeks.

hvorragend, 2020-10-04:
I just want to mention that I have two PCBs to sell. 
Please see the German FHEM forum: https://forum.fhem.de/index.php/topic,110493.0.html
Sorry to abuse this issue here.

damienmorrissey, 2020-10-05:
Would be happy to take any boards you have available. Let me know how much and how to pay for them shipped to Australia. I'll PM the address.

closed 2020-10-13 by hvorragend