absalom-muc/MHI-AC-Ctrl #9: Version 1.3 Frequenz out of Range

coyote1, 2020-03-04:
Hey, habe heute deine Version 1.3 ausprobiert, leider startet der Wemos immer neu und der Monitor zeigt "Frequenzy out of range"
Hier mal ein Teil des seriellen Monitors:

`MHI-AC-Ctrl starting
Measure frequency for SCK, MOSI and MISO pin
SCK frequency=4160Hz (expected: 3000 .. 3200Hz) out of range!

Soft WDT reset

ctx: cont 
sp: 3ffffd70 end: 3fffffd0 offset: 01b0

>>>stack>>>
3fffff20:  0000003f 0000000d 3ffef58c 40208188  
3fffff30:  0000003f 3ffef340 3ffef58c 40208469  
3fffff40:  3ffe93ac 00000020 3ffe8fc4 40208469  
3fffff50:  3ffe8ad0 feefeffe 3ffef58c 4020861c  
3fffff60:  40106808 4010642c 3ffef58c 40208690  
3fffff70:  0000003f 3ffef340 3ffef58c 40202e7d  
3fffff80:  3ffe8500 3ffef344 3ffef58c 40203346  
3fffff90:  feefeffe feefeffe feefeffe feefeffe  
3fffffa0:  feefeffe feefeffe feefeffe feefeffe  
3fffffb0:  3fffdad0 00000000 3ffef65c 40209600  
3fffffc0:  feefeffe feefeffe 3ffe8608 40100739  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld

MHI-AC-Ctrl starting
Measure frequency for SCK, MOSI and MISO pin
SCK frequency=4168Hz (expected: 3000 .. 3200Hz) out of range!

Soft WDT reset

ctx: cont 
sp: 3ffffd70 end: 3fffffd0 offset: 01b0

>>>stack>>>
3fffff20:  0000003e 0000000d 3ffef58c 40208188  
3fffff30:  0000003e 3ffef340 3ffef58c 40208469  
3fffff40:  3ffe93ac 00000020 3ffe8fc4 40208469  
3fffff50:  3ffe8ad0 feefeffe 3ffef58c 4020861c  
3fffff60:  40106808 4010642c 3ffef58c 40208690  
3fffff70:  0000003e 3ffef340 3ffef58c 40202e7d  
3fffff80:  3ffe8500 3ffef344 3ffef58c 40203346  
3fffff90:  feefeffe feefeffe feefeffe feefeffe  
3fffffa0:  feefeffe feefeffe feefeffe feefeffe  
3fffffb0:  3fffdad0 00000000 3ffef65c 40209600  
3fffffc0:  feefeffe feefeffe 3ffe8608 40100739  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld

MHI-AC-Ctrl starting
Measure frequency for SCK, MOSI and MISO pin
SCK frequency=4304Hz (expected: 3000 .. 3200Hz) out of range!

Soft WDT reset

ctx: cont 
sp: 3ffffd70 end: 3fffffd0 offset: 01b0

>>>stack>>>
3fffff20:  0000003f 0000000d 3ffef58c 40208188  
3fffff30:  0000003f 3ffef340 3ffef58c 40208469  
3fffff40:  3ffe93ac 00000020 3ffe8fc4 40208469  
3fffff50:  3ffe8ad0 feefeffe 3ffef58c 4020861c  
3fffff60:  40106808 4010642c 3ffef58c 40208690  
3fffff70:  0000003f 3ffef340 3ffef58c 40202e7d  
3fffff80:  3ffe8500 3ffef344 3ffef58c 40203346  
3fffff90:  feefeffe feefeffe feefeffe feefeffe  
3fffffa0:  feefeffe feefeffe feefeffe feefeffe  
3fffffb0:  3fffdad0 00000000 3ffef65c 40209600  
3fffffc0:  feefeffe feefeffe 3ffe8608 40100739  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld

MHI-AC-Ctrl starting
Measure frequency for SCK, MOSI and MISO pin
SCK frequency=4184Hz (expected: 3000 .. 3200Hz) out of range!

Soft WDT reset

ctx: cont 
sp: 3ffffd70 end: 3fffffd0 offset: 01b0

>>>stack>>>
3fffff20:  0000003e 0000000d 3ffef58c 40208188  
3fffff30:  0000003e 3ffef340 3ffef58c 40208469  
3fffff40:  3ffe93ac 00000020 3ffe8fc4 40208469  
3fffff50:  3ffe8ad0 feefeffe 3ffef58c 4020861c  
3fffff60:  40106808 4010642c 3ffef58c 40208690  
3fffff70:  0000003e 3ffef340 3ffef58c 40202e7d  
3fffff80:  3ffe8500 3ffef344 3ffef58c 40203346  
3fffff90:  feefeffe feefeffe feefeffe feefeffe  
3fffffa0:  feefeffe feefeffe feefeffe feefeffe  
3fffffb0:  3fffdad0 00000000 3ffef65c 40209600  
3fffffc0:  feefeffe feefeffe 3ffe8608 40100739  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld

MHI-AC-Ctrl starting
Measure frequency for SCK, MOSI and MISO pin
SCK frequency=4296Hz (expected: 3000 .. 3200Hz) out of range!

Soft WDT reset

ctx: cont 
sp: 3ffffd70 end: 3fffffd0 offset: 01b0

>>>stack>>>
3fffff20:  0000003e 0000000d 3ffef58c 40208188  
3fffff30:  0000003e 3ffef340 3ffef58c 40208469  
3fffff40:  3ffe93ac 00000020 3ffe8fc4 40208469  
3fffff50:  3ffe8ad0 feefeffe 3ffef58c 4020861c  
3fffff60:  40106808 4010642c 3ffef58c 40208690  
3fffff70:  0000003e 3ffef340 3ffef58c 40202e7d  
3fffff80:  3ffe8500 3ffef344 3ffef58c 40203346  
3fffff90:  feefeffe feefeffe feefeffe feefeffe  
3fffffa0:  feefeffe feefeffe feefeffe feefeffe  
3fffffb0:  3fffdad0 00000000 3ffef65c 40209600  
3fffffc0:  feefeffe feefeffe 3ffe8608 40100739  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld

MHI-AC-Ctrl starting
Measure frequency for SCK, MOSI and MISO pin
SCK frequency=4171Hz (expected: 3000 .. 3200Hz) out of range!

Soft WDT reset

ctx: cont 
sp: 3ffffd70 end: 3fffffd0 offset: 01b0

>>>stack>>>
3fffff20:  0000003f 0000000d 3ffef58c 40208188  
3fffff30:  0000003f 3ffef340 3ffef58c 40208469  
3fffff40:  3ffe93ac 00000020 3ffe8fc4 40208469  
3fffff50:  3ffe8ad0 feefeffe 3ffef58c 4020861c  
3fffff60:  40106808 4010642c 3ffef58c 40208690  
3fffff70:  0000003f 3ffef340 3ffef58c 40202e7d  
3fffff80:  3ffe8500 3ffef344 3ffef58c 40203346  
3fffff90:  feefeffe feefeffe feefeffe feefeffe  
3fffffa0:  feefeffe feefeffe feefeffe feefeffe  
3fffffb0:  3fffdad0 00000000 3ffef65c 40209600  
3fffffc0:  feefeffe feefeffe 3ffe8608 40100739  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld

MHI-AC-Ctrl starting
Measure frequency for SCK, MOSI and MISO pin
SCK frequency=4160Hz (expected: 3000 .. 3200Hz) out of range!
`

Und noch eine Frage, warum lässt du den Wemos zusätzlich im AP Modus laufen? Hat das einen speziellen Grund?

Gruß

absalom-muc, 2020-03-04:
Ich sehe 2 Möglichkeiten für die zu hohe Frequenz beim SCK:
1. Dein SRK50 ZM-S hat ein anderes Timing als mein SRK 35 ZS-S
2. Dein Compiler kennt nicht die Frequenz, mit der dein ESP läuft, daher kommt es zu einem Messfehler

Mein Tipp wäre die zweite Option. Welches Board verwendest du und stimmt die eingestelle CPU Frequency mit der Realität überein?
[image]
Wenn du die 3 Zeilen mit
`while(1);`
in MHI-AC-Ctrl.h auskommentierst, wird der ESP nicht mehr in die Endlosschleife feststecken.

Ich habe zumindest nicht absichtlich den AP Modus aktiviert - wie kommst du darauf, dass er aktiv ist?

coyote1, 2020-03-05:
Verwende einen Wemos D1 Mini (ESP8266-12F). Ich habe aber jetzt auch 160MHz ausprobiert mit dem gleichen Ergebnis.
Also die 3 Zeilen auskommentiert, dann bekomme ich das im Monitor:

`sync=0
Wrong MOSI signature: 0x6C 0x80 0x04 or checksum received!
async: millis() - lastDatapacketMillis > 35
async: millis() - lastDatapacketMillis > 35
async: millis() - lastDatapacketMillis > 35
async: millis() - lastDatapacketMillis > 35
sync=1
sync=0
Wrong MOSI signature: 0x36 0x40 0x02 or checksum received!
async: millis() - lastDatapacketMillis > 35
async: millis() - lastDatapacketMillis > 35
async: millis() - lastDatapacketMillis > 35
async: millis() - lastDatapacketMillis > 35
sync=1
sync=0
Wrong MOSI signature: 0x6C 0x80 0x04 or checksum received!
async: millis() - lastDatapacketMillis > 35
async: millis() - lastDatapacketMillis > 35
async: millis() - lastDatapacketMillis > 35
async: millis() - lastDatapacketMillis > 35
Wrong MOSI signature: 0x36 0x40 0x02 or checksum received!
async: millis() - lastDatapacketMillis > 35
async: millis() - lastDatapacketMillis > 35
async: millis() - lastDatapacketMillis > 35
async: millis() - lastDatapacketMillis > 35
sync=1
sync=0
Wrong MOSI signature: 0x6C 0x80 0x04 or checksum received!
async: millis() - lastDatapacketMillis > 35
async: millis() - lastDatapacketMillis > 35
async: millis() - lastDatapacketMillis > 35
async: millis() - lastDatapacketMillis > 35
Wrong MOSI signature: 0x36 0x40 0x02 or checksum received!
async: millis() - lastDatapacketMillis > 35
async: millis() - lastDatapacketMillis > 35
async: millis() - lastDatapacketMillis > 35
async: millis() - lastDatapacketMillis > 35
sync=1
sync=0
Wrong MOSI signature: 0x2C 0x40 0x02 or checksum received!
async: millis() - lastDatapacketMillis > 35
async: millis() - lastDatapacketMillis > 35
async: millis() - lastDatapacketMillis > 35
async: millis() - lastDatapacketMillis > 35
sync=1
sync=0
Wrong MOSI signature: 0x6C 0x40 0x02 or checksum received!
async: millis() - lastDatapacketMillis > 35
async: millis() - lastDatapacketMillis > 35
async: millis() - lastDatapacketMillis > 35
async: millis() - lastDatapacketMillis > 35
`
Aber läuft dann trotzdem, kann alles steuern. Kann ich das dann so lassen oder gibts da Probleme, wenn das mit der frequenzmessung nicht passt?

Mit dem AP Modus: Nutze 3 ESP's an 3 Klimageräten und alle 3 werden mir im Netztwerk in den WLAN Einstellungen angezeigt (ESP_XXXX). Kann mich dann auch mit diesem verbinden.
Wenn ich richtig gelesen habe, wird automatisch der AP Modus mit aktiviert, wenn nicht "WiFi.mode(WIFI_STA);" in void setup geschrieben wird, erst dann nimmt er den AP Mode weg. Denke du kennst dich da bestimmt besser aus, aber ich habe das mal ausprobiert und dann ist er auch nicht mehr als AP sichtbar

absalom-muc, 2020-03-06:
DIe Frequenzmessung hatte ich eingebaut, um sicherzustellen, dass der ESP richtig angeschlossen ist. Ansonsten werden die Messergebnisse nicht mehr verwendet. Allerdings deuten die Log-Ausgaben darauf hin, dass es Timing Probleme gibt, da öfters die Synchronisation verloren geht.
Kannst du bitte nochmal ein Log erzeugen während der Startphase, also incl. der Frequenzmessung? Dabei sollte aber die Zeit mitangezeigt werden, dazu bitte den Zeitstempel im Seriellen Monitor aktivieren. Danke!
Wegen STA vs. AP muss ich mich einlesen ...

coyote1, 2020-03-06:
Ok, also kann ich die Version ruhig laufen lassen.
Hier das Log:

`12:41:17.314 -> MHI-AC-Ctrl starting
12:41:17.314 -> Measure frequency for SCK, MOSI and MISO pin
12:41:18.311 -> SCK frequency=4160Hz (expected: 3000 .. 3200Hz) out of range!
12:41:18.311 -> MOSI frequency=600Hz (expected: 300 .. 3000Hz) o.k.
12:41:18.311 -> MISO frequency=0Hz (expected: ~0Hz) o.k.
12:41:18.808 -> . connected to CoyoteIoT, IP address: 192.168.59.5
12:41:18.808 -> OTA Ready, IP address: 192.168.59.5
12:41:18.808 -> MQTTreconnect
12:41:18.808 -> Attempting MQTT connection...connected
12:41:21.303 -> Wrong MOSI signature: 0x36 0x40 0x02 or checksum received!
12:41:21.303 -> async: millis() - lastDatapacketMillis > 35
12:41:21.303 -> async: millis() - lastDatapacketMillis > 35
12:41:21.303 -> async: millis() - lastDatapacketMillis > 35
12:41:21.338 -> async: millis() - lastDatapacketMillis > 35
12:41:21.440 -> Wrong MOSI signature: 0x36 0x40 0x02 or checksum received!
12:41:21.440 -> async: millis() - lastDatapacketMillis > 35
12:41:21.440 -> async: millis() - lastDatapacketMillis > 35
12:41:21.440 -> async: millis() - lastDatapacketMillis > 35
12:41:21.440 -> async: millis() - lastDatapacketMillis > 35
12:41:21.510 -> sync=1
12:41:22.998 -> sync=0
12:41:22.998 -> Wrong MOSI signature: 0x36 0x40 0x02 or checksum received!
12:41:23.036 -> async: millis() - lastDatapacketMillis > 35
12:41:23.036 -> async: millis() - lastDatapacketMillis > 35
12:41:23.036 -> async: millis() - lastDatapacketMillis > 35
12:41:23.036 -> async: millis() - lastDatapacketMillis > 35
12:41:23.072 -> Wrong MOSI signature: 0x36 0x40 0x02 or checksum received!
12:41:23.106 -> async: millis() - lastDatapacketMillis > 35
12:41:23.106 -> async: millis() - lastDatapacketMillis > 35
12:41:23.106 -> async: millis() - lastDatapacketMillis > 35
12:41:23.106 -> async: millis() - lastDatapacketMillis > 35
12:41:23.175 -> sync=1
12:41:23.279 -> sync=0
12:41:23.279 -> Wrong MOSI signature: 0x36 0x40 0x02 or checksum received!
12:41:23.279 -> async: millis() - lastDatapacketMillis > 35
12:41:23.279 -> async: millis() - lastDatapacketMillis > 35
12:41:23.279 -> async: millis() - lastDatapacketMillis > 35
12:41:23.279 -> async: millis() - lastDatapacketMillis > 35
12:41:23.420 -> Wrong MOSI signature: 0x36 0x40 0x02 or checksum received!
12:41:23.420 -> async: millis() - lastDatapacketMillis > 35
12:41:23.420 -> async: millis() - lastDatapacketMillis > 35
12:41:23.420 -> async: millis() - lastDatapacketMillis > 35
12:41:23.420 -> async: millis() - lastDatapacketMillis > 35
12:41:23.492 -> sync=1
12:41:25.248 -> sync=0
12:41:25.248 -> Wrong MOSI signature: 0x6C 0x80 0x04 or checksum received!
12:41:25.248 -> async: millis() - lastDatapacketMillis > 35
12:41:25.248 -> async: millis() - lastDatapacketMillis > 35
12:41:25.248 -> async: millis() - lastDatapacketMillis > 35
12:41:25.248 -> async: millis() - lastDatapacketMillis > 35
12:41:25.346 -> sync=1
12:41:26.355 -> sync=0
12:41:26.355 -> Wrong MOSI signature: 0x36 0x40 0x02 or checksum received!
12:41:26.355 -> async: millis() - lastDatapacketMillis > 35
12:41:26.355 -> async: millis() - lastDatapacketMillis > 35
12:41:26.355 -> async: millis() - lastDatapacketMillis > 35
12:41:26.355 -> async: millis() - lastDatapacketMillis > 35
12:41:26.459 -> sync=1
12:41:27.965 -> sync=0
12:41:27.965 -> Wrong MOSI signature: 0x36 0x40 0x02 or checksum received!
12:41:27.965 -> async: millis() - lastDatapacketMillis > 35
12:41:27.965 -> async: millis() - lastDatapacketMillis > 35
12:41:28.012 -> async: millis() - lastDatapacketMillis > 35
12:41:28.012 -> async: millis() - lastDatapacketMillis > 35
12:41:28.045 -> sync=1
12:41:28.527 -> sync=0
12:41:28.527 -> Wrong MOSI signature: 0x6C 0x80 0x04 or checksum received!
[... 60 lines cut ...]
12:41:34.242 -> async: millis() - lastDatapacketMillis > 35
12:41:34.276 -> async: millis() - lastDatapacketMillis > 35
12:41:34.276 -> async: millis() - lastDatapacketMillis > 35
12:41:34.344 -> sync=1
12:41:35.264 -> sync=0
12:41:35.264 -> Wrong MOSI signature: 0x6C 0x80 0x04 or checksum received!
12:41:35.298 -> async: millis() - lastDatapacketMillis > 35
12:41:35.298 -> async: millis() - lastDatapacketMillis > 35
12:41:35.298 -> async: millis() - lastDatapacketMillis > 35
12:41:35.298 -> async: millis() - lastDatapacketMillis > 35
12:41:35.374 -> sync=1
12:41:35.598 -> sync=0
12:41:35.598 -> Wrong MOSI signature: 0x6C 0x80 0x00 or checksum received!
12:41:35.598 -> async: millis() - lastDatapacketMillis > 35
12:41:35.598 -> async: millis() - lastDatapacketMillis > 35
12:41:35.598 -> async: millis() - lastDatapacketMillis > 35
12:41:35.598 -> async: millis() - lastDatapacketMillis > 35
12:41:35.669 -> Wrong MOSI signature: 0x36 0x40 0x02 or checksum received!
12:41:35.669 -> async: millis() - lastDatapacketMillis > 35
12:41:35.669 -> async: millis() - lastDatapacketMillis > 35
12:41:35.669 -> async: millis() - lastDatapacketMillis > 35
12:41:35.703 -> async: millis() - lastDatapacketMillis > 35
12:41:35.772 -> sync=1`

Siehst du bei dir den ESP als AP? Also ich habe das jetzt eingefügt und dann ist er nicht mehr sichtbar.

absalom-muc, 2020-03-06:
Du kannst die Version weiterlaufen lassen, da kann m.E. nichts passieren, außer dass du Instabilitäten/Aussetzer hast.
Kannst du das log-file nochmal schicken, aber diesmal mit Zeilenumbrüchen? Genre auch als Dateianhang und nicht innerhalb des Posts.

Ich habe einige ESPs bei mir am laufen, keiner kommt als AP daher. Kann es sein, dass du den ESP - bevor du ihn in der Klima verwendet hast - evtl. in einer anderen Anwendung incl. AP benutzt hast und er sich das im Flash gemerkt hat? Wenn ich die Thematik richtig verstehe, solltest du den Befehl
`WiFi.mode(WIFI_STA);`
jetzt wieder aus deinem Programm rausnehmen können, weil er sich den STA mode im Flash gemerkt hat.

coyote1, 2020-03-06:
Ok.
Ja, hier als Datei, wusste nicht wie ich das hier mache mit Zeilenumbruch.

Hm, das könnte sein, werde ich nochmal probieren.

[Log.txt](https://github.com/absalom-muc/MHI-AC-Ctrl/files/4298210/Log.txt)

absalom-muc, 2020-03-06:
Viel besser - danke! Dort ist schön zu sehen, dass die Synchronisation dauernd verloren geht (sync=0) und das passiert offensichtlich wegen des unerwarteten Timings deiner Klima. Es kann daran liegen, dass die SCK Frequenz höher als erwartet ist oder dass zwischen zwei Bytes oder Frames die Pausenzeit verkürzt ist.
WIe auch immer, wenn die Bedienbarkeit nicht eingeschränkt ist, würde ich es so belassen.

coyote1, 2020-03-06:
Ah ok, dachte das muss immer wechseln.
Also, es funktioniert soweit eigentlich recht gut. Ab und an habe ich bei Vanes Swing stehen oder ein Fragezeichen, obwohl ich z.B. 2 ausgewählt habe und die Klappen auf der richtigen Position stehen. Aber ansonsten klappt es.
Dann werde ich deine aktuelle Version auf alle 3 Wemos hochladen. Ich wollte auch noch schauen ,ob OTA funktioniert. Bekomme die nicht in Arduino angezeigt, aber das ist auch nicht so problematisch. Hauptsache die Bedienung und Rückmeldung funktioniert, der Sommer kann kommen ;-)

absalom-muc, 2020-03-06:
Das "?" bei Vanes ist normal, wenn der letzte Vanes Befehl über deine IR-Fernbedienung und nicht über MHI-AC-Ctrl kam. Das ist bisher nur in [SPI.md](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/SPI.md) erklärt, sollte ich wohl auch in [MQTT.md](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/MQTT.md) erwähnen. Die Geschichte mit Vanes Swing ist mir auch schon aufgefallen, ist wahrscheinlich ein Fehler im Programm. Werde ich die Tage beheben.

coyote1, 2020-03-06:
Ok, danke für die Infos. Das hab ich nicht gelesen in SPI.md, sorry :-)
Nochmal danke für deine Arbeit, ist echt gut gelungen, sieht professioneller aus und ich finde es läuft stabiler als das "alte" Projekt von rjdekker

absalom-muc, 2020-03-15:
Danke!
Um das Vanes Problem zu beheben, füge die Zeile 
```          
            tx_SPIframe[DB0] = 0b10000000; // disable swing
```
vor [Zeile 282 in MHI-AC-Ctrl.ino](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/src/MHI-AC-Ctrl.ino#L282) ein. Es sollte dann so aussehen:
```
        if (set_Vanes) {
          if (new_Vanes == 5) { //5: swing
            tx_SPIframe[DB0] = 0b11000000;
          }
          else {
            tx_SPIframe[DB0] = 0b10000000; // disable swing
            tx_SPIframe[DB1] = (1 << 7) | ((new_Vanes - 1) << 4);
          }
          set_Vanes = false;
        }
```

coyote1, 2020-03-18:
Zeile habe ich heute eingefügt, sieht soweit alles gut aus. Der AP Mode am Wemos ist auch weg, nachdem ich mit "flash all contents" geflasht habe, war wohl, wie du gesagt hast, noch im Speicher von einem vorigen Programm.

absalom-muc, 2020-03-18:
gut zu hören, danke für die Rückmeldung.

closed 2020-03-31 by absalom-muc