absalom-muc/MHI-AC-Ctrl #12: SSL eventually?

ligangyi, 2020-05-07:
Have this working on my home network, I was gonna implement SSL on my MQTT broker (Mosquitto), but then I realised as of now, there's no option to support SSL for this project.

absalom-muc, 2020-05-08:
MHI-AC-Ctrl and Mosquitto support SSL, what is the problem?

ligangyi, 2020-05-09:
MHI-AC-Ctrl now implements a plaintext password. TLS/SSL encrypts data transmission and uses port 8883.

absalom-muc, 2020-05-09:
O.k., I see your point. Currently I have not planned to implement SSL/TLS, because I don't use it. When you are interested to add it, the [MQTT client](https://github.com/knolleary/pubsubclient) might be a good starting point. There were some discussions related to TLS/SSL.

closed 2020-07-03 by absalom-muc