absalom-muc/MHI-AC-Ctrl #13: Library integration proposal

b0rder, 2020-05-07:
Hi,

    I am using Tasmota firmware because it has reach set of modules. But it is lack of support MHI control.
I started development Tasmota module to support MHI control, but I found MHI-AC-Ctrl has implemented most needed feature for this.
Probably, the better way will be to refactor current project to be able use it as library which will be easy to integrate or reuse.
What do you think?

absalom-muc, 2020-05-08:
I agree with you that it would probably have been good to start with a framework like Tasmato. But since it runs quite stable, I see no big advantage in refactoring MHI-AC-Ctrl now.
Which benefit would you expect?

b0rder, 2020-05-12:
The main benefit is the one source of your library (current repository), which can be easy to include into other projects (like Tasmota).
New features (or fixes) in the library will be automatically applied to any dependent framework/project.

absalom-muc, 2020-05-12:
O.k., I see your point. I will consider it for the future. But please be aware that the integration in another framework could be timing critical because there are only about 40ms available between two frames for serving WiFi, MQTT, OTA etc.

b0rder, 2020-05-13:
I just started to put your code inside Tasmota and it seems works fine.
The only thing need to do is to move out MQTT-related stuff and system output (serial output) out of the functions (or call it using callbacks which can be passed to library during initialization).

absalom-muc, 2020-05-15:
Since there are inquiries from several sides for the implementation of MHI-AC-Ctrl in different frameworks, I consider - following your suggestion - to separate the core functionality (SPI communication) from the input / output (MQTT, OTA). This could simplify the Tasmato integration for you. (but I expect you finish the integration before I'm ready)

b0rder, 2020-05-18:
I've just committed initial version of your library integrated into Tasmota here: https://github.com/b0rder/Tasmota/commit/ad150afa3ebdf0ee1bcfbffcf05c30ffeeaf7ab8
There are 2 files added (this should be Yours [library](https://github.com/b0rder/Tasmota/tree/development/lib/MHI-AC-Ctrl)):
[lib/MHI-AC-Ctrl/MHI_AC_Ctrl.cpp
](https://github.com/b0rder/Tasmota/blob/development/lib/MHI-AC-Ctrl/MHI_AC_Ctrl.cpp)[lib/MHI-AC-Ctrl/MHI_AC_Ctrl.h
](https://github.com/b0rder/Tasmota/blob/development/lib/MHI-AC-Ctrl/MHI_AC_Ctrl.h)
And other stuff is related to Tasmota:
[tasmota/xdrv_40_mhi.ino
](https://github.com/b0rder/Tasmota/blob/development/tasmota/xdrv_40_mhi.ino)
Compilation of Tasmota is very easy with [platformio](http://platformio.org/):

```
git clone https://github.com/b0rder/Tasmota.git
cd Tasmota
pio run -e tasmota-lite
# flash firmware:
esptool.py --port /dev/cu.wchusbserial1420 --baud 921600 write_flash -fm dout 0x00000 ./.pioenvs/tasmota-lite/firmware.bin

```

After connected to Wifi and entering web Console, we can see AC events and perform commands (using console or MQTT):

[image]

Commands (using web console or MQTT):
mhi On|Off
Mode Auto|Dry|Cool|Fan|Heat
Fan Auto|Low|Med|High
Vanes Down|DownMiddle|UpMiddle|Up|Swing
Tsetpoint 17...30
ErrOpData

There are a lot of TODO's which I would like to do with Tasmota (like add resending MHI events statuses after MQTT reconnect like you have in your implementation) and your original code also need to be refactored to be able use the same library...

absalom-muc, 2020-05-21:
looks good - thanks for the update!

absalom-muc, 2020-06-05:
Today I published a refactored version. There is now a strict separation between the core functionality (SPI read / write) and the user interface stuff. I hope it makes the integration to Tasmota and other frameworks simpler.

dvisser, 2021-01-06:
> I've just committed initial version of your library integrated into Tasmota here: [b0rder/Tasmota@ad150af](https://github.com/b0rder/Tasmota/commit/ad150afa3ebdf0ee1bcfbffcf05c30ffeeaf7ab8)
> There are 2 files added (this should be Yours [library](https://github.com/b0rder/Tasmota/tree/development/lib/MHI-AC-Ctrl)):
> [lib/MHI-AC-Ctrl/MHI_AC_Ctrl.cpp ](https://github.com/b0rder/Tasmota/blob/development/lib/MHI-AC-Ctrl/MHI_AC_Ctrl.cpp)[lib/MHI-AC-Ctrl/MHI_AC_Ctrl.h ](https://github.com/b0rder/Tasmota/blob/development/lib/MHI-AC-Ctrl/MHI_AC_Ctrl.h)
> And other stuff is related to Tasmota:
> [tasmota/xdrv_40_mhi.ino ](https://github.com/b0rder/Tasmota/blob/development/tasmota/xdrv_40_mhi.ino)
> Compilation of Tasmota is very easy with [platformio](http://platformio.org/):
> 
> ```
> git clone https://github.com/b0rder/Tasmota.git
> cd Tasmota
> pio run -e tasmota-lite
> # flash firmware:
> esptool.py --port /dev/cu.wchusbserial1420 --baud 921600 write_flash -fm dout 0x00000 ./.pioenvs/tasmota-lite/firmware.bin
> ```
> 
> After connected to Wifi and entering web Console, we can see AC events and perform commands (using console or MQTT):
> 
> [image]
> 
> Commands (using web console or MQTT):
> mhi On|Off
> Mode Auto|Dry|Cool|Fan|Heat
> Fan Auto|Low|Med|High
> Vanes Down|DownMiddle|UpMiddle|Up|Swing
> Tsetpoint 17...30
> ErrOpData
> 
> There are a lot of TODO's which I would like to do with Tasmota (like add resending MHI events statuses after MQTT reconnect like you have in your implementation) and your original code also need to be refactored to be able use the same library...

@b0rder I have tried to replicate this and test it on a unit with no success. I can see in your xdrv_40_mhi.ino file you have a check to see if "USE_MHI_SPI" is set and I cant find any further reference of this in your files? I don't believe it is ever enabled.

Can you please advise what is needed to enable this so that I can get the console output above similar to yours?

absalom-muc, 2021-08-14:
Hello @b0rder,
I just wanted to try your Tasmota integration but I ran in the same issue as @dvisser. It seems that tasmota/xdrv_40_mhi.ino is not considered for compilation. Do you have an idea how to resolve it?
Thanks
absalom

closed 2022-06-16 by absalom-muc