absalom-muc/MHI-AC-Ctrl #14: Home-assistant setup

apateluk, 2020-05-13:
Many thanks for creating this!

I am trying to integrate into home-assistant. If you are doing the same, any chance you could share your HA configuration please as I'm having some difficulties with the names of the MQTT labels ?

Thanks.

ThomasHFWright, 2020-05-14:
I'm working on this myself, though I've never worked with C++ before so it's a bit slow going.
Attempting to add a ModePower MQTT topic which combines both, and leave the existing Mode and Power topics as is.
Aim is that when the units are off, then the new topic will be OFF, when the unit is on, it will have the same value as the Mode topic.
Current progress: https://github.com/ThomasHFWright/MHI-AC-Ctrl/tree/status_edits

apateluk, 2020-05-14:
I've hacked a modification that's working for me this morning. 

I've removed tthe "Power" topics and anything relating to it completly, and addred an extra state to the Mode.

```
bool is_on = false;

void MQTT_subscribe_callback(char* topic, byte* payload, unsigned int length) {
  char payload_str[20];
  memcpy(payload_str, payload, length);
  payload_str[length] = '\0';

    if (strcmp(topic, MQTT_SET_PREFIX "Mode") == 0) {
      byte write_mask = 0b00111111;
      //Don't write power bit if already on
      if (is_on) 
        write_mask = 0b00111100;

      if (strcmp(payload_str, "off") == 0) {
      set_Mode = true;
      //new_Mode = 0b00100010;
      new_Mode = 0b00000010;
      publish_cmd_ok();
    }
    else if (strcmp(payload_str, "auto") == 0) {
      set_Mode = true;
      new_Mode = 0b00100011 & write_mask;
      publish_cmd_ok();
    }
    else if (strcmp(payload_str, "dry") == 0) {
      set_Mode = true;
      new_Mode = 0b00100111 & write_mask;
      publish_cmd_ok();
    }
    else if (strcmp(payload_str, "cool") == 0) {
      set_Mode = true;
      new_Mode = 0b00101011 & write_mask;
      publish_cmd_ok();
    }
    else if (strcmp(payload_str, "fan_only") == 0) {
      set_Mode = true;
      new_Mode = 0b00101111 & write_mask;
      publish_cmd_ok();
    }
    else if (strcmp(payload_str, "heat") == 0) {
      set_Mode = true;
      new_Mode = 0b00110011 & write_mask;
      publish_cmd_ok();
    }
    else
      publish_cmd_invalidparameter();
  }
  else if (strcmp(topic, MQTT_SET_PREFIX "Tsetpoint") == 0) {
    new_Tsetpoint = atoi(payload_str);
    if ((new_Tsetpoint >= 18) & (new_Tsetpoint <= 30)) {
      set_Tsetpoint = true;
      publish_cmd_ok();
    }
    else
      publish_cmd_invalidparameter();
  }
  else if (strcmp(topic, MQTT_SET_PREFIX "Fan") == 0) {
    new_Fan = atoi(payload_str);
    if ((new_Fan >= 1) & (new_Fan <= 4)) {
      set_Fan = true;
      publish_cmd_ok();
    }
    else
      publish_cmd_invalidparameter();
  }
  else if (strcmp(topic, MQTT_SET_PREFIX "Vanes") == 0) {
    if(strcmp(payload_str, "Swing") == 0) {
      new_Vanes = 5;
      set_Vanes = true;
      publish_cmd_ok();
    }
    else {
      new_Vanes = atoi(payload_str);
      if ((new_Vanes >= 1) & (new_Vanes <= 5)) {
        set_Vanes = true;
        publish_cmd_ok();
      }
      else 
        publish_cmd_invalidparameter();  
    }
  }
  else if (strcmp(topic, MQTT_SET_PREFIX "ErrOpData") == 0) {
    request_erropData = true;
    publish_cmd_ok();
  }
  else if (strcmp(topic, MQTT_SET_PREFIX "reset") == 0) {
    if(strcmp(payload_str, "reset") == 0) {
      publish_cmd_ok();
      delay(500);
      ESP.restart();
    }
    else 
      publish_cmd_invalidparameter();  
  }
  else
    publish_cmd_unknown();
}
```

and

```
if (updateMQTTStatus | ((rx_SPIframe[DB0] & 0x1d) != mode_old)) { // Mode
          mode_old = rx_SPIframe[DB0] & 0x1d;
          if ((mode_old & 0x01) == 0) {
            is_on = false;
            MQTTclient.publish(MQTT_PREFIX "Mode", "off", true);
          } else 
          {
            is_on = true;
            switch (mode_old & 0x1c) {
              case 0x00:
                MQTTclient.publish(MQTT_PREFIX "Mode", "auto", true);
                break;
              case 0x04:
                MQTTclient.publish(MQTT_PREFIX "Mode", "dry", true);
                break;
              case 0x08:
                MQTTclient.publish(MQTT_PREFIX "Mode", "cool", true);
                break;
              case 0x0c:
                MQTTclient.publish(MQTT_PREFIX "Mode", "fan_only", true);
                break;
              case 0x10:
                MQTTclient.publish(MQTT_PREFIX "Mode", "heat", true);
                break;
              default:
                MQTTclient.publish(MQTT_PREFIX "Mode", "invalid", true);
                break;
            }
          }
        }
```

absalom-muc, 2020-05-15:
it is no issue to set the power bit again if the unit is already on. For the future I plan to include an option for this.

apateluk, 2020-05-16:
Thanks for that, I've updated the snippets above taking out checking of the power bit. Also I put your original code back in for separate power control, as don't think there is an issue having both. 

Now you can power on the device by either setting power to on, or setting the mode, and power off by setting power to off or mode to off.

absalom-muc, 2020-06-05:
In v2.0 I added an option to enable this behaviour, please check [SW-Configuration.md](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/SW-Configuration.md#behaviour-when-changing-ac-mode-mhi-ac-ctrlh)

apateluk, 2020-06-26:
Had a quick look at the changes and although i've not tested, based on the description I think it will still not completly work with HA to power off the unit. From a HA persepctive, these are the only modes recognised and all sent via the mode topic.
      - "auto"
      - "dry"
      - "cool"
      - "fan_only"
      - "heat"
      - "off"

absalom-muc, 2020-06-26:
That should be feasible. What is the problem?

apateluk, 2020-06-27:
That's right you can't power offer through the mode-topic or read status.

absalom-muc, 2020-06-28:
I just uploaded a new version (hopefully) supporting the right behaviour for HA. Please test it.

apateluk, 2020-07-01:
Have just tried this and seems to be stuck in a boot loop with the following output.

```
MHI-AC-Ctrl starting
CPU frequency[Hz]=80000000
Attempting WiFi connection .... connected to homewifi, IP address: 192.168.0.225
OTA Ready
Attempting MQTT connection... connected
status=65 topic=connected payload=1

Exception (3):
epc1=0x40216c74 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40247f48 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd20 end: 3fffffc0 offset: 0190
3ffffeb0:  40247f56 40247fa4 3ffffe90 3ffffec0  
3ffffec0:  40247f48 00000000 40100362 3ffee800  
3ffffed0:  00000041 40247f4c 40247f48 402027e4  
3ffffee0:  6e49484d 66726574 2f656361 6e6e6f63  
3ffffef0:  65746365 00000064 00000000 00000000  
3fffff00:  00000000 00000000 00000000 00000000  
3fffff10:  00000000 00000000 00000000 00000000  
3fffff20:  00000000 00000000 00000000 00000000  
3fffff30:  00000000 00000000 00000000 00000000  
3fffff40:  00000000 00000000 3ffee6d4 4020690c  
3fffff50:  3ffee594 00000000 3ffee6d4 40202884  
3fffff60:  00000001 3ffe86a3 3ffee614 40202a5c  
3fffff70:  3ffefc04 00000000 3ffee594 402034ba  
3fffff80:  3fffdad0 00000000 3ffee594 40202437  
3fffff90:  40201a9c feefeffe 4020e9ec 4020e9d4  
3fffffa0:  feefeffe feefeffe 3ffee7c0 40207d7c  
3fffffb0:  feefeffe feefeffe 3ffe850c 40101101  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3456, room 16 
tail 0
chksum 0x84
csum 0x84
va5432625
~ld

MHI-AC-Ctrl starting
```

The only difference to your implementation is that I'm using the Wemos D1 Mini rather than nodemcu, but that wasn't;t an issue last time. Seems quite a bit of code has changed since the initial release so will compare against what was working for me.

absalom-muc, 2020-07-01:
Have you seen that there is a problem with the latest MQTT client library, please see https://github.com/absalom-muc/MHI-AC-Ctrl#software. PubSubClient should be v2.7.0, not v2.8.0

apateluk, 2020-07-01:
That fixed that issue. That'll teach me to RTFM :).

The issue now seems to be the Mode topic is not being published when the unit is switched on.

Steps:

1. Unit is powered off
2. Set Mode to 'cool'
3. Unit powers on and is running in the correct mode ("cool")
4. Published Mode topic still reports "off"
5. Set Mode to 'heat'
6. Unit switches to heat
7. Published Mode topic reports "heat".

As I said above, issue seems to only be occurring when the unit starts in off state.

absalom-muc, 2020-07-01:
Hm , sponatenously no idea. Could you try again and generate a log file (timestamp enabled, 115200 Baud)?

apateluk, 2020-07-02:
I'll have a proper go at debugging and providing a log later today. My unit is in an arkward location that makes it hard to connect a USB cable.

I the meantime I think spot the issue in the code (not certain though).

```
case status_power:
          if (value == power_on)
            output_P(status, (TOPIC_POWER), PSTR(PAYLOAD_POWER_ON));
          else {
            output_P(status, (TOPIC_POWER), (PAYLOAD_POWER_OFF));
#ifdef POWERON_WHEN_CHANGING_MODE
            output_P(status, PSTR(TOPIC_MODE), PSTR(PAYLOAD_MODE_OFF));
#endif
          }
          break;
        case status_mode:
        case opdata_mode:
        case erropdata_mode:
          switch (value) {
            case mode_auto:
              if (status != erropdata_mode)
                output_P(status, PSTR(TOPIC_MODE), PSTR(PAYLOAD_MODE_AUTO));
              else
                output_P(status, PSTR(TOPIC_MODE), PSTR(PAYLOAD_MODE_STOP));
              break;
            case mode_dry:
              output_P(status, PSTR(TOPIC_MODE), PSTR(PAYLOAD_MODE_DRY));
              break;
            case mode_cool:
              output_P(status, PSTR(TOPIC_MODE), PSTR(PAYLOAD_MODE_COOL));
              break;
            case mode_fan:
              output_P(status, PSTR(TOPIC_MODE), PSTR(PAYLOAD_MODE_FAN));
              break;
            case mode_heat:
              output_P(status, PSTR(TOPIC_MODE), PSTR(PAYLOAD_MODE_HEAT));
              break;
          }
          break;
```

I'm guessing that when the unit is powered on it hit's case: "status_power", which only publishes PAYLOAD_POWER_ON. I assume that the unit is not flagging a corespinding mode change. "status_mode" is only hit when the mode is explicitly changed ?

apateluk, 2020-07-02:
Log output below. 
The first bold line is when I switch the AC on, seems only the Power topic is published. The last two lines are when I switch the unit off. There is a bit in between when I change modes whilst the unit is on.
```

MHI-AC-Ctrl starting
CPU frequency[Hz]=80000000
Attempting WiFi connection .... connected to BTWholeHome-XN5, IP address: 192.168.0.225
OTA Ready
Attempting MQTT connection... connected
status=65 topic=connected payload=1
status=64 topic=RSSI payload=-52
Measure frequency for SCK, MOSI and MISO pin
status=68 value=3200
status=68 topic=fSCK payload=3200
status=69 value=560
status=69 topic=fMOSI payload=560
status=70 value=0
status=70 topic=fMISO payload=0
SCK frequency=3200Hz (expected: >3000Hz) o.k.
MOSI frequency=560Hz (expected: <SCK frequency) o.k.
MISO frequency=0Hz (expected: ~0Hz) o.k.
Setup done!
status=72 value=8
status=72 topic=Mode payload=cool
status=71 value=0
status=71 topic=Power payload=off
status=71 topic=Mode payload=off
status=73 value=2
status=73 topic=Fan payload=3
status=74 value=0
status=74 topic=Vanes payload=?
status=75 value=165
status=75 topic=Troom payload=26.00
status=76 value=23
status=76 topic=Tsetpoint payload=23
status=77 value=0
status=77 topic=Errorcode payload=0
status=128 value=8
status=128 topic=Mode payload=cool
status=129 value=23
status=129 topic=Tsetpoint payload=23
status=139 value=4
status=139 topic=TOTAL-COMP-RUN payload=400
status=144 value=0
status=144 topic=PROTECTION-NO payload=0
status=130 value=165
status=130 topic=RETURN-AIR payload=26.25
status=134 value=82
status=134 topic=THI-R1 payload=15
status=133 value=0
status=133 topic=IU-FANSPEED payload=0
status=138 value=3
status=138 topic=TOTAL-IU-RUN payload=300
status=131 value=173
status=131 topic=OUTDOOR payload=19.75
status=135 value=254
status=135 topic=THI-R2 payload=71
status=140 value=0
status=140 topic=COMP payload=0.00
status=71 value=1
**status=71 topic=Power payload=on**
status=132 value=57
status=132 topic=THO-R1 payload=7
status=142 value=24
status=142 topic=TD payload=44
status=136 value=81
status=136 topic=THI-R3 payload=15
status=141 value=0
status=141 topic=CT payload=0.00
status=137 value=0
status=137 topic=OU-FANSPEED payload=0
status=145 value=0
status=145 topic=DEFROST payload=off
status=72 value=16
status=72 topic=Mode payload=heat
status=143 value=19
status=143 topic=TDSH payload=19
status=128 value=16
status=128 topic=Mode payload=heat
status=129 value=25
status=129 topic=Tsetpoint payload=25
status=146 value=470
status=146 topic=OU-EEV1 payload=470
status=72 value=0
status=72 topic=Mode payload=auto
status=76 value=24
status=76 topic=Tsetpoint payload=24
status=134 value=84
status=134 topic=THI-R1 payload=16
status=133 value=2
status=133 topic=IU-FANSPEED payload=2
status=131 value=172
status=131 topic=OUTDOOR payload=19.50
status=72 value=8
status=72 topic=Mode payload=cool
status=76 value=23
status=76 topic=Tsetpoint payload=23
status=71 value=0
**status=71 topic=Power payload=off
status=71 topic=Mode payload=off**
```

absalom-muc, 2020-07-02:
O.k., now I see your point. You used the IR-RC for your tests. I don't longer use the IR-RC, only MHI-AC-Ctrl, therefore I've never tested what happens in case that the IR-RC is used. But I agree, also with the IR-RC the right status should be provided via MHI-AC-Ctrl. I will continue my efforts to find a solution.

apateluk, 2020-07-02:
The issue is the same for if IR-RC is used or MHI-AC-Ctrl is used. 

```
    // evaluate status
    if ((MOSI_frame[DB0] & 0x1c) != status_mode_old) { // Mode
      status_mode_old = MOSI_frame[DB0] & 0x1c;
      m_cbiStatus->cbiStatusFunction(status_mode, status_mode_old);
    }

    if ((MOSI_frame[DB0] & 0x01) != status_power_old) { // Power
      status_power_old = MOSI_frame[DB0] & 0x01;
      m_cbiStatus->cbiStatusFunction(status_power, status_power_old);
    }
```
I assume the top one of these two conditions are not being met when the power is switched on, hence the mode is not being published.

apateluk, 2020-07-03:
I've made a change that I think will work. Possibly not the most elegent, but will test later today.

```
class CallbackInterface_Status {
  public: virtual void cbiStatusFunction(ACStatus status, int value, int mode = 0) = 0;
};
```

```
    if ((MOSI_frame[DB0] & 0x01) != status_power_old) { // Power
      status_power_old = MOSI_frame[DB0] & 0x01;
      m_cbiStatus->cbiStatusFunction(status_power, status_power_old, status_mode_old);
    }
```

```
case status_power:
          if (value == power_on) {
            output_P(status, (TOPIC_POWER), PSTR(PAYLOAD_POWER_ON));
switch (mode) {
            case mode_auto:
                output_P(status, PSTR(TOPIC_MODE), PSTR(PAYLOAD_MODE_AUTO));
              break;
            case mode_dry:
              output_P(status, PSTR(TOPIC_MODE), PSTR(PAYLOAD_MODE_DRY));
              break;
            case mode_cool:
              output_P(status, PSTR(TOPIC_MODE), PSTR(PAYLOAD_MODE_COOL));
              break;
            case mode_fan:
              output_P(status, PSTR(TOPIC_MODE), PSTR(PAYLOAD_MODE_FAN));
              break;
            case mode_heat:
              output_P(status, PSTR(TOPIC_MODE), PSTR(PAYLOAD_MODE_HEAT));
              break;
          }
      }
          else {
            output_P(status, (TOPIC_POWER), (PAYLOAD_POWER_OFF));
#ifdef POWERON_WHEN_CHANGING_MODE
            output_P(status, PSTR(TOPIC_MODE), PSTR(PAYLOAD_MODE_OFF));
#endif
```

absalom-muc, 2020-07-03:
very good 👍 

To be honest, I would prefer a way without changing the interface of cbiStatusFunction. Else other persons who wants to re-use MHI-AC-Ctrl-core have to change their implementation, too. My approach is that MHI-AC-Ctrl-core is only changed if there are core related functionalities which have to be adapted. So the HA specific implementation should not touch MHI-AC-Ctrl-core.

I hacked another way, but have currently not the possibility to test it.
[MHI-AC-Ctrl_v2.02a.txt](https://github.com/absalom-muc/MHI-AC-Ctrl/files/4869534/MHI-AC-Ctrl_v2.02a.txt)
If you want to test it, you have only to rename it to .ino

apateluk, 2020-07-03:
Yep your version works great and it's much better solution. Agree keeping the core functionality platform agnostic is better :). Thanks for doing this so quickly.

absalom-muc, 2020-07-03:
Happy to hear it, then I will add it to the repository next week. Thank you for your quick feedback!

closed 2020-07-09 by absalom-muc