absalom-muc/MHI-AC-Ctrl #17: Add supported MHI models

waynethomas69, 2020-06-06:
Hi not really an issue, but thought you would like to know your base code works with the SRK50ZJX-S1 heat pump also. Only issue noted is the SCK frequency is measured incorrectly, but this doesn't affect operation. Checked SPI output using a logic analyser and actual SCK was around 60kHz (from memory) but measured as 2500Hz.
Haven't tested v2.0 yet (only just noticed you updated).
Cheers
Wayne

absalom-muc, 2020-06-06:
I am absolutely interested in such feedback, thank you. 

The frequency measurement counts the rising edges for 1 second. The result is only meaningful, if the timing of your AC is identical to the timing of the SRK xx ZS-S. That means only if you have 20 frames in one second (1frame=20byte=20 x 8bit=160bit => 20frame x 160bit=3200). So also with the latest version you will see no meaningful results. But as you mentioned, that is no restriction for the normal usage.

ThatsIT67, 2020-06-14:
Hi, I'm using your program for the AC SRK60ZSX-W and it works fine! You have done a great job!!!

absalom-muc, 2020-06-15:
Thank you for your feedback @ThatsIT67. I'm glad to hear that.

makonrus, 2020-06-19:
Hello! Thanks for the project. I connected the air conditioner SRK25ZJX-S.
1) Power On/Off - Perfect!
2) Fan speed. In total, the air conditioner has 3 fan speeds. Do I understand correctly that mode 4 is an automatic mode? - Perfect!
3) Mode. Fail... When I try to turn on any of the modes, the Heat mode is turned on:
```
MQTT_subscribe_callback, topic=MHI-AC-Ctrl/set/Mode payload=Cool
status=67 value=16
status=67 topic=Mode payload=Heat
status=128 value=16
status=128 topic=Mode payload=Heat
status=129 value=26
status=129 topic=Tsetpoint payload=26
status=137 value=4
status=137 topic=OU-FANSPEED payload=4
status=133 value=1
status=133 topic=IU-FANSPEED payload=1
mhi_ac_ctrl_core.loop error: -2
```
4) Setpoint temp. - Perfect!

absalom-muc, 2020-06-20:
Thanks for your feedback. Please find my response below:

>     2. Fan speed. In total, the air conditioner has 3 fan speeds. Do I understand correctly that mode 4 is an automatic mode? - Perfect!
There should be 4 different fan speeds, no auto mode for the fan.

> 3. Mode. Fail... When I try to turn on any of the modes, the Heat mode is turned on:
> ```
> MQTT_subscribe_callback, topic=MHI-AC-Ctrl/set/Mode payload=Cool
> status=67 value=16
> status=67 topic=Mode [...]

Please create a log with time stamp and some more entries before and after the effect. In addition, how does row 30 of [MHI-AC-Ctrl-core.h](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/f762d00571a6530e3519ce7a6949f2a7054d8a81/src/MHI-AC-Ctrl-core.h#L30)
`//#define POWERON_WHEN_CHANGING_MODE true`
looks like?, Unchanged or have you commented it in?

>     2. A large number of errors in the terminal:
> ```
> mhi_ac_ctrl_core.loop error: -2
> mhi_ac_ctrl_core.loop error: -2
> status=130 value=170
> status=130 topic=RETURN-AIR payload=27.50
> mhi_ac_ctrl_core.loop error: -1
> mhi_ac_ctrl_core.loop error: -2
> mhi_ac_ctrl_core.loop error: -2
> mhi_ac_ctrl_core.loop error: -2
> mhi_ac_ctrl_core.loop error: -2
> mhi_ac_ctrl_core.loop error: -2
> mhi_ac_ctrl_core.loop error: -2
> mhi_ac_ctrl_core.loop error: -1
> mhi_ac_ctrl_core.loop error: -2
> mhi_ac_ctrl_core.loop error: -1
> mhi_ac_ctrl_core.loop error: -2
> ```
We've seen this instability alrady e.g. in the [issue10](https://github.com/absalom-muc/MHI-AC-Ctrl/issues/10).  I couldn't find out the root cause, but since this seems to be not critical, I postponed it. btw: If you have an oscilloscope, a measurement would be helpful.

waynethomas69, 2020-06-20:
Me again, also works with my SRF50ZJX-S1. Only issue was this model AC seems to vomit room temperature readings constantly in pairs (0.5 degrees difference). I added an averaging filter for the pairs and another filter to reduce the number of outputs. No issue with your code, just an oddity about this AC model. Thanks again for your great work!!!
Cheers

absalom-muc, 2020-06-20:
Thank you for the feedback @waynethomas69!

absalom-muc, 2020-06-20:
sorry, closed it unintenionally...

absalom-muc, 2020-06-21:
I added a testprogram to folder [testprog](https://github.com/absalom-muc/MHI-AC-Ctrl/tree/master/testprog) and I would appreciate when you could upload here the according logfile. My intention is to find out, if the timing of MHI-AC-Ctrl should be adapted to improve it for further AC models.

scottm1985, 2020-10-07:
@absalom-muc thanks for sharing this, it works perfectly on my SRF35ZMX-S

absalom-muc, 2020-10-07:
Great 👍 thanks for your feedback @scottm1985

L4RGER, 2020-11-19:
@absalom-muc awesome work man, can confirm this works with 
SRK95ZRA-W
Have control through Apple Home app using Home Assistant. 
Changes made as per @deverelo post above

drdoener, 2021-03-06:
Hi, I'm using it for the AC MHI SRK 35 ZSX - W and it works perfect!
It´s a superb project.
Please advise if I can support you anyhow.

absalom-muc, 2021-03-07:
Thanks for your feedback @drdoener - I'm happy to hear that.
Have you used the latest PCB layout from Feb 2021 or the previous version?

drdoener, 2021-03-07:
I use the previous version. Due to the limited space in the device, I focussed already ne new PCB layout. It seems to be really good improvement.

joshuammex, 2021-08-03:
Thanks for your work on this.
The hardware and code works for a SETSRK25ZSA-W (C2.5kW H3.2kW Reverse Cycle Split System). I did have issues with Mosquitto in Home Assistant picking up the new MQQT device (probably unrelated to this project), so I've instead installed the ESPHome fork (https://github.com/ginkage/MHI-AC-Ctrl-ESPHome).

kARTechnology, 2022-01-20:
MHI SRK18YXS2-W6 Model supported. working without any bugs. highest Indian ISEER model, 1.65ton..
installed in the case itself, discreet.

tharealmb, 2022-04-05:
AC MHI SRK 35 ZSX - WT works too. It's the same as the other ZSX that are already listed. Just a different color (W=white, WB=White/Black, WT=Titanium)

FloR012, 2022-12-29:
Hi,
Thanks for your awesome work. I ordered the OCB with the provided Gerber Files from JLCPCB and it works perfectly with my SRF25-ZJX-S and SRF25-ZJX-S. But the PCB does hardly fit into the cabinet so I placed outside next to the power supply cables. There is a lot of space.

The DC-DC converter was hard to get, but the TSR 1-2450E seems to be the successor of TSR 1-2450 and can also be used.
Best regards from Germany

absalom-muc, 2023-01-02:
Thanks for yor feedback @FloR012 

A discussion related to the TSR1-2450E was started [here](https://github.com/absalom-muc/MHI-AC-Ctrl/issues/102). I had also the impression that the 'E'-version it an adequate replacement - good to hear that you can confirm.

Fix1985, 2023-06-18:
Thanks for the good work.
I have a SRK25ZS-W that works great.
I also have an SRF50ZSX-W as a chest.
Unfortunately it is not read. I installed MHI-AC_CTRL as ESP Home. I haven't found the testprog yet.

glsf91, 2023-10-02:
Closed because of no activity for a long time

closed 2023-10-02 by glsf91

Krava615, 2024-04-08:
Hello! Does someone know is it Homebridge compatible?