absalom-muc/MHI-AC-Ctrl #18: v2.0 MQTT Verbindungsprobleme

coyote1, 2020-06-13:
Hey, habe gerade die v2.0 ausprobiert, bei der Version habe ich aber stetige MQTT Neuverbindungen, ca. alle 0,5 Sekunden. Hier ein Auszug aus dem Arduino LOG, da stertet er das Programm scheinbar ständig neu:

19:24:12.423 -> MHI-AC-Ctrl starting
19:24:12.423 -> CPU frequency[Hz]=80000000
19:24:12.423 -> Attempting WiFi connection ....... connected to CoyoteIoT, IP address: 192.168.XX.XX
19:24:14.412 -> OTA Ready
19:24:14.412 -> Attempting MQTT connection... connected
19:24:14.457 -> status=65 topic=connected payload=1
19:24:14.457 -> 
19:24:14.457 -> Exception (3):
19:24:14.457 -> epc1=0x40216db4 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40247e60 depc=0x00000000
19:24:14.457 -> 
19:24:14.457 -> >>>stack>>>
19:24:14.457 -> 
19:24:14.457 -> ctx: cont
19:24:14.457 -> sp: 3ffffd10 end: 3fffffc0 offset: 0190
19:24:14.457 -> 3ffffea0:  40247e6e c0247e63 3ffee9e8 4020b6a4  
19:24:14.457 -> 3ffffeb0:  40247e60 40247e60 3ffee9e8 00000041  
19:24:14.457 -> 3ffffec0:  40247e64 40247e60 3ffee9e8 40202630  
19:24:14.457 -> 3ffffed0:  5f49484d 746e6957 61677265 6e657472  
19:24:14.505 -> 3ffffee0:  6e6f632f 7463656e 00006465 4020df9d  
19:24:14.505 -> 3ffffef0:  3ffe8ac7 00045454 4d002e01 4020df9d  
19:24:14.505 -> 3fffff00:  3ffe8ac7 00000000 00000001 3ffeeb30  
19:24:14.505 -> 3fffff10:  4020b02c 3ffee9e8 3ffe8ac5 4020b038  
19:24:14.505 -> 3fffff20:  4020b02c 3ffee9e8 3ffe8ac5 4020b425  
19:24:14.505 -> 3fffff30:  3ffee834 0000000a 3ffee9e8 4020b6a4  
19:24:14.505 -> 3fffff40:  3ffee834 00000000 3ffee9e8 3ffeeb30  
19:24:14.505 -> 3fffff50:  3ffee834 00000000 3ffee9e8 402026cc  
19:24:14.505 -> 3fffff60:  00000001 3ffe86ad 3ffee8b4 40202850  
19:24:14.505 -> 3fffff70:  3ffefd54 00000000 3ffee834 4020504a  
19:24:14.505 -> 3fffff80:  3fffdad0 00000000 3ffee834 402021f8  
19:24:14.539 -> 3fffff90:  402018c0 feefeffe 4020eb10 4020eaf8  
19:24:14.539 -> 3fffffa0:  feefeffe feefeffe 3ffeeaf0 4020cea8  
19:24:14.539 -> 3fffffb0:  feefeffe feefeffe 3ffe850c 40101135  
19:24:14.539 -> <<<stack<<<
19:24:14.539 -> 
19:24:14.539 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
19:24:14.539 -> 
19:24:14.539 -> load 0x4010f000, len 3456, room 16 
19:24:14.572 -> tail 0
19:24:14.572 -> chksum 0x84
19:24:14.572 -> csum 0x84
19:24:14.572 -> va5432625
19:24:14.572 -> ~ld
19:24:14.654 -> 
19:24:14.654 -> MHI-AC-Ctrl starting
19:24:14.654 -> CPU frequency[Hz]=80000000
19:24:14.654 -> Attempting WiFi connection ....... connected to CoyoteIoT, IP address: 192.168.XX.XX
19:24:16.633 -> OTA Ready
19:24:16.633 -> Attempting MQTT connection... connected
19:24:16.673 -> status=65 topic=connected payload=1
19:24:16.673 -> 
19:24:16.673 -> Exception (3):
19:24:16.673 -> epc1=0x40216db4 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40247e60 depc=0x00000000
19:24:16.673 -> 
19:24:16.673 -> >>>stack>>>
19:24:16.673 -> 
19:24:16.673 -> ctx: cont
19:24:16.673 -> sp: 3ffffd10 end: 3fffffc0 offset: 0190
19:24:16.673 -> 3ffffea0:  40247e6e c0247e63 3ffee9e8 4020b6a4  
19:24:16.673 -> 3ffffeb0:  40247e60 40247e60 3ffee9e8 00000041  
19:24:16.673 -> 3ffffec0:  40247e64 40247e60 3ffee9e8 40202630  
19:24:16.721 -> 3ffffed0:  5f49484d 746e6957 61677265 6e657472  
19:24:16.721 -> 3ffffee0:  6e6f632f 7463656e 00006465 4020df9d  
19:24:16.721 -> 3ffffef0:  3ffe8ac7 00045454 4d002e01 4020df9d  
19:24:16.721 -> 3fffff00:  3ffe8ac7 00000000 00000001 3ffeeb30  
19:24:16.721 -> 3fffff10:  4020b02c 3ffee9e8 3ffe8ac5 4020b038  
19:24:16.721 -> 3fffff20:  4020b02c 3ffee9e8 3ffe8ac5 4020b425  
19:24:16.721 -> 3fffff30:  3ffee834 0000000a 3ffee9e8 4020b6a4  
19:24:16.721 -> 3fffff40:  3ffee834 00000000 3ffee9e8 3ffeeb30  
19:24:16.721 -> 3fffff50:  3ffee834 00000000 3ffee9e8 402026cc  
19:24:16.721 -> 3fffff60:  00000001 3ffe86ad 3ffee8b4 40202850  
19:24:16.755 -> 3fffff70:  3ffefd54 00000000 3ffee834 4020504a  
19:24:16.755 -> 3fffff80:  3fffdad0 00000000 3ffee834 402021f8  
19:24:16.755 -> 3fffff90:  402018c0 feefeffe 4020eb10 4020eaf8  
19:24:16.755 -> 3fffffa0:  feefeffe feefeffe 3ffeeaf0 4020cea8  
19:24:16.755 -> 3fffffb0:  feefeffe feefeffe 3ffe850c 40101135  
19:24:16.755 -> <<<stack<<<
19:24:16.805 -> 
19:24:16.805 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
19:24:16.805 -> 
19:24:16.805 -> load 0x4010f000, len 3456, room 16 
19:24:16.805 -> tail 0
19:24:16.805 -> chksum 0x84
19:24:16.805 -> csum 0x84
19:24:16.805 -> va5432625
19:24:16.805 -> ~ld
19:24:16.839 ->

Tomakava, 2020-06-13:
I have exact same error and I think this problem happens with newest PubSubClient version. It starts working when PubSubClient is downgraded to 2.7.0

absalom-muc, 2020-06-13:
Thank you @Tomakava, you are right. v2.8.0 of the PubSubClient doesn't work. Until I found the cause of the error, please stay with the old version 2.7.0

coyote1, 2020-06-14:
Ok, downgrade to 2.7.0 works. Thank you.

Just another problem, i set it to "On" and after few seconds, it turns automatically "Off" without a mqtt request. Any idea?

09:06:06.896 -> **MQTT_subscribe_callback, topic=MHI_Wintergarten/set/Power payload=On**
09:06:07.643 -> mhi_ac_ctrl_core.loop error: -2
09:06:07.907 -> status=66 value=1
09:06:07.907 -> status=66 topic=Power payload=On
09:06:09.675 -> status=133 value=5
09:06:09.675 -> status=133 topic=IU-FANSPEED payload=5
09:06:09.924 -> mhi_ac_ctrl_core.loop error: -2
09:06:10.819 -> mhi_ac_ctrl_core.loop error: -2
09:06:14.681 -> mhi_ac_ctrl_core.loop error: -2
09:06:15.075 -> status=132 value=51
09:06:15.075 -> status=132 topic=THO-R1 payload=5
09:06:16.412 -> mhi_ac_ctrl_core.loop error: -2
09:06:19.509 -> mhi_ac_ctrl_core.loop error: -2
09:06:20.453 -> mhi_ac_ctrl_core.loop error: -2
09:06:22.340 -> status=141 value=0
09:06:22.340 -> status=141 topic=CT payload=0.00
09:06:22.412 -> mhi_ac_ctrl_core.loop error: -2
09:06:22.689 -> mhi_ac_ctrl_core.loop error: -2
09:06:24.547 -> mhi_ac_ctrl_core.loop error: -2
09:06:29.777 -> mhi_ac_ctrl_core.loop error: -2
09:06:33.063 -> status=66 value=0
09:06:33.063 -> **status=66 topic=Power payload=Off**
09:06:33.309 -> status=70 value=154
09:06:33.309 -> status=70 topic=Troom payload=23.25
09:06:33.380 -> status=70 value=151
09:06:33.380 -> status=70 topic=Troom payload=22.50
09:06:33.632 -> mhi_ac_ctrl_core.loop error: -2
09:06:34.649 -> status=70 value=153
09:06:34.649 -> status=70 topic=Troom payload=23.00
09:06:36.224 -> mhi_ac_ctrl_core.loop error: -2
09:06:36.432 -> status=70 value=151

absalom-muc, 2020-06-14:
The SPI connection appears to be very fragile. You have many _err_msg_invalid_checksum_ errors (mhi_ac_ctrl_core.loop error: -2). This is unusual, especially because the signature bytes are o.k. Did you have the problem in an earlier version (e.g. v1.4)?

coyote1, 2020-06-14:
No, v1.4 works fine on all 3 Devices

absalom-muc, 2020-06-14:
You could try to increase the CPU frequency to 160Mhz, but that's just guessed. I have no idea what the problem is.

coyote1, 2020-06-14:
unfortunately no luck, even with 160 MHz frequency, same behavior. After few seconds it turns off.

absalom-muc, 2020-06-15:
Could you comment-out all operating data in MHI-AC-Ctrl-core.h (beside the dummy) and try again?

coyote1, 2020-06-15:
Yes, I'll try tonight.
 Two SRK25 ZM-S and one SRK50 ZM-S

absalom-muc, 2020-06-15:
I was just able to comprehend the error, some more checks are required. But no more testing by you needed currently

absalom-muc, 2020-06-15:
It is fixed. There was a bug (wrong value MISO_DB6=0xff) in MHI-AC-Ctrl-core.h. This led to the shutdown of the air conditioning system.

coyote1, 2020-06-15:
OK, cool. I will try the fixed version.

coyote1, 2020-06-17:
Works fine, thanks for your work

closed 2020-06-17 by coyote1