absalom-muc/MHI-AC-Ctrl #21: Problem with SRK25ZJX-S

makonrus, 2020-06-22:
> Please create a log with time stamp and some more entries before and after the effect. In addition, how does row 30 of [MHI-AC-Ctrl-core.h](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/f762d00571a6530e3519ce7a6949f2a7054d8a81/src/MHI-AC-Ctrl-core.h#L30)
> `//#define POWERON_WHEN_CHANGING_MODE true`
> looks like?, Unchanged or have you commented it in?

I just commented this line and the problems with switching modes disappeared. But if this line is not commented, then the problem remains.

>`There should be 4 different fan speeds, no auto mode for the fan.`

My remote control has 3 speeds and Auto mode:
[http://splitoff.ru/wp-content/uploads/pults/ESC-RC-195.jpg](url)

> We've seen this instability alrady e.g. in the [issue10](https://github.com/absalom-muc/MHI-AC-Ctrl/issues/10). I couldn't find out the root cause, but since this seems to be not critical, I postponed it. btw: If you have an oscilloscope, a measurement would be helpful.

Yes, I have an oscilloscope, but I can connect it to the AC no earlier than next week.

Here is the test program log:
```
Frequency measurement v1.0
CPU frequency [Hz] : 80000000

acquisition-please wait, ready: acq. time=982ms f=2560 time to 1st falling edge=6004µs
more than 3 ranges detected, skip measurement, sum acq_fails=1

acquisition-please wait, ready: acq. time=921ms f=2400 time to 1st falling edge=37013µs
more than 3 ranges detected, skip measurement, sum acq_fails=2

acquisition-please wait, ready: acq. time=983ms f=2560 time to 1st falling edge=7357µs
more than 3 ranges detected, skip measurement, sum acq_fails=3

acquisition-please wait, ready: acq. time=919ms f=2400 time to 1st falling edge=39466µs
more than 3 ranges detected, skip measurement, sum acq_fails=4

acquisition-please wait, ready: acq. time=982ms f=2559 time to 1st falling edge=7883µs
more than 3 ranges detected, skip measurement, sum acq_fails=5

acquisition-please wait, ready: acq. time=921ms f=2400 time to 1st falling edge=39438µs
more than 3 ranges detected, skip measurement, sum acq_fails=6

acquisition-please wait, ready: acq. time=983ms f=2560 time to 1st falling edge=8291µs
more than 3 ranges detected, skip measurement, sum acq_fails=7

acquisition-please wait, ready: acq. time=918ms f=2400 time to 1st falling edge=39714µs
more than 3 ranges detected, skip measurement, sum acq_fails=8

acquisition-please wait, ready: acq. time=982ms f=2560 time to 1st falling edge=8167µs
more than 3 ranges detected, skip measurement, sum acq_fails=9

acquisition-please wait, ready: acq. time=920ms f=2400 time to 1st falling edge=39546µs
more than 3 ranges detected, skip measurement, sum acq_fails=10

acquisition-please wait, ready: acq. time=983ms f=2560 time to 1st falling edge=8691µs
more than 3 ranges detected, skip measurement, sum acq_fails=11

acquisition-please wait, ready: acq. time=920ms f=2400 time to 1st falling edge=40235µs
more than 3 ranges detected, skip measurement, sum acq_fails=12

acquisition-please wait, ready: acq. time=982ms f=2560 time to 1st falling edge=9158µs
more than 3 ranges detected, skip measurement, sum acq_fails=13

acquisition-please wait, ready: acq. time=919ms f=2400 time to 1st falling edge=40066µs
more than 3 ranges detected, skip measurement, sum acq_fails=14

acquisition-please wait, ready: acq. time=982ms f=2560 time to 1st falling edge=9013µs
more than 3 ranges detected, skip measurement, sum acq_fails=15

acquisition-please wait, ready: acq. time=918ms f=2400 time to 1st falling edge=40543µs
more than 3 ranges detected, skip measurement, sum acq_fails=16

acquisition-please wait, ready: acq. time=983ms f=2560 time to 1st falling edge=8463µs
more than 3 ranges detected, skip measurement, sum acq_fails=17
[... 231 lines cut ...]

acquisition-please wait, ready: acq. time=983ms f=2560 time to 1st falling edge=9936µs
more than 3 ranges detected, skip measurement, sum acq_fails=95

acquisition-please wait, ready: acq. time=919ms f=2400 time to 1st falling edge=40667µs
more than 3 ranges detected, skip measurement, sum acq_fails=96

acquisition-please wait, ready: acq. time=982ms f=2560 time to 1st falling edge=9556µs
more than 3 ranges detected, skip measurement, sum acq_fails=97

acquisition-please wait, ready: acq. time=920ms f=2400 time to 1st falling edge=40249µs
more than 3 ranges detected, skip measurement, sum acq_fails=98

acquisition-please wait, ready: acq. time=983ms f=2560 time to 1st falling edge=9995µs
more than 3 ranges detected, skip measurement, sum acq_fails=99

acquisition-please wait, ready: acq. time=919ms f=2400 time to 1st falling edge=41784µs
more than 3 ranges detected, skip measurement, sum acq_fails=100

acquisition-please wait, ready: acq. time=983ms f=2560 time to 1st falling edge=9739µs
more than 3 ranges detected, skip measurement, sum acq_fails=101

acquisition-please wait, ready: acq. time=920ms f=2400 time to 1st falling edge=40809µs
more than 3 ranges detected, skip measurement, sum acq_fails=102
```

absalom-muc, 2020-06-22:
> I just commented this line and the problems with switching modes disappeared. But if this line is not commented, then the problem remains.

I believe the bug is in [row 137](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/c0b168714d072bba03c8d2b6c24f13be72ff41a6/src/MHI-AC-Ctrl-core.cpp#L137). Please replace
`        MISO_frame[DB0] |= 0x11;`
by
`        MISO_frame[DB0] |= 0b11;`
and try it again. Will come back to you for the other topics.

absalom-muc, 2020-06-22:
> > `There should be 4 different fan speeds, no auto mode for the fan.`
> 
> My remote control has 3 speeds and Auto mode:
> [http://splitoff.ru/wp-content/uploads/pults/ESC-RC-195.jpg](url)

The IR remote control supports  Fan level 1 ... 4 and 'auto'. Unfortunately Fan 'auto' is not supported via a SPI remote control.

makonrus, 2020-06-23:
> I believe the bug is in [row 137 of MHI-AC-Ctrl-core.cpp](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/c0b168714d072bba03c8d2b6c24f13be72ff41a6/src/MHI-AC-Ctrl-core.cpp#L137). Please replace
> ` MISO_frame[DB0] |= 0x11;`
> by
> ` MISO_frame[DB0] |= 0b11;`
> and try it again. Will come back to you for the other topics.

Yes! Now everything works fine.

> The IR remote control supports Fan level 1 ... 4 and 'auto'. Unfortunately Fan 'auto' is not supported via a SPI remote control.

That is, the Auto fan mode and the vanes position switching mode are not sent by the AC to the SPI bus? It would be necessary to check by oscilloscope.

absalom-muc, 2020-06-23:
> Yes! Now everything works fine.
Thanks for confirmation!

absalom-muc, 2020-06-24:
I just uploaded a [SPI logger](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/testprog/SPI_logger.ino). Please use it for creating another log.

absalom-muc, 2020-08-09:
If you are still interested in a follow up, please feel free to re-open this issue

closed 2020-08-09 by absalom-muc