absalom-muc/MHI-AC-Ctrl #22: MISO frequency issue?

Daryn-L, 2020-06-26:
Hey mate,

I wonder if you can help me.  I have just soldered up your circuit and have a problem with the MISO frequency it seems.

Here is the running log:
```
MHI-AC-Ctrl starting
CPU frequency[Hz]=80000000
Found 0 DS18xxx family devices.
Unable to find address for Device 0
Device 0 Address: 0000000000000000
Attempting WiFi connection ....... connected to 42 Grey Street, IP address: 192.168.1.91
OTA Ready
Attempting MQTT connection... connected
status=65 topic=connected payload=1
status=64 topic=RSSI payload=-89
Measure frequency for SCK, MOSI and MISO pin
SCK frequency=3196Hz (expected: >3000Hz) o.k.
MOSI frequency=478Hz (expected: <SCK frequency) o.k.
MISO frequency=3435Hz (expected: ~0Hz) out of range!

Soft WDT reset

>>>stack>>>

ctx: cont
sp: 3ffffdb0 end: 3fffffc0 offset: 01a0
3fffff50:  3ffee8bc 3ffee8c4 3ffeeb34 40202b8d  
3fffff60:  35333433 3ffe8900 3ffeea00 40202a00  
3fffff70:  0039382d 3ffee8d8 3ffee908 3ffeec7c  
3fffff80:  3fffdad0 00000000 3ffee908 402021ff  
3fffff90:  402018c0 feefeffe 4020f600 4020f5e8  
3fffffa0:  feefeffe feefeffe 3ffeec3c 4020d9a0  
3fffffb0:  feefeffe feefeffe 3ffe850c 40101135  
<<<stack<<<

ets Jan  8 2013,rst cause:2, boot mode:(3,2)

load 0x4010f000, len 3456, room 16 
tail 0
chksum 0x84
csum 0x84
va5432625
~ld
```

Here is the freqency log:

```
acquisition-please wait, ready: acq. time=999ms f=3200 time to 1st falling edge=35µs
cycle_time_min[0]=33µs, delta_time[0]=1µs, cycle_time_cnt=2800
cycle_time_min[1]=266µs, delta_time[1]=3µs, cycle_time_cnt=379
cycle_time_min[2]=40268µs, delta_time[2]=1µs, cycle_time_cnt=20
delta_cycle_time_max 9 6 4

acquisition-please wait, ready: acq. time=959ms f=3200 time to 1st falling edge=13156µs
cycle_time_min[0]=33µs, delta_time[0]=1µs, cycle_time_cnt=2800
cycle_time_min[1]=265µs, delta_time[1]=4µs, cycle_time_cnt=380
cycle_time_min[2]=40266µs, delta_time[2]=3µs, cycle_time_cnt=19
delta_cycle_time_max 9 6 4
```

And the SPI log:
https://www.dropbox.com/s/1co2uv9fxcv0zp7/SPI%20Log.txt?dl=0

I'm not really sure how any of this works, but if you could give me an idea of what might be wrong it would be greatly appreciated.

Daryn-L, 2020-06-26:
Actually I think I may have bought the wrong type of logic shifter, 2-channel instead of 4-channel...

damienmorrissey, 2020-06-26:
That'd do it.

absalom-muc, 2020-06-26:
The MISO which is an output of the ESP8266 detected a signal during start-up, and that shouldn't be. Maybe you have a short between MISO and SCK.

Daryn-L, 2020-06-28:
Yes I just had bought the wrong level shifter.  It is working now.  But there is a new issue which I think warrants a new thread...

closed 2020-06-28 by absalom-muc