absalom-muc/MHI-AC-Ctrl #23: Home Assistant

artbrayko, 2020-06-26:
Do you have an example of integration in Home assistant?

absalom-muc, 2020-06-26:
Unfortunatelly I've no experience with Home assistant. In general you have to setup a MQTT server for receiving the messages from MHI-AC-Ctrl. But maybe @apateluk or @ThomasHFWright who discussed a Home assistant topic in [this issue](https://github.com/absalom-muc/MHI-AC-Ctrl/issues/14) could support you.

apateluk, 2020-06-26:
You need to add the climate mqtt platform to your configuration. Note, I havn't tried the latest build of MHI-AC-Ctrl yet, but is meant to be more compatible with HA. 
Below is my configuration,  but note some of the string and topic names may be different to the defaults and HA likes everything to be lowercase.

`climate:
  - platform: mqtt
    unique_id: mhi0001a
    name: LoftAC
    initial: 22
    min_temp: 18
    max_temp: 30
    modes:
      - "auto"
      - "dry"
      - "cool"
      - "fan_only"
      - "heat"
      - "off"
    swing_modes:
      - "1"
      - "2"
      - "3"
      - "4"
      - "Swing"
    fan_modes:
      - "1"
      - "2"
      - "3"
      - "4"
    mode_command_topic: "MHInterface/set/Mode"
    mode_state_topic: "MHInterface/Mode"
    temperature_command_topic: "MHInterface/set/Tsetpoint"
    temperature_state_topic: "MHInterface/Tsetpoint"
    fan_mode_command_topic: "MHInterface/set/Fan"
    fan_mode_state_topic: "MHInterface/Fan"
    swing_mode_command_topic: "MHInterface/set/Vanes"
    swing_mode_state_topic: "MHInterface/Vanes"
    current_temperature_topic: "MHInterface/Troom"`

artbrayko, 2020-06-26:
Thanks, at least some understanding. If you still know what, please write.

artbrayko, 2020-08-25:
I added to configuration.yaml as you wrote above,
only the topic names were left by default (MHI-AC-Ctrl) but nothing happens. Please tell me what am I doing wrong?
In MQTT Explorer I can see that data is being transferred.
.........  
............
    mode_command_topic: "MHI-AC-Ctrl/set/Mode"
    mode_state_topic: "MHI-AC-Ctrl/Mode"
    temperature_command_topic: "MHI-AC-Ctrl/set/Tsetpoint"
    temperature_state_topic: "MHI-AC-Ctrl/Tsetpoint"
    fan_mode_command_topic: "MHI-AC-Ctrl/set/Fan"
    fan_mode_state_topic: "MHI-AC-Ctrl/Fan"
    swing_mode_command_topic: "MHI-AC-Ctrl/set/Vanes"
    swing_mode_state_topic: "MHI-AC-Ctrl/Vanes"
    current_temperature_topic: "MHI-AC-Ctrl/Troom"

sstmelbourne, 2020-09-17:
At first thanks @absalom-muc for the great effort in the compiling. 

I have set up my yaml as above, everything works except;

AC unit is off - can't switch the ac unit on from the UI of HA. However when a command "On" is sent to the the topic MHI-AC-Ctrl/set/Power via an mqtt.fx the AC unit turns on. 

Once the AC is on, modes (heat, cool, dry) from the front UI of HA function, it will change on the AC as well.

When I switch to "Off" front he UI of HA it doesn't work and unit doesn't respond. - error I get when looked up on serial communication of D1 mini is "invalid command". 

However, when an "Off" command is published from the mqtt.fx to the power topic it switches off. 

In summary the on and off command are not working from the UI of the HA. 

Anyone come up with this issue, if so any solution. 

Thanks

L4RGER, 2021-01-21:
did you check the updated wiring picture absalin posted?

Also whats the topic for the DS18B20, i cannot get a readinf out of it. 
i tried subscribing to TOPIC_TDS1820 "Tds1820"

absalom-muc, 2022-06-16:
closed since there was no update for a long time. Please feel free to re-open in case of follow-up topics.

hart0x, 2022-06-17:
> At first thanks @absalom-muc for the great effort in the compiling.
>
> I have set up my yaml as above, everything works except;
>
> AC unit is off - can't switch the ac unit on from the UI of HA. However [...]

If you add this line in HA:
    power_command_topic: "MHI-AC-Ctrl/set/Power"  
and change the following 2 lines in MHI-AC-Ctrl.h from:
#define PAYLOAD_POWER_ON "On"
#define PAYLOAD_POWER_OFF "Off"
to:
#define PAYLOAD_POWER_ON "ON"
#define PAYLOAD_POWER_OFF "OFF"
It seems to work then..

scottm1985, 2023-11-09:
Has anybody got this working in Home Assistant with the new config layout?
I've got temperature working but for some reason I can't see the current operation or power status even though the MQTT topics are populated.  This is the configuration I have currently:
`
mqtt:
  climate:
    unique_id: mhi0001a
    name: HeatPump
    initial: 18
    min_temp: 18
    max_temp: 30
    modes:
      - "auto"
      - "dry"
      - "cool"
      - "fan_only"
      - "heat"
      - "off"
    swing_modes:
      - "1"
      - "2"
      - "3"
      - "4"
      - "Swing"
    fan_modes:
      - "1"
      - "2"
      - "3"
      - "4"
    mode_command_topic: "MHI-AC-Ctrl/set/Mode"
    mode_state_topic: "MHI-AC-Ctrl/Mode"
    temperature_command_topic: "MHI-AC-Ctrl/set/Tsetpoint"
    temperature_state_topic: "MHI-AC-Ctrl/Tsetpoint"
    fan_mode_command_topic: "MHI-AC-Ctrl/set/Fan"
    fan_mode_state_topic: "MHI-AC-Ctrl/Fan"
    swing_mode_command_topic: "MHI-AC-Ctrl/set/Vanes"
    swing_mode_state_topic: "MHI-AC-Ctrl/Vanes"
    current_temperature_topic: "MHI-AC-Ctrl/Troom"
`

Timple, 2023-11-09:
Honestly I just use: https://github.com/ginkage/MHI-AC-Ctrl-ESPHome 
It's based on this nice project. But works out of the box for home assistant.

Allinijhof, 2023-11-21:
I have been able to toggle / Target / Fan mode and Swing mode successfully, hence unfortunately I am not able to enable the features displayed under Mode.

If I select "Heat" = I can see a message has been send through MQTT topic: MHI-AC-Ctrl/set/Mode = heat

I can see a response through MQTT topic: MHI-AC-Ctrl/cmd_received = invalid parameter

Can you advice how to resolve the parameter issue?

Please find attached the software I have loaded on to the Wemos D1;

[MHI-AC-Ctrl.zip](https://github.com/absalom-muc/MHI-AC-Ctrl/files/13430988/MHI-AC-Ctrl.zip)

Please find below a screenshot of Home Assistant entity and MQTT explorer;

[image]

[image]

[image]

Please find below the code in configuration.yaml

mqtt:
  climate:
    unique_id: mhiacctrl
    name: mhictrlac
    initial: 15
    min_temp: 15
    max_temp: 30
    modes:
      - "auto"
      - "dry"
      - "cool"
      - "fan_only"
      - "heat"
      - "off"
    swing_modes:
      - "1"
      - "2"
      - "3"
      - "4"
      - "Swing"
    fan_modes:
      - "1"
      - "2"
      - "3"
      - "4"
    mode_command_topic: "MHI-AC-Ctrl/set/Mode"
    mode_state_topic: "MHI-AC-Ctrl/Mode"
    temperature_command_topic: "MHI-AC-Ctrl/set/Tsetpoint"
    temperature_state_topic: "MHI-AC-Ctrl/Tsetpoint"
    fan_mode_command_topic: "MHI-AC-Ctrl/set/Fan"
    fan_mode_state_topic: "MHI-AC-Ctrl/Fan"
    swing_mode_command_topic: "MHI-AC-Ctrl/set/Vanes"
    swing_mode_state_topic: "MHI-AC-Ctrl/Vanes"
    current_temperature_topic: "MHI-AC-Ctrl/Troom"

Many thanks for your help!

Regards,
Allard

scottm1985, 2023-11-21:
I figured this out after reading the documentation.  There is a line in support.h
```cpp
#define POWERON_WHEN_CHANGING_MODE true
```
that needs to be uncommented to be able to turn on and off with Home Assistant.
And then you need to change MHI-AC-Ctrl.h so that your mode payloads are as follows
```cpp
#define PAYLOAD_MODE_AUTO "auto"
#define PAYLOAD_MODE_STOP "stop"
#define PAYLOAD_MODE_DRY "dry"
#define PAYLOAD_MODE_COOL "cool"
#define PAYLOAD_MODE_FAN "fan_only"
#define PAYLOAD_MODE_HEAT "heat"
```
Home Assistant is receiving "Heat" when it should be "heat".
Your YAML looks correct.

Allinijhof, 2023-11-22:
Excellent it working perfectly!

Allinijhof, 2023-11-24:
@scottm1985  I have do want to add a second AC to Home Assistant, I have created a prefix HOSTNAME Room1 and Room2 however unfortunately the Dasboard in Home Assistant is no longer working. Once I delete either Room1 or Room2 (so basically just have one HOSTNAME / HOSTNAME prefix) the dashboard is working.

Could you please advise what I am missing?
 
#define MQTT_PREFIX HOSTNAME "/Room1/"                    // basic prefix used for publishing AC data (e.g. for status),
                                                    // replace "/" by e.g. "/Living-Room/" when you have multiple ACs
 
Please find below the details of the configuration.yaml file in Home Assistant.

mqtt:
  climate:
    unique_id: MHI-AC-Ctrl-Room1
    name: MHI-AC-Ctrl-Room1
    initial: 15
    min_temp: 15
    max_temp: 30
    modes:
      - "auto"
      - "dry"
      - "cool"
      - "fan_only"
      - "heat"
      - "off"
    swing_modes:
      - "1"
      - "2"
      - "3"
      - "4"
      - "Swing"
    fan_modes:
      - "1"
      - "2"
      - "3"
      - "4"
    mode_command_topic: "MHI-AC-Ctrl/Room1/set/Mode"
    mode_state_topic: "MHI-AC-Ctrl/Room1/Mode"
    temperature_command_topic: "MHI-AC-Ctrl/Room1/set/Tsetpoint"
    temperature_state_topic: "MHI-AC-Ctrl/Room1/Tsetpoint"
    fan_mode_command_topic: "MHI-AC-Ctrl/Room1/set/Fan"
    fan_mode_state_topic: "MHI-AC-Ctrl/Room1/Fan"
    swing_mode_command_topic: "MHI-AC-Ctrl/Room1/set/Vanes"
    swing_mode_state_topic: "MHI-AC-Ctrl/Room1/Vanes"
    current_temperature_topic: "MHI-AC-Ctrl/Room1/Troom"
  climate:
    unique_id: MHI-AC-Ctrl-Room2
    name: MHI-AC-Ctrl-Room2
    initial: 15
    min_temp: 15
    max_temp: 30
    modes:
      - "auto"
      - "dry"
      - "cool"
      - "fan_only"
      - "heat"
      - "off"
    swing_modes:
      - "1"
      - "2"
      - "3"
      - "4"
      - "Swing"
    fan_modes:
      - "1"
      - "2"
      - "3"
      - "4"
    mode_command_topic: "MHI-AC-Ctrl/Room2/set/Mode"
    mode_state_topic: "MHI-AC-Ctrl/Room2/Mode"
    temperature_command_topic: "MHI-AC-Ctrl/Room2/set/Tsetpoint"
    temperature_state_topic: "MHI-AC-Ctrl/Room2/Tsetpoint"
    fan_mode_command_topic: "MHI-AC-Ctrl/Room2/set/Fan"
    fan_mode_state_topic: "MHI-AC-Ctrl/Room2/Fan"
    swing_mode_command_topic: "MHI-AC-Ctrl/Room2/set/Vanes"
    swing_mode_state_topic: "MHI-AC-Ctrl/Room2/Vanes"
    current_temperature_topic: "MHI-AC-Ctrl/Room2/Troom"

Beste regards,

Allard

Allinijhof, 2023-11-26:
By creating unique HOSTNAME's and PREFIX'es MQTT has been able to divide the messages. The issue has been resolved and I am now able to control my three AC's through Home Assistant. Many thanks for your help!

glsf91, 2023-11-27:
I added this issue in the Integration examples list for as Home Assistant.
As it is solved I will close this item.

closed 2023-11-27 by glsf91