absalom-muc/MHI-AC-Ctrl #24: Mqtt commands not effecting heat pump.  Can receive but not set.

Daryn-L, 2020-06-28:
I have my unit up and running.

I can receive all the status' of the heat pump correctly in Home Assistant.  But setting the states has no effect on the heat pump.
Here is my log:

init
`14:12:42.945 -> MHI-AC-Ctrl starting
14:12:42.945 -> CPU frequency[Hz]=80000000
14:12:42.945 -> Found 0 DS18xxx family devices.
14:12:42.986 -> Unable to find address for Device 0
14:12:42.986 -> Attempting WiFi connection ....... connected to 42 Grey Street, IP address: 192.168.1.75
14:12:45.023 -> OTA Ready
14:12:45.023 -> Attempting MQTT connection... connected
14:12:48.873 -> status=65 topic=connected payload=1
14:12:48.873 -> status=64 topic=RSSI payload=-78
14:12:48.873 -> Measure frequency for SCK, MOSI and MISO pin
14:12:49.897 -> status=68 value=3200
14:12:49.897 -> status=68 topic=fSCK payload=3200
14:12:49.897 -> status=69 value=464
14:12:49.897 -> status=69 topic=fMOSI payload=464
14:12:49.897 -> status=70 value=0
14:12:49.897 -> status=70 topic=fMISO payload=0
14:12:49.897 -> SCK frequency=3200Hz (expected: >3000Hz) o.k.
14:12:49.897 -> MOSI frequency=464Hz (expected: <SCK frequency) o.k.
14:12:49.897 -> MISO frequency=0Hz (expected: ~0Hz) o.k.
`

**At first I am trying to set the different states to the heat pump via Home Assistant.  Nothing happens to the heat pump on any of these commands:**

14:12:49.977 -> status=71 value=0
14:12:49.977 -> status=71 topic=Power payload=OFF
14:12:49.977 -> status=72 value=0
14:12:49.977 -> status=72 topic=Mode payload=auto
14:12:49.977 -> status=73 value=2
14:12:49.977 -> status=73 topic=Fan payload=3
14:12:49.977 -> status=74 value=0
14:12:49.977 -> status=74 topic=Vanes payload=?
14:12:49.977 -> status=75 value=152
14:12:49.977 -> status=75 topic=Troom payload=22.75
14:12:49.977 -> status=76 value=24
14:12:49.977 -> status=76 topic=Tsetpoint payload=24
14:12:49.977 -> status=77 value=0
14:12:49.977 -> status=77 topic=Errorcode payload=0
14:13:09.047 -> MQTT_subscribe_callback, topic=MHInterface/set/Mode payload=off payload_length=3
14:13:09.047 -> status=66 topic=cmd_received payload=invalid parameter
14:13:14.983 -> MQTT_subscribe_callback, topic=MHInterface/set/Mode payload=heat payload_length=4
14:13:14.983 -> status=66 topic=cmd_received payload=o.k.
14:13:19.926 -> new DS18x20 temperature=-127.00°C
14:13:19.926 -> status=67 topic=Tds1820 payload=-127.00
14:13:31.332 -> MQTT_subscribe_callback, topic=MHInterface/set/Power payload=ON payload_length=2
14:13:31.332 -> status=66 topic=cmd_received payload=o.k.
14:13:33.834 -> MQTT_subscribe_callback, topic=MHInterface/set/Mode payload=auto payload_length=4
14:13:33.834 -> status=66 topic=cmd_received payload=o.k.
14:13:35.938 -> MQTT_subscribe_callback, topic=MHInterface/set/Tsetpoint payload=23.0 payload_length=4
14:13:35.938 -> status=66 topic=cmd_received payload=o.k.

**Then using the remote to control the heat pump (which works)**

14:15:42.582 -> status=71 value=1
14:15:42.582 -> status=71 topic=Power payload=ON
14:15:59.662 -> status=76 value=26
14:15:59.662 -> status=76 topic=Tsetpoint payload=26
14:16:02.716 -> status=73 value=0
14:16:02.716 -> status=73 topic=Fan payload=1
14:16:03.918 -> status=75 value=150
14:16:03.918 -> status=75 topic=Troom payload=22.25
14:16:04.272 -> status=73 value=1
14:16:04.272 -> status=73 topic=Fan payload=2
14:16:05.976 -> status=73 value=2
14:16:05.976 -> status=73 topic=Fan payload=3
14:16:09.021 -> status=73 value=3
14:16:09.021 -> status=73 topic=Fan payload=4
14:16:14.642 -> status=73 value=2
14:16:14.642 -> status=73 topic=Fan payload=3
14:16:18.851 -> status=73 value=0
14:16:18.851 -> status=73 topic=Fan payload=1
14:16:23.661 -> status=73 value=3
14:16:23.661 -> status=73 topic=Fan payload=4
14:16:32.433 -> status=73 value=2
14:16:32.433 -> status=73 topic=Fan payload=3
14:16:35.746 -> status=75 value=148
14:16:35.746 -> status=75 topic=Troom payload=21.75
14:16:49.678 -> status=76 value=25
14:16:49.678 -> status=76 topic=Tsetpoint payload=25
14:16:53.687 -> status=72 value=16
14:16:53.687 -> status=72 topic=Mode payload=heat
14:17:00.702 -> status=72 value=4
14:17:00.702 -> status=72 topic=Mode payload=dry
14:17:09.662 -> status=72 value=12
14:17:09.662 -> status=72 topic=Mode payload=fan_only
14:17:17.704 -> status=72 value=0
14:17:17.704 -> status=72 topic=Mode payload=auto
14:17:17.704 -> status=76 value=26
14:17:17.704 -> status=76 topic=Tsetpoint payload=26

**Then trying to turn off the heat pump from Home Assistant (didn't turn off)**

14:17:30.110 -> MQTT_subscribe_callback, topic=MHInterface/set/Power payload=OFF payload_length=3
14:17:30.110 -> status=66 topic=cmd_received payload=o.k.

Any advise would be appreciated

absalom-muc, 2020-06-28:
Please test the latest version. Should work now.

Daryn-L, 2020-06-28:
Nope still not working for me.  Even with latest version (2.2)
It's like the values are not writing to the heat pump but the program thinks they are...

absalom-muc, 2020-06-29:
If cmd_received payload=o.k. then the program writes to the AC. If you see no reaction, the connection of the MOSI line may be broken.

Daryn-L, 2020-07-01:
Thanks for that.

Yes it was a bad connection on my JST connector. 

Is working now.

Thank you very much for this excellent work!  

The only thing not working for me is the swing controls.  I get back "?"

absalom-muc, 2020-07-01:
👍

closed 2020-07-01 by absalom-muc