absalom-muc/MHI-AC-Ctrl #26: First run problem

aleksandrdobrynin, 2020-06-30:
Greetings.

Thanks a lot for such a cool thing. But I can’t start something: /

I assembled the module, flashed it with the latest firmware, connected it to the air conditioner, in MQTT Explorer I see only that (see screen). What could be wrong?
[image]

aleksandrdobrynin, 2020-06-30:
It is also alarming that the pings are interrupted, maybe it doesn’t catch the wifi network well, but another device is nearby and works fine.
[image]

absalom-muc, 2020-06-30:
Hello,
1. Which AC model do you have? Please tell me the exact name e.g. SRK 35 ZS-S
2. Please provide an extract of the serial monitor output (115200Baud, Timestamp enabled)

aleksandrdobrynin, 2020-06-30:
1. SRK 25ZS-W
2. Please tell me how to do this? I do not quite understand (

aleksandrdobrynin, 2020-06-30:
Still, instead of the power that is implemented in the project, I set this with an output of 5v:
https://aliexpress.ru/item/4000035788248.html?spm=a2g0s.9042311.0.0.402133edyUWAbH

[image]

absalom-muc, 2020-06-30:
> 
> 
>     1. SRK 25ZS-W
> 
>     2. Please tell me how to do this? I do not quite understand (

if you use the Arduino IDE, select tools -> serial monitor. Since this is very basic, please use Google for further details how to use the serial monitor.

aleksandrdobrynin, 2020-06-30:
11:49:41.688 -> MHI-AC-Ctrl starting
11:49:41.688 -> CPU frequency[Hz]=80000000
11:49:41.721 -> Attempting WiFi connection .... connected to ilink2G, IP address: 192.168.252.215
11:49:42.207 -> OTA Ready
11:49:42.207 -> Attempting MQTT connection... connected
11:49:42.207 -> status=65 topic=connected payload=1
11:49:42.207 -> status=64 topic=RSSI payload=-58
11:49:42.241 -> Measure frequency for SCK, MOSI and MISO pin
11:49:43.238 -> status=68 value=0
11:49:43.238 -> status=68 topic=fSCK payload=0
11:49:43.238 -> status=69 value=0
11:49:43.238 -> status=69 topic=fMOSI payload=0
11:49:43.238 -> status=70 value=0
11:49:43.238 -> status=70 topic=fMISO payload=0
11:49:43.238 -> SCK frequency=0Hz (expected: >3000Hz) out of range!
11:49:43.238 -> MOSI frequency=0Hz (expected: <SCK frequency) out of range!
11:49:43.238 -> MISO frequency=0Hz (expected: ~0Hz) o.k.
11:49:44.229 -> mhi_ac_ctrl_core.loop error: -1
11:49:44.583 -> mhi_ac_ctrl_core.loop error: -1
11:49:44.931 -> mhi_ac_ctrl_core.loop error: -1
11:49:45.294 -> mhi_ac_ctrl_core.loop error: -1
11:49:45.621 -> mhi_ac_ctrl_core.loop error: -1
11:49:45.963 -> mhi_ac_ctrl_core.loop error: -1
11:49:46.317 -> mhi_ac_ctrl_core.loop error: -1
11:49:46.661 -> mhi_ac_ctrl_core.loop error: -1
11:49:47.026 -> mhi_ac_ctrl_core.loop error: -1
11:49:47.394 -> mhi_ac_ctrl_core.loop error: -1

absalom-muc, 2020-06-30:
You mixed MOSI and SCK pins. Swap the connections or adapt MHI-AC-Ctrl-core.h
old
```
#define SCK_PIN  14
#define MOSI_PIN 13
```
new
```
#define SCK_PIN  13
#define MOSI_PIN 14
```
And ensure that MISO is connected to Pin 12

aleksandrdobrynin, 2020-06-30:
[image]
The data ran! )))
Only does not display the position of the curtain. Is he just writing a sign? I tried to change the position from the remote control, the situation has not changed.

aleksandrdobrynin, 2020-06-30:
I tried to send Off or On in RAW format to the MHI-AC-Ctrl-LR / Power topic via MQTT Explorer, but the air conditioner does not react in any way.

absalom-muc, 2020-06-30:
> 
> 
> I tried to send Off or On in RAW format to the MHI-AC-Ctrl-LR / Power topic via MQTT Explorer, but the air conditioner does not react in any way.

Please provide an extract of the serial monitor output (115200Baud, Timestamp enabled).
Have you changed something in the program? I'm surprised that fSCK=0, it should be aprox. 3200.
The curtain topic is described in https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/SW-Configuration.md#mqtt-status

aleksandrdobrynin, 2020-06-30:
Changed only this:
old

#define SCK_PIN 14
#define MOSI_PIN 13
new

#define SCK_PIN 13
#define MOSI_PIN 14

And reloaded the firmware. Didn’t change anything anymore.

fSCK is ok
[image]

aleksandrdobrynin, 2020-07-01:
[image]

I worked for a couple of hours and fell off, does not respond to / set commands. fSCK shows 0

Also there was an open wifi network. Or should it be?

aleksandrdobrynin, 2020-07-01:
12:15:37.205 -> mhi_ac_ctrl_core.loop error: -2
12:15:37.312 -> mhi_ac_ctrl_core.loop error: -2
12:15:37.420 -> mhi_ac_ctrl_core.loop error: -2
12:15:37.520 -> mhi_ac_ctrl_core.loop error: -2
12:15:37.624 -> mhi_ac_ctrl_core.loop error: -2
12:15:37.727 -> mhi_ac_ctrl_core.loop error: -2
12:15:37.801 -> mhi_ac_ctrl_core.loop error: -2
12:15:37.902 -> mhi_ac_ctrl_core.loop error: -2
12:15:38.012 -> mhi_ac_ctrl_core.loop error: -2
12:15:38.118 -> mhi_ac_ctrl_core.loop error: -2

absalom-muc, 2020-07-01:
In your log are no MQTT accesses visible. Have you considered the desciption in https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/SW-Configuration.md#mqtt-status? Especially that the MQTT path for setting values is different from the path for reading values, e.g. MHI-AC-Ctrl/**set**/Power

I don't understand your statement:
>  fSCK shows 0

If fSCK=0 then the program would do nothing, and [here](https://github.com/absalom-muc/MHI-AC-Ctrl/issues/26#issuecomment-651960971) you show that fSCK=3200.

If you used previously the ESP8266 for an open WiFi then it could happen that the old configuration is stored. To prevent this, you could use once the option tools -> Erase Flash -> All Flash Contents when compiling/uploading the sketch in the Arduino IDE.

Your log shows many errors, but these are not critical and we take care later once the basic functionality works.

aleksandrdobrynin, 2020-07-01:
I download the firmware with the checkmark on clearing flash checked, but after downloading, the Wi-Fi point is visible anyway.
Tell me which version of arduino ide  to use and what settings to set with firmware?

aleksandrdobrynin, 2020-07-01:
Now there is such a log in the console (cyclic reboot).

18:15:23.787 -> >>>stack>>>
18:15:23.787 -> 
18:15:23.787 -> ctx: cont
18:15:23.787 -> sp: 3ffffdb0 end: 3fffffc0 offset: 01a0
18:15:23.787 -> 3fffff50:  3ffee54c 3ffee554 3ffee7b0 402011b1  
18:15:23.787 -> 3fffff60:  00000001 3ffe86a3 3ffee67c 3ffee8f8  
18:15:23.787 -> 3fffff70:  3fffdad0 00000000 3ffee558 402011d6  
18:15:23.825 -> 3fffff80:  3fffdad0 00000000 3ffee584 4020245c  
18:15:23.825 -> 3fffff90:  40201aa4 feefeffe 4020e8d0 4020e8b8  
18:15:23.825 -> 3fffffa0:  feefeffe feefeffe 3ffee8b8 4020cc70  
18:15:23.825 -> 3fffffb0:  feefeffe feefeffe 3ffe850c 40101145  
18:15:23.825 -> <<<stack<<<
18:15:23.825 -> 
18:15:23.825 ->  ets Jan  8 2013,rst cause:1, boot mode:(3,6)
18:15:23.825 -> 
18:15:23.825 -> load 0x4010f000, len 3456, room 16 
18:15:23.859 -> tail 0
18:15:23.859 -> chksum 0x84
18:15:23.859 -> csum 0x84
18:15:23.859 -> va5432625
18:15:23.859 -> ~ld
18:15:23.927 -> 
18:15:23.927 -> MHI-AC-Ctrl starting
18:15:23.927 -> CPU frequency[Hz]=80000000
18:15:23.927 -> Attempting WiFi connection .... connected to ilink2G, IP address: 192.168.252.215
18:15:24.428 -> OTA Ready
18:15:24.428 -> Attempting MQTT connection... connected
18:15:24.465 -> status=65 topic=connected payload=1
18:15:24.465 -> status=64 topic=RSSI payload=-64
18:15:24.465 -> Measure frequency for SCK, MOSI and MISO pin
18:15:25.458 -> status=68 value=3200
18:15:25.458 -> status=68 topic=fSCK payload=3200
18:15:25.458 -> status=69 value=500
18:15:25.458 -> status=69 topic=fMOSI payload=500
18:15:25.458 -> status=70 value=3200
18:15:25.458 -> status=70 topic=fMISO payload=3200
18:15:25.458 -> SCK frequency=3200Hz (expected: >3000Hz) o.k.
18:15:25.458 -> MOSI frequency=500Hz (expected: <SCK frequency) o.k.
18:15:25.458 -> MISO frequency=3200Hz (expected: ~0Hz) out of range!
18:15:27.305 -> 
18:15:27.305 -> Soft WDT reset

absalom-muc, 2020-07-01:
Have you touched the board connections? According your log
```
18:18:50.143 -> SCK frequency=3200Hz (expected: >3000Hz) o.k.
18:18:50.143 -> MOSI frequency=500Hz (expected: <SCK frequency) o.k.
18:18:50.180 -> MISO frequency=3200Hz (expected: ~0Hz) out of range!
```
there is a signal on MISO. Seems that you shorted SCK and MISO.

aleksandrdobrynin, 2020-07-02:
Yes, apparently something was not soldered so. But the strangest thing was that it worked for a while. I assembled a new circuit and it seems like everything works.

I hope the last question, is it possible to somehow get the status of an air conditioner in Node-RED? Some kind of request, something like get nodes?

coyote1, 2020-07-02:
> If you used previously the ESP8266 for an open WiFi then it could happen that the old configuration is stored. To prevent this, you could use once the option tools -> Erase Flash -> All Flash Contents when compiling/uploading the sketch in the Arduino IDE.

For me it is also the case that all 3 ESPs show an open network, even though I have selected "flash all contents".

absalom-muc, 2020-07-02:
According to the [ESP8266WiFi documentation](https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/station-class.html#start-here),
[line 9 in support.cpp](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/2a87076fd7c401a906a94f64d7c20b699d0efc02/src/support.cpp#L9)
`WiFi.begin(WIFI_SSID, WIFI_PASSWORD);`
is used to switch to the Station mode, so module can connect to specific Access Point. An AP related command is not used.
Since I can't see an open network I suggest that one of you tries to find out the root cause.

absalom-muc, 2020-07-03:
I rechecked this topic and finally the solution is what coyote1 already proposed. Please insert the lines
```
  WiFi.persistent(false);
  WiFi.mode(WIFI_STA);
```
to the begin of void setupWiFi() in support.cpp and give feedback if it works for you. That means there should be no longer an open WiFi visible.

Background (from my understanding):
A fresh ESP8266 comes with enabled STA **and** AP mode (WIFI_AP_STA). The line
`WiFi.begin(WIFI_SSID, WIFI_PASSWORD);`
does not switch to the STA mode, this line provides only the parameters for the STA mode, still STA and AP mode are enabled. The [ESP8266WiFi documentation](https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/station-class.html#start-here) seems to be wrong or at least too fuzzy. 
You must explicitely write 
```
  WiFi.mode(WIFI_STA);
```
to enable the STA mode only.
When using the option tools -> Erase Flash -> All Flash Contents, then ESP8266 uses a default and this default is using the AP mode (WIFI_AP). I know it is confusing ...

Addtionally you find above the line
```
  WiFi.persistent(false);
```
because with every call of 
`WiFi.begin(WIFI_SSID, WIFI_PASSWORD);`
these settings are written to specific sectors of flash memory. Frequently calling these functions could cause wear on the flash memory.
```
  WiFi.persistent(false);
```
prevents the writing to the flash memory. This behaviour is described [here](https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/generic-class.html#persistent).

aleksandrdobrynin, 2020-07-03:
Tell the correct syntax:

void setupWiFi (WiFi.persistent (false), WiFi.mode (WIFI_STA));

So I get a compilation error.

absalom-muc, 2020-07-03:
Please copy it from my post

aleksandrdobrynin, 2020-07-03:
Sorry, I made changes to the wrong file.

Forgiven, the access point has disappeared. But the connection disappeared again. Log:

01:28:45.630 -> >>>stack>>>
01:28:45.664 -> 
01:28:45.664 -> ctx: cont
01:28:45.664 -> sp: 3ffffdb0 end: 3fffffc0 offset: 01a0
01:28:45.664 -> 3fffff50:  3ffee54c 3ffee554 3ffee7b0 402011b1  
01:28:45.664 -> 3fffff60:  00000001 3ffe86a3 3ffee67c 3ffee8f8  
01:28:45.664 -> 3fffff70:  3fffdad0 00000000 3ffee558 402011d6  
01:28:45.664 -> 3fffff80:  3fffdad0 00000000 3ffee584 40202440  
01:28:45.664 -> 3fffff90:  40201a9c feefeffe 4020e8cc 4020e8b4  
01:28:45.664 -> 3fffffa0:  feefeffe feefeffe 3ffee8b8 4020cc6c  
01:28:45.700 -> 3fffffb0:  feefeffe feefeffe 3ffe850c 40101145  
01:28:45.700 -> <<<stack<<<
01:28:45.700 -> 
01:28:45.700 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,7)
01:28:45.700 -> 
01:28:45.700 -> load 0x4010f000, len 3456, room 16 
01:28:45.700 -> tail 0
01:28:45.700 -> chksum 0x84
01:28:45.700 -> csum 0x84
01:28:45.700 -> va5432625
01:28:45.700 -> ~ld
01:28:45.775 -> 
01:28:45.775 -> MHI-AC-Ctrl starting
01:28:45.775 -> CPU frequency[Hz]=80000000
01:28:45.775 -> Attempting WiFi connection .......... connected to ilink2G, IP address: 192.168.252.215
01:28:50.020 -> OTA Ready
01:28:50.020 -> Attempting MQTT connection... connected
01:28:50.020 -> status=65 topic=connected payload=1
01:28:50.020 -> status=64 topic=RSSI payload=-63
01:28:50.020 -> Measure frequency for SCK, MOSI and MISO pin
01:28:51.009 -> status=68 value=3200
01:28:51.009 -> status=68 topic=fSCK payload=3200
01:28:51.009 -> status=69 value=500
01:28:51.043 -> status=69 topic=fMOSI payload=500
01:28:51.043 -> status=70 value=500
01:28:51.043 -> status=70 topic=fMISO payload=500
01:28:51.043 -> SCK frequency=3200Hz (expected: >3000Hz) o.k.
01:28:51.043 -> MOSI frequency=500Hz (expected: <SCK frequency) o.k.
01:28:51.043 -> MISO frequency=500Hz (expected: ~0Hz) out of range!
01:28:52.165 -> 
01:28:52.165 -> Soft WDT reset

absalom-muc, 2020-07-04:
```
01:28:51.043 -> SCK frequency=3200Hz (expected: >3000Hz) o.k.
01:28:51.043 -> MOSI frequency=500Hz (expected: <SCK frequency) o.k.
01:28:51.043 -> MISO frequency=500Hz (expected: ~0Hz) out of range!
```
This time it is a short between MOSI and MISO.

coyote1, 2020-07-05:
Flashed all 3 ESPs with Changes lines, works fine now. No AP anymore

aleksandrdobrynin, 2020-07-06:
Yes, I did it too! WiFi points are no more. Thank.

And how can I make sure that the air conditioner sends its basic settings (operating mode, temperature, fan speed, shutters) every 10 seconds? I add air conditioning to Apple HomeKit through Node-RED and without receiving changes, it may turn out that the air conditioner works in one mode, and the Apple application will display another.

And yet, I have 5 fan speed modes - auto, 1,2, 3, 4. When I set auto mode, it says that this is the third speed, not auto. Can I somehow add support for auto mode?

absalom-muc, 2020-07-08:
> Yes, I did it too! WiFi points are no more. Thank.

Thanks to you and @coyote1 for your feedback

> And how can I make sure that the air conditioner sends its basic settings (operating mode, temperature, fan speed, shutters) every 10 seconds? I add air conditioning to Apple HomeKit through [...]

Better rethink your approach. An update of the AC status is visible on your MQTT broker. This event driven approach is usually the best approach because it is ensured that the latest status is always available. There should be no reason for regulary update the status when there is no change. If you still want to have the status resend w/o update you have to adapt the program.
 
> And yet, I have 5 fan speed modes - auto, 1, 2, 3, 4. When I set auto mode, it says that this is the third speed, not auto. Can I somehow add support for auto mode?

Please check section "Known Limitations" in the [Troubleshooting guide](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/Troubleshooting.md#known-limitations)

aleksandrdobrynin, 2020-07-09:
Thank you very much for the help!!!
Added two air conditioners to HomeKIT thanks to your efforts.

closed 2020-07-09 by aleksandrdobrynin

L4RGER, 2020-11-16:
> You mixed MOSI and SCK pins. Swap the connections or adapt MHI-AC-Ctrl-core.h
> old
> 
> ```
> #define SCK_PIN  14
> #define MOSI_PIN 13
> ```
> 
> new
> 
> ```
> #define SCK_PIN  13
> #define MOSI_PIN 14
> ```
> 
> And ensure that MISO is connected to Pin 12

In 2.03 it’s still showing 
#define SCK_PIN  14
#define MOSI_PIN 13

Do I need to swap this?

absalom-muc, 2020-11-17:
No, if you connect it according to [Hardware.md](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/Hardware.md)
```
#define SCK_PIN  14
#define MOSI_PIN 13
```
is correct.