absalom-muc/MHI-AC-Ctrl #27: Vanes not Working

Daryn-L, 2020-07-01:
On my unit, I have L/R, U/D and 3D Auto as Vanes options.

My MQTT response for the vanes from this program is:
```

Message 323 received on MHInterface/Vanes at 14:56:

?

QoS: 0 - Retain: true 
```

Do you have any advice on how to solve this?

absalom-muc, 2020-07-01:
Is the behaviour different from the description [here](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/SW-Configuration.md#mqtt-status) or [here](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/SPI.md#vanes)?

`When the last command was received via the infrared remote control then the Vanes status is unknown and the "?" is published.`

Daryn-L, 2020-07-01:
Actually it is working.  I had a misunderstanding on what they meant and needing to change the swing command to lower case.

However, is there any way to get the left/right working and also the 3D auto?  (up/down + left/right)

absalom-muc, 2020-07-02:
> 
> 
> Actually it is working. I had a misunderstanding on what they meant and needing to change the swing command to lower case.
> 
> However, is there any way to get the left/right working and also the 3D auto? (up/down + left/right)

I don't know a way and since also commercial wired remote controls (e.g. RC-E5) doesn't support it I assume it is not possible

absalom-muc, 2020-07-03:
I believe the question is answered, so I close this topic, but feel free to re-open it, if something is unclear.

closed 2020-07-03 by absalom-muc