absalom-muc/MHI-AC-Ctrl #28: Veroboard design using ESP8266 NodeMCU

ThomasHFWright, 2020-07-01:
If anyone is interested in an alternative design not requiring a PCB, I remapped the existing circuit diagram to Veroboard. I also swapped out to a NodeMCU module as they were cheaper for me. Note I swapped the MISO/MOSI pins directly on the board, so there isn't any crossover required in the wiring.

I used some software called [VeroRoute ](https://sourceforge.net/projects/veroroute/) to create these designs

[aircon-veroroute-design.zip](https://github.com/absalom-muc/MHI-AC-Ctrl/files/4858392/aircon-veroroute-design.zip)
[Veroroute cutout.pdf](https://github.com/absalom-muc/MHI-AC-Ctrl/files/4858388/Veroroute.cutout.pdf)
[Veroroute top.pdf](https://github.com/absalom-muc/MHI-AC-Ctrl/files/4858389/Veroroute.top.pdf)

absalom-muc, 2020-07-02:
Thank you very much for sharing. It will definitely help users with their first steps! :thumbsup:

(only one hint: I would not recommend to locate the PCB near the power supply connectors without a proper isolation)

ThomasHFWright, 2020-07-03:
I do have the whole circuit shrink wrapped in clear plastic, would you say that is not enough isolation? (I'm quite a newbie to electronics in general)

absalom-muc, 2020-07-03:
From "electrical safety" point of view it is not enough. I assume the isolation voltage is not defined for the plastic. Or if you have an unexpected heating of the ESP8266 board, the plastic could meld. Or if you connect the USB to the board for testing, you are near the power supply connectors, you could accidentally touch the connectors. Therefore I recommend to locate the PCB outside of the AC housing.

Rescla, 2021-01-17:
Thanks for sharing! I've never heard of VeroRoute before. And although I wouldn't call it intuitive, the tutorial (Help -> Tutorial) is pretty helpful. I used a slightly different layout (since I couldn't easily source a TSR 1-2450).

This design uses a LOLIN D1 Mini for the ESP8266, a Polulu 2595 for the 5v supply and a Polulu 2843 for the level shifter.  I used a 5cm*7cm perfboard with mostly solder traces and a couple of wires for the longer 5V and GND traces. The board is located in the AC housing, wrapped in 400V isolation tape and fixed in place using the same tape.

[image]

As a first functional electronics project it was pretty fun to do. The hardest part was actually getting to the connector. I have the SRK35ZS-W, which requires you to take off the top and bottom panel to reach the low voltage electronics. 
[instructions.pdf](https://github.com/absalom-muc/MHI-AC-Ctrl/files/5826417/instructions.pdf)

For additional fun, my unit has been installed with the refrigerant lines running through a hole in the bottom panel, which means I can't actually remove it. It's still possible to get the top panel off and then temporarily screw the bottom panel back to the unit, so you can work without the bottom panel dangling, but it's still a bit annoying.

On the software side, I had to change some config in support.h and MHI-AC-Ctrl.h to suit Home Assistant. Namely changing the MQTT payloads to lowercase,  TEMP_MEASURE_PERIOD to 0 and POWERON_WHEN_CHANGING_MODE to true, in addition to the usual configuration.

[aricon-veroroute-design-alt.zip](https://github.com/absalom-muc/MHI-AC-Ctrl/files/5826408/aricon-veroroute-design-alt.zip)

SpatialJ, 2021-06-28:
@Rescla could you post some images of your design? Maybe you took some before closing the unit :-)

Rescla, 2021-06-28:
> 
> 
> @Rescla could you post some images of your design? Maybe you took some before closing the unit :-)

I hadn't, of course. I was just happy to get it working :P But I will take some pictures when I have the chance.

Rescla, 2021-06-28:
@SpatialJ Does this help?
[image]
[image]
[image]

SpatialJ, 2021-06-28:
Yes, thank you very much!

Siumba, 2023-01-22:
I've created a PCB a while ago, attached are the gerber files.
[Gerber_v1.5 - COPPER_2023-01-22.zip](https://github.com/absalom-muc/MHI-AC-Ctrl/files/10474234/Gerber_v1.5.-.COPPER_2023-01-22.zip)

glsf91, 2023-10-02:
Closed because of no activity for a long time

closed 2023-10-02 by glsf91