absalom-muc/MHI-AC-Ctrl #29: Home Assistant Enhancement Idea

Daryn-L, 2020-07-02:
I use home assistant for this.

My HA config is:

`climate heatpump:
  - platform: mqtt
    unique_id: mhi0001a
    name: HeatPump
    initial: 22
    min_temp: 18
    max_temp: 30
    modes:
      - "auto"
      - "dry"
      - "cool"
      - "fan_only"
      - "heat"
      - "off"
    swing_modes:
      - "1"
      - "2"
      - "3"
      - "4"
      - "Swing"
    fan_modes:
      - "1"
      - "2"
      - "3"
      - "4"
    mode_command_topic: "MHI-AC-Ctrl/set/Mode"
    temperature_command_topic: "MHI-AC-Ctrl/set/Tsetpoint"
    fan_mode_command_topic: "MHI-AC-Ctrl/set/Fan"
    swing_mode_command_topic: "MHI-AC-Ctrl/set/Vanes"
    current_temperature_topic: "MHI-AC-Ctrl/Troom"
    mode_state_topic: "MHI-AC-Ctrl/Mode"
    temperature_state_topic: "MHI-AC-Ctrl/Tsetpoint"
    fan_mode_state_topic: "MHI-AC-Ctrl/Fan"
    swing_mode_state_topic: "MHI-AC-Ctrl/Vanes"
`

Home assistant will only accept certain topic commands which can't be changed easily e.g. "MHI-AC-Ctrl/set/Mode" -> "fan_only"

My suggestion is to have something like follows:

`
//Platform
#define USE_HOME_ASSISTANT true

#define PAYLOAD_MODE_AUTO HOME_ASSISTANT == true  ? "auto" : "Auto"
#define PAYLOAD_MODE_STOP HOME_ASSISTANT == true  ? "stop" : "Stop"
#define PAYLOAD_MODE_DRY HOME_ASSISTANT == true  ? "dry" : "Dry"
#define PAYLOAD_MODE_COOL HOME_ASSISTANT == true  ? "cool" : "Cool"
#define PAYLOAD_MODE_FAN HOME_ASSISTANT == true  ? "fan_only" : "Fan"
#define PAYLOAD_MODE_HEAT HOME_ASSISTANT == true  ? "heat" : "Heat"
#define PAYLOAD_OP_DEFROST_ON HOME_ASSISTANT == true  ? "on" : "On"
#define PAYLOAD_OP_DEFROST_OFF HOME_ASSISTANT == true  ? "off" : "Off"
`

The only problem I am getting is in the line
`
#define PAYLOAD_MODE_FAN HOME_ASSISTANT == true  ? "fan_only" : "Fan"
`

Not sure what it is as I do not know C.  If it was in Java I could probably figure it out.

Anyways, I think it would be a nice addition to have this work 'out of the box' with HA.

absalom-muc, 2020-07-02:
I like the approach and I absolute agree that an 'out of the box' with HA would be beneficial for the user.

But I planned this repository to provide only a generic solution for all platforms. For specific platforms e.g. HA, I suggest that you open a new github repository where you describe the customizations for HA and where you store files specific for HA (e.g. the configuration file and the specific MHI-AC-Ctrl.h). Of course I could then reference the new repository here.

That would give you the freedom to adapt it according your needs and reduces the complexity of the generic solution. I prefer not to have specific platforms (which I don't use) supported by this repository.
What do you think?

apateluk, 2020-07-02:
> That would give you the freedom to adapt it according your needs and reduces the complexity of the generic solution. I prefer not to have specific platforms (which I don't use) supported by this repository.
> What do you think?

I agree with your approach of a more generic soloution as there a number of platforms that need to be considered in addition to HA. Also i've made other modifications that integrate WiFi manager for easier WiFi set-up.

```
The only problem I am getting is in the line
#define PAYLOAD_MODE_FAN HOME_ASSISTANT == true ? "fan_only" : "Fan"
```
Do the other equivalent lines like this work ? I didn't know you could use conditions in C defines like this. The only way i would have thought is to use:

//Platform
#define USE_HOME_ASSISTANT

#ifdef USE_HOME_ASSISTANT
#define PAYLOAD_MODE_AUTO  "auto"
#else
#define PAYLOAD_MODE_AUTO  "Auto"
#endif

absalom-muc, 2020-07-02:
I believe 
```
#define USE_HOME_ASSISTANT

#ifdef USE_HOME_ASSISTANT
#define PAYLOAD_MODE_AUTO  "auto"
#else
#define PAYLOAD_MODE_AUTO  "Auto"
#endif
```
is the right way.

But when you are writing a specific version for HA now, why do you still want to differentiate generic vs. HA?

apateluk, 2020-07-02:
I agree with you. A generic version is preferable as it keeps it clean and easy for others to adapt for other platforms if needed.

Daryn-L, 2020-07-03:
I see the point on a generic version and agree with that now.  A better option, if possible, would be a webserver where you could select the SSID and password and set the topic commands. If anyone had the time...

apateluk, 2020-07-03:
You can use WiFiManager to do this. Infact, i'm doing exactly that for the Wifi configuration part of this as that's the only thing that may possible change in my usage. The topics etc are pretty stable when it comes to HA, so once modified in code never expect to change it.

absalom-muc, 2020-07-03:
@b0rder made already an [integration of MHI-AC-Ctrl in Tasmota](https://github.com/absalom-muc/MHI-AC-Ctrl/issues/13), seems to be similar what you plan.

ginkage, 2021-04-19:
One way to do this is to use ESPHome and implement integration for it.
I took a stab at it here: https://github.com/ginkage/MHI-AC-Ctrl-ESPHome (note that MHI-AC-Ctrl-core.* files are directly forked, the rest is done from scratch, and your network config is in *.yaml).

deku-m, 2021-04-29:
> One way to do this is to use ESPHome and implement integration for it.
> I took a stab at it here: https://github.com/ginkage/MHI-AC-Ctrl-ESPHome (note that MHI-AC-Ctrl-core.* files are directly forked, the rest is done from scratch, and your network config is in *.yaml).
> This way you don't need MQTT at all, you get auto-discovery and even OTA for free.

Use it in my home assistant works perfectly THX

absalom-muc, 2022-06-16:
closed since there was no update for a long time. Please feel free to re-open in case of follow-up topics.

closed 2022-06-16 by absalom-muc