absalom-muc/MHI-AC-Ctrl #31: Jumper

hvorragend, 2020-07-07:
Could you please answer me for what purpose are the jumpers on the PCB?

Thank you very much

absalom-muc, 2020-07-08:
There is one jumper JP1. It connects the voltage output of the DC/DC converter to the voltage input of the ESP8266 board. Originally I planned it because I feared problems if the ESP8266 is supplied simultaneously via the DC/DC converter and during test / debug via USB. But I have never seen a problem with it.
You can replace the jumper by a piece of wire.

hvorragend, 2020-07-08:
What do you think? Should't this be located in the Wiki?

absalom-muc, 2020-07-09:
Yes, I added a comment to [Hardware.md](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/Hardware.md)

closed 2020-07-27 by hvorragend