absalom-muc/MHI-AC-Ctrl #33: Run exception on Wemos D1 Mini

coen800, 2020-07-20:
First of all thank you for your hard work on this project. Saves me a lot of headaches ;) so thanks for that.

While trying this out i noticed the d1 mini was in a boot loop. After hooking it up to my serial monitor I saw the following output running in a loop:

`MHI-AC-Ctrl starting
CPU frequency[Hz]=80000000
Attempting WiFi connection .......... connected to xxxxxx, IP address: 192.168.xxx.xxx
OTA Ready
Attempting MQTT connection... connected
status=65 topic=connected payload=1

Exception (3):
epc1=0x40216c6c epc2=0x00000000 epc3=0x00000000 excvaddr=0x40247ed4 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd20 end: 3fffffc0 offset: 0190
3ffffeb0:  40247ee2 40247f30 3ffffe90 3ffffec0  
3ffffec0:  40247ed4 00000000 401004d2 3ffee890  
3ffffed0:  00000041 40247ed8 40247ed4 40202814  
3ffffee0:  2d49484d 432d4341 2f6c7274 6e6e6f63  
3ffffef0:  65746365 00000064 00000000 00000000  
3fffff00:  00000000 00000000 00000000 00000000  
3fffff10:  00000000 00000000 00000000 00000000  
3fffff20:  00000000 00000000 00000000 00000000  
3fffff30:  00000000 00000000 00000000 00000000  
3fffff40:  00000000 00000000 3ffee748 4020b5fc  
3fffff50:  3ffee594 00000000 3ffee748 402028ac  
3fffff60:  00000001 3ffe86c2 3ffee614 40202a84  
3fffff70:  3ffefbf4 00000000 3ffee594 40204fde  
3fffff80:  3fffdad0 00000000 3ffee594 40202459  
3fffff90:  40201a9c feefeffe 4020e9c4 4020e9ac  
3fffffa0:  feefeffe feefeffe 3ffee850 4020cd64  
3fffffb0:  feefeffe feefeffe 3ffe8510 40101145  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3456, room 16 
tail 0
chksum 0x84
csum 0x84
va5432625
~ld`

Unfortunately my knowledge stops here and after a lot of googling I wasn't able to resolve it myself.

absalom-muc, 2020-07-20:
Have you already checked the [troubleshooting guide](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/Troubleshooting.md), especially related to the [PubSub Client version](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/Troubleshooting.md#fire-you-use-pubsubclient-v280)?

coen800, 2020-07-21:
I did not, didn't even see it. Will try it out, thanks.

coen800, 2020-07-21:
Can confirm it works now. Thank you.

closed 2020-07-21 by coen800

absalom-muc, 2020-07-21:
Great, happy to hear that :thumbsup: