absalom-muc/MHI-AC-Ctrl #34: Unable to compile

thomas41587, 2020-07-24:
Hello,
I am trying to compile the project. I already added the reqired libraries but I get a bunch of error with the pubsub library. I checked the library itself and was able to compile the built in examples. So I guess the library itself should be OK.
Here is the output of the Arduino IDE: https://pastebin.com/0Q3QUWcs

absalom-muc, 2020-07-25:
Hello Thomas,
I just compiled the [recently released version](https://github.com/absalom-muc/MHI-AC-Ctrl/releases/tag/v2.03) without problems. You could compare the versions of the used libraries:
[image]

You find in the web many posts related to ESP8266 compile errors. It seems to be not related to MHI-AC-Ctrl.

thomas41587, 2020-07-29:
Thank you very much for the fast reply. I played a little bit around with the Arduino IDE but always ended up with the same error. Therefore I tried on a different VM-Image where nothing was install. Did the same steps there and...it compiled without any errors :-)
I have no idea what exactly caused these compile error. looks like something with the system itself seemed to be strange.
But now everything is working perfectly! Thank you very much for this awesome work!!

closed 2020-07-29 by thomas41587