absalom-muc/MHI-AC-Ctrl #37: How to configure / update OTA?

manoloxxl, 2020-07-29:
Hi, 

I've seen there is a OTA option. But i didn't find any advice how to handle it. By using IP or host-name i cannot access the device.

Is there a way to change the MQTT server adress or run an update without unmount the AC / remove the device from the AC?

Any hint?

Best Regards

Manuel

thomas41587, 2020-07-29:
Hello Manuel,
never tried (or even thought) about OTA update. But the appropriate option is available in the editor: 
[image]

absalom-muc, 2020-07-29:
In the web you find many examples how to use OTA. And maybe this [hint](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/Troubleshooting.md#fire-cant-see-the-device-in-in-arduino-ide-tools---port-for-ota) in the Troubleshooting Guide could help.

manoloxxl, 2020-07-30:
My Arduino IDE is not showing any network-device.

As this is an issue on my system, i close this one. Thanks!

closed 2020-07-30 by manoloxxl