absalom-muc/MHI-AC-Ctrl #39: Wiring btw AC and MHI-AC-Ctrl

drdoener, 2020-08-05:
Hi absalom,

thanks a lot for your great work. I´m about to solder the board and I´ve a question and input concerning the layout / wiring.
On the orange placards (/images/SRK-PCB.jpg and /images/SRK-PCB-RC.jpg) the order is GND,MISO,MOSI,SCL,+12V, beginning from 1.
This would mean 1 equals to GND.
On the pic below (AC_to_MHI-AC-Ctrl.png), PIN 1 equals to +12 V.
Please, could you explain the relationship between the pins on the boards and the information on the orange placards?

Thanks a lot,
Best,
drdoner

absalom-muc, 2020-08-05:
Thank you @drdoener - you found a bug in AC_to_MHI-AC-Ctrl.png. There is the numbering of the AC connector wrong, it should be
[image]
But the photos (incl. the orange placards in [SRK-PCB.jpg](../tree/master/images/SRK-PCB.jpg) and [SRK-PCB-RC.jpg](../tree/master/images/SRK-PCB-RC.jpg)) and the PCB show still the correct signal names.
I will update AC_to_MHI-AC-Ctrl.png accordingly.

absalom-muc, 2020-08-09:
corrected [here](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/images/AC_to_MHI-AC-Ctrl.png)

closed 2020-08-09 by absalom-muc

drdoener, 2020-08-15:
Hi,
fine, looks better now.
As soon as I will have installed the board, you receive feedback for my SRK 35 ZSX-W.

Greetz