absalom-muc/MHI-AC-Ctrl #40: Dht22 temp sensor

gamblor999, 2020-08-07:
Hi absolom, I have been using your code on my Mitsubishi’s for a while now (Thank you heaps) and noticed you have changed the code and added a temp sensor which is great and has been a pain point for me as the Mitsubishi’s don’t report the indoor temp unless active. I was wondering if any users of your code have modified it to use a DHT22 temp sensor and if so could I get access to the code.

Much appreciated

absalom-muc, 2020-08-07:
Hello gamblor9999,
I have no program for the DHT22, but I'm surprised that you say "has been a pain point for me as the Mitsubishi’s don’t report the indoor temp unless active". Usually the room temperature is also reported during stand-by, so maybe re-check it.

absalom-muc, 2021-09-18:
I believe your question is answered, so I close this topic. But feel free to re-open it in case of further questions.

(in general please close an issue once it is answered - thank you)

closed 2021-09-18 by absalom-muc