absalom-muc/MHI-AC-Ctrl #41: AC is set to "Off" after Wifi Connection was lost

manoloxxl, 2020-08-09:
Hi,

I'm running into an issue where the AC is powered off when the wifi network was down.

How to reproduce:

> Connect the module with wifi
> Power AC an by IR Control
> Power Off WiFi
> Power On Wifi

After the module reconnects, the AC is set to "Off" and stops working.

Im not pretty sure if this is a behavior in combination with ioBroker / MQTT Adapter as i don't have any other setup available. Does anybody else having this issue?

Best Regards
Manuel

absalom-muc, 2020-08-09:
I believe this behaviour comes from ioBroker or your MQTT adapter. You could use [MQTT Explorer](http://mqtt-explorer.com/) - a great all-round MQTT client that provides a structured topic overview - for your investigation.

manoloxxl, 2020-08-17:
Found the issue. The MQTT adapter sends all states on connect / subscribe. 

So when the AC is stopped by **not** the same device as it was startet, the "set" command contains the last submitted value that is different to the recent state.

I disabled the MQTT Adapter "send on start / subcribe" settings and now the AC is working as expected.

closed 2020-08-17 by manoloxxl