absalom-muc/MHI-AC-Ctrl #48: MHI-AC-Ctrl in a multi-split configuration

pdufrenne, 2020-10-13:
Hi,
Great work you have done. I appreciate your effort for this project.

Question though: Has this solution even been deployed in a multi-split system (one outside unit with multiple inside convertors).

One convertor works fine. The moment I connect the second one, MQTT reports AC "connected = 0" and the MQTT broker is flooded with this connect message. Sometime "1" but most of the time "0".

Which part of the code is responsible for retransmitting the "connected" status?

Thanks,

Peter

absalom-muc, 2020-10-13:
Hi Peter,
I believe you are the first one using MHI-AC-Ctrl with a Multi-Split-System.
In this [code line](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/080b90289c25d90ea8d801cd867b21753bf4342a/src/support.cpp#L46) it is checked if there is still a MQTT connection and - if needed - a reconnect is triggered. But I believe this information will not help you.

I assume you have one MHI-AC-Ctrl PCB per indoor unit, if this is correct have you used different [HOSTNAMES](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/080b90289c25d90ea8d801cd867b21753bf4342a/src/support.h#L8) for the different MHI-AC-Ctrl PCBs? If not then you have competing devices trying to establish a connection to the MQTT broker and I could comprehend the behaviour you describe.
Good luck!
absalom

pdufrenne, 2020-10-16:
Hi Absalom,

It works now.
Giving a unique HOSTNAME to every PCB did the trick.

[image]

The third one still needs to be installed so that one is not connected yet.

Thanks for the help and response.

Have a nice weekend,
Peter

absalom-muc, 2020-10-17:
Hi Peter, Thanks for your feedback - glad to hear that it works 👍

closed 2020-10-17 by absalom-muc