absalom-muc/MHI-AC-Ctrl #49: Timer and Night Setback

scottm1985, 2020-11-06:
More a question than an issue:
1. It looks as though having anything connected to a MHI CNS port will always disable the built-in timer functions.  The timer light will not even turn on with the MHI-AC-Ctrl connected.
2. It doesn't look as though it's possible to enable Night Setback mode through the MHI-AC-Ctrl.  I can see that when I enable Night Setback through the remote then the unit has a Tsetpoint of 10 but if I try to pass 10 to set/Tsetpoint I get the Invalid Parameter for cmd_received.  Has anybody figured out how to turn on Night Setback as that would mean I no longer need the timer function mentioned above?

absalom-muc, 2020-11-06:
When entering Night Setback then Tsetpoint = 10°C **and** the operating data SET-TEMP=10°C. It is usual that Tsetpoint has the same value as SET-TEMP. On the SPI signals no specific "Night Setback" configuration is visible.
You could try to adapt row 74 [here](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/4c11dc6c944c48b910df78fe77abb345e2407e5d/src/MHI-AC-Ctrl.ino#L74) from
`if ((atoi((char*)payload) >= 18) & (atoi((char*)payload) <= 30)) {`
to
`if ((atoi((char*)payload) >= 10) & (atoi((char*)payload) <= 30)) {`
Then you can select a temperature >=10°C, e.g. 10°C
But a set value of 10°C doesn't change SET-TEMP, only Tsetpoint. Therefore I'm not sure if it works.
You could try it. The room where my AC is installed becomes never <14°C therefore I can't test it.

scottm1985, 2020-11-07:
Thanks, I've made that change and it does allow me to select 10° but I'll have to wait and see if it gets down to that temperature and stops.  I see what you mean about the operating data not changing to 10° though.
Can you help me understand what the operating data is for?  I've noticed that it's always similar but not quite the same.  Are these some sort of internal variables that we can only read?

absalom-muc, 2020-11-07:
Hi Scott,

I don't know the exact background of the operating data. I assume they will be mainly used for service / maintenance. And unfortunatelly I couldn't find a detailed description from the manufacter related to the usage and decoding. I never found out if some of the operating data might be writeable.

Additionally not all settings of the infrared remote control are supported by the SPI (e.g. Vanes left/right or 3D) or at least I don't know how to interpret. I compared it also with some other commercial SPI based remote controls. They all have limited settings.

absalom

scottm1985, 2020-11-09:
Hi

I tested this last night and with it set to 10° it behaves as if it is set to 18°.  OpData/Tsetpoint went down to 17 but I guess that's because it was on eco mode.  I'm going to try it again with eco mode off and see if that makes a difference.  Otherwise, I'll see if I can get it to update the Opdata.

Thanks
Scott

absalom-muc, 2021-04-30:
Scott,
if your question is answered or if you plan no follow-up, please close this issue.
Thank you

closed 2021-05-01 by scottm1985