absalom-muc/MHI-AC-Ctrl #51: Hardware buy

vergoossen, 2020-12-10:
Hi,

Is there a place where to buy ready to install hardware?

serrj-sv, 2021-01-27:
I bought more components that I need (for spare) so, if you are in EU, I can assemble and ship for 25EUR + shipping costs. 
This is how it looks like, I can assemble 2 more:
[image]
[image]
[image]
[image]

vergoossen, 2021-01-27:
I want to buy both please.
Can you send me an e-mail to stef [at] [myusername] .nl please?

hellvideorazor, 2021-02-06:
I would be very happy to get one. How could we arrange?

> I bought more components that I need (for spare) so, if you are in EU, I can assemble and ship for 25EUR + shipping costs.
> This is how it looks like, I can assemble 2 more:
> [image]
> [image]
> [image]
> [image]
> [image]

serrj-sv, 2021-02-07:
I already assembled and sold both adapters to Stef. I have to order
components then, it may take 2-3 weeks to arrive. If you're willing to wait
I can proceed with ordering but I will need a pre-payment first.

hellvideorazor, 2021-02-07:
That should be fine as my skills might be a bit lacking. please drop me a
mail to adiehr at gmail dot com

serrj-sv, 2021-02-11:
> That should be fine as my skills might be a bit lacking.

Ok, I already placed an order for new components since I received another request. But due to Chinese new year there may be a delay. I will let you know once components arrived and device assembled

th3sniper86, 2021-02-18:
@serrj-sv if you have some more spare time and spare parts. I'd also be happy to receive two ;)

serrj-sv, 2021-02-18:
yes,I ordered more parts. I put you on the list ))

deku-m, 2021-02-25:
@serrj-sv 
Do you still have 2 more? 
Also a question did you try tasmota software aswel?

drdoener, 2021-02-27:
Hi all,
could anybody tell me which tye of connector has to be used to plug it on the board of the MHI-AC?
There are so many types of JST connectors, I already bought a wrong one.
Thanks !
Michael

serrj-sv, 2021-02-27:
I use this one (choose 5P version)
https://a.aliexpress.com/_uJNE1f

drdoener, 2021-02-28:
Thanks.
I ordered this one/this type (on another shop within EU).

Greets

absalom-muc, 2021-03-07:
> 
> 
> @serrj-sv
> Also a question did you try tasmota software aswel?

You could have a look on this [issue](https://github.com/absalom-muc/MHI-AC-Ctrl/issues/13)

deku-m, 2021-03-27:
Yes i did. Sorry i was just curious if someone did try.

@serrj-sv 
Do you perhaps still sell those adapters, i would like to buy 2 if possible..?
Im also located in Netherlands btw.

serrj-sv, 2021-03-28:
Guys, I'm very sorry to say that I'm done with orders. To the people whom I promised here and didn't sent: I'm also canceling these orders.

koelkast88, 2021-03-31:
Today I got all parts for soldering a few adaptors, I have parts for 4 in total. If you are interested, contact me.

th3sniper86, 2021-04-01:
@absalom-muc What type of condensator did you use for C2 and C3? Tantalum?

absalom-muc, 2021-04-01:
> 
> 
> @absalom-muc What type of condensator did you use for C2 and C3? Tantalum?

simple ceramic, no Tantalum

deku-m, 2021-04-02:
> Today I got all parts for soldering a few adaptors, I have parts for 4 in total. If you are interested, contact me.

How can i contact you? im interested in 2 adaptors :)

synegic, 2021-04-03:
@koelkast88 I'd also be interested in two adapters, if they are still available. :)

koelkast88, 2021-04-03:
@deku-m and @synegic contact me at fokke[at]neefjes.info , I have still parts for 4 adaptors.

DYLaKo, 2021-04-09:
If someone in the Netherlands is still looking for these adapters, I have some for sale, in DIY and complete soldered version, see:
https://gathering.tweakers.net/forum/list_messages/2065058

th3sniper86, 2021-05-21:
I also have 1 module available for sale. 
Plug & Play (please provide your wifi credentials and mqtt broker IP). Tested in my unit.
Easy accessible after installation because of the cable connection.

[image]
[image]
[image]

€30,- (total, including shipping cost: €34)

buennerbernd, 2021-05-21:
I would buy it. buennerbernd-github@yahoo.com

swa72, 2021-06-27:
I'm too looking for the assembled device (actually I need two). I can cover the software side and reflash the device if necessary but hardware is beyond my skill ;-) @DYLaKo link is dead.

DYLaKo, 2021-06-27:
> I'm too looking for the assembled device (actually I need two). I can cover the software side and reflash the device if necessary but hardware is beyond my skill ;-) @DYLaKo link is dead.

Just tested the link, it is still working. I still have some units available.

swa72, 2021-06-27:
My fault. I misinterpreted the error message as 404ish - I probably didn't authenticate. Anyway, would you ship to Germany? How much do you charge for a unit? Best, Stefan

DYLaKo, 2021-06-27:
> My fault. I misinterpreted the error message as 404ish - I probably didn't authenticate. Anyway, would you ship to Germany? I'm interested in two units (the premium option with no soldering, please ;-)). Best, Stefan

I need to check for the shipping cost to Germany, I can do that tomorrow and let you know. How can I contact you? Do you have an email address?

swa72, 2021-06-27:
Posted my mail at https://gathering.tweakers.net/forum/list_messages/2065058 (which is a little more private than github).

DYLaKo, 2021-06-27:
> Posted my mail at https://gathering.tweakers.net/forum/list_messages/2065058 (which is a little more private than github).

Found it. You should have an email now.

jenshorn, 2021-07-27:
@DYLaKo Do you have any additional devices left? Drop me a mail: mhiac at sharklasers.com

DYLaKo, 2021-07-28:
@jenshorn I just send you an email. You can remove your email address from your post above.

jenshorn, 2021-07-28:
@DYLaKo Hm, unfortunately sharklasers went down so couldn't receive it. Would you try again with jenshorn at gmail dot com instead?

DYLaKo, 2021-07-28:
@jenshorn I just forwarded the mail to your gmail address.

RobertJansen1, 2021-11-05:
i have 2 of these units available for purchase, if someone needs it: https://tweakers.net/aanbod/2835762/mitsubishi-ac-wifi-controller.html

darkenbula, 2021-12-09:
I'm looking for a pair of these in Australia. Would anyone be willing to ship me an assembled pair please

DYLaKo, 2021-12-09:
> I'm looking for a pair of these in Australia. Would anyone be willing to ship me an assembled pair please

I still have some, but I see that shipping to Australia from the Netherlands is about $86,50, which is very very expensive.

darkenbula, 2021-12-12:
> > I'm looking for a pair of these in Australia. Would anyone be willing to ship me an assembled pair please
> 
> I still have some, but I see that shipping to Australia from the Netherlands is about $86,50, which is very very expensive.

How much would you want for the pair posted to Australia?

juan11perez, 2022-01-11:
Good day anyone attempted to use this with sez units?

kayjaybees, 2022-02-27:
Is anyone in Australia selling any units?
I'm chasing a couple of them.

darkenbula, 2022-02-27:
Would also love a pair of these in Australia

RobertJansen1, 2022-02-28:
I can ship them to Australia for a total price of €60 for a complete soldered set including a lolin d1 and shipping fee 

A soldered package without esp will be €50,-

kayjaybees, 2022-02-28:
> 

Is this for 2 complete units? Plug in and ready to go?
What are your contact details?

WorlockM, 2022-02-28:
I think not. Shipping to Australia alone with DHL is already € 50,-

I would be happy to ship the modules to Australia, but I think it would be cheaper to order parts yourself. My price for ready to go (plug&play) modules is € 35,- per set. So 2 of them would be € 75,- but if you order 2 I would give a discount, so it would be € 70,- and then the € 50,- shipping makes € 120,- (still way cheaper than the Airconwithme modules)

If anyone is interested I can be contacted at `mhi-wifi [at] slinkos [dot] nl`

Rescla, 2022-02-28:
Shipping to Australia with PostNL is much cheaper than €50, €11 for a "brievenbuspakje" <350g. No experience with using them to send stuff to Australia though.

WorlockM, 2022-02-28:
You are totally right Rescla, did not see that. Well in that case I can deliver for a total of € 70,- for 2 modules.

RobertJansen1, 2022-02-28:
> > 
> 
> Is this for 2 complete units? Plug in and ready to go? What are your contact details?

You can contact me through mail by my first name and then the thing you would normally find in an emailadres dot nl

DYLaKo, 2022-06-20:
> Does anyone have units available with shipping to EU?
How many units and to which country?

mkaleu, 2022-06-20:
Same answer: How many and where are you. I have about 10 available.

ddrag001, 2022-06-21:
I need just one at the moment, to see if its performing well and if it suits my needs. I have a MHI 5 way multi-split AC , so I will need 4 more if the module is doing ok. Location is Bulgaria

DYLaKo, 2022-06-21:
> I need just one at the moment, to see if its performing well and if it suits my needs. I have a MHI 5 way multi-split AC , so I will need 4 more if the module is doing ok. Location is Bulgaria

If you send me an email at mhi@[my-username].NL we can check the options together.

mkaleu, 2022-06-21:
> 
What do you need? Just the circuit board or all of it? It will be cheaper if only the circuit board is sent in a letter.

Boki3333, 2022-07-06:
> > 
> 
> What do you need? Just the circuit board or all of it? It will be cheaper if only the circuit board is sent in a letter.

Hello. I would need 3 fully assembled modules , but no need to put Wemos on it. 
How can i get in contact with you ? 
If you have WhatsUp , that would be awesome )) +38631236837
If you do not have Whats Up, then my email : moja.poraba@gmail.com

Please contact me asap :)

Thanks

DYLaKo, 2022-07-07:
Hi @Boki3333, I have send you an email with what I can offer you. If it fits your need, I suggest you remove your contact details above so it won't get picket up by spambots. :)

ampersandhd, 2022-07-20:
Anyone here who could ship a final product to Germany or could do it if I send the PCB to you?

ampersandhd, 2022-07-20:
> > Anyone here who could ship a final product to Germany or could do it if I send the PCB to you?
> 
> 
> 
> Possible, please send me an email to exchange details. See my email address a few posts earlier .
> 
> 

You got mail?

mkaleu, 2022-07-20:
Yes, it's hot! People keep asking for a finished board. If you then name the prices, it quickly becomes quiet. The cost of a complete board is around 32 euros, because you have to buy the components here in Germany and pay customs duty on the board yourself from China. If you would like such a finished board, then contact me again with a private mail.

ampersandhd, 2022-07-20:
How to Contact you?

ampersandhd, 2022-07-20:
@Boki3333 i think the 32 Euro include soldering and the other components. How much is your Board?

mkaleu, 2022-07-20:
My dear @Boki3333, obviously you have no idea about the costs in Germany. The material costs: 18d20=-.80;DCDC=3.90;Elko=-.31;2xC=.12;D1=4.80;JST=-.87;Converter=-.80;Board=4.50and 2.40Custom, all together 18.50 without the shipping costs that are incurred because not everything comes from one supplier. Then it should be soldered together, guaranteed and also shipped. 32 euros are absolutely OK. I don't trade with it and don't have to earn my money with it, and I don't have anything to give away either. With such stupid discussions, I lose the desire to sell the boards. Or are you not concerned with the boards, but only with presenting your idea?

[62 more replies]