absalom-muc/MHI-AC-Ctrl #53: MHI-AC-Ctrl + wired remote RC-EX3A or RC-E5 + SC-BIKN2-E

Dmitriy-Petukhov, 2021-02-01:
Hello!

Have you had any experience with connecting the MHI-AC-Ctrl and connecting a wired remote controller such as RC-EX3A or RC-E5 at the same time?

To connect a wired RC-EX3A or RC-E5 remote controller, I use the SC-BIKN2-E interface.

First, I connect MHI-AC-Ctrl to the CNS connector on the main board of the indoor unit, then I connect the second connector on MHI-AC-Ctrl to connect SC-BIKN2-E.

When the SC-BIKN2-E is connected, only the RC-EX3A wired remote control works, and although the MHI-AC-Ctrl is connected, it does not connect to the Home Assistant and does not receive topics and does not exchange information.

As soon as you turn off the SC-BIKN2-E interface, MHI-AC-Ctrl immediately starts working, receiving topics and exchanging information with the Home Assistant.

The question is, what do you think can MHI-AC-Ctrl + RC-EX3A or RC-E5 + SC-BIKN2-E interface work simultaneously? Or just individually?

Thanks for your reply.

I am using MHI - SCM60ZM-S1 + SRR25ZM-S + SRR35ZM-S.

absalom-muc, 2021-02-01:
> First, I connect MHI-AC-Ctrl to the CNS connector on the main board of the indoor unit, then I connect the second connector on MHI-AC-Ctrl to connect SC-BIKN2-E.

This wiring creates a short of the MISO outputs of MHI-AC-Ctrl and SC-BIKN2-E. This could damage one of these devices.

The SPI bus is a single-master system, so it is not possible to have multiple masters. You could remove the MISO output of MHI-AC-Ctrl, then you should be able to monitor the system status. (I used this configuration to observe some details of the applied protocol). But of course it is no longer possible to change the AC settings via MHI-AC-Ctrl.

Dmitriy-Petukhov, 2021-02-02:
Thank you for your answer

closed 2021-02-02 by Dmitriy-Petukhov