absalom-muc/MHI-AC-Ctrl #54: Static ipaddress

vergoossen, 2021-02-07:
Hi,

Can you make it possible to set a static ipaddress for the device?
I tried some things myself, but they fail. Im not familiar with arduinoide...

absalom-muc, 2021-03-07:
You could insert between row 10 and 11 of [support.cpp](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/src/support.cpp) something similar as:
```
  IPAddress ip(192, 168, 178, 125);
  IPAddress gateway(192, 168, 178, 1);
  IPAddress subnet(255, 255, 255, 0);
  IPAddress dns(192, 168, 178, 1);
  WiFi.config(ip, dns, gateway, subnet); 
```
Please find some docu related to WiFi.config() [here](https://www.arduino.cc/en/Reference/WiFiConfig)

vergoossen, 2021-03-08:
Thankyou.
And is it possible to put ip/gateway/dns in support.h?
When i try myself i get 'IPAddress' does not name a type

absalom-muc, 2021-04-04:
I believe it is not possible to assign a value to a variable from type IPAddress in support.h. If you are interested in details have a look into [IPAddress.cpp#L38](https://github.com/esp8266/Arduino/blob/1c57b3408cfdc2956d3c11307b79e2e0055db038/cores/esp8266/IPAddress.cpp#L38).

btw: There was a typo in my [example ](https://github.com/absalom-muc/MHI-AC-Ctrl/issues/54#issuecomment-792355585)above. The right order for WiFi.config is
`bool config(IPAddress local_ip, IPAddress gateway, IPAddress subnet, IPAddress dns1 = (uint32_t)0x00000000, IPAddress dns2 = (uint32_t)0x00000000);`
according to the [ESP8266WiFiSTA.h](https://github.com/esp8266/Arduino/blob/1c57b3408cfdc2956d3c11307b79e2e0055db038/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.h#L45).

closed 2021-04-12 by vergoossen