absalom-muc/MHI-AC-Ctrl #55: WDT Soft Reset question

aidbish, 2021-03-31:
HI 
Just built the controller and uploaded the code, however it keeps on soft resseeting
I have not connected it to the AC yet, so im just wondering if there is no connection would this be the cause of the soft reset

HI-AC-Ctrl starting
CPU frequency[Hz]=80000000
Attempting WiFi connection ............... connected to IoT, IP address: 10.0.0.101
OTA Ready
Attempting MQTT connection... connected
status=65 topic=connected payload=1
status=64 topic=RSSI payload=-65
Measure frequency for SCK, MOSI and MISO pin
status=68 value=0
status=68 topic=fSCK payload=0
status=69 value=0
status=69 topic=fMOSI payload=0
status=70 value=0
status=70 topic=fMISO payload=0
SCK frequency=0Hz (expected: >3000Hz) out of range!
MOSI frequency=0Hz (expected: <SCK frequency) out of range!
MISO frequency=0Hz (expected: ~0Hz) o.k.

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

>>>stack>>>

ctx: cont
sp: 3ffffdb0 end: 3fffffc0 offset: 01a0
3fffff50:  00000064 3ffee508 00001e9d 40201289  
3fffff60:  00000000 00000000 00000008 00000000  
3fffff70:  3ffe84dc 3ffee4f8 3ffee62c 4020622a  
3fffff80:  3fffdad0 00000000 3ffee534 3ffee8a8  
3fffff90:  3fffdad0 00000000 3ffee508 4020249f  
3fffffa0:  feefeffe feefeffe 3ffee868 4020cd18  
3fffffb0:  feefeffe feefeffe 3ffe8514 40101105  
<<<stack<<<

Rescla, 2021-04-01:
I don't remember where in the code this is handled, but I remember having the same behavior when bench testing my own board. It worked fine when connected to the MHI though.

absalom-muc, 2021-04-01:
yes, this behavior is common if not connected to the AC.

aidbish, 2021-04-01:
Thanks for confirming.

closed 2021-04-01 by aidbish