absalom-muc/MHI-AC-Ctrl #56: PCB improvement for SRK-ZSX serie

DYLaKo, 2021-04-26:
First of all: Excellent project & development. I have 2 MHI airco's and both are equipped with your solution and it works great!

Last week I have provided one of my ordered PCB's to someone who also wanted to build an airco controller based on your software. After installation, he provided me with some feedback for PCB improvement, since he had to modify the PCB to make it fit in his airco.

The airco used is one from the Mitsubishi Heavi Industries SRK-ZSX series. Next to the connector on the mainboard of the airco, there is a big elco which prevents the PCB to be fitted in the connector. This could offcourse be solved with an extension cable, but he removed a small piece of the PCB to make it fit.

Photos are attached for your reference, maybe you can use them to further improve your PCB design. 

[image]
[image]

absalom-muc, 2021-04-28:
Thank you for the hint @DYLaKo, I will add a link to this issue to the HW description and consider it for a future update of the PCB.

closed 2021-04-28 by absalom-muc

absalom-muc, 2022-01-20:
The [supported units list](https://github.com/absalom-muc/MHI-AC-Ctrl#prerequisites) gives you only the infromation if MHI-AC-Ctrl is basically supported by a unit, e.g. related to the SPI protocol and the electrical connection. The feature for plug-in MHI-AC-Ctrl directly into the AC without using a cable is only since the last PCB update (Sept 2021). And I'm not sure when we received the confirmation that SRK ZS-W units work with MHI-AC-Ctrl. Before or after Sept 2021. So worst case there could be a collision. And I'm also not sure if there is enough space for an extension.
Let's keep finger crossed.

DYLaKo, 2022-01-20:
I have 2 SRK35ZS-W units in my house, running like a charm with the MHI-AC-Ctrl board, so that works. :)

birarda, 2022-01-21:
> The [supported units list](https://github.com/absalom-muc/MHI-AC-Ctrl#prerequisites) gives you only the infromation if MHI-AC-Ctrl is basically supported by a unit, e.g. related to the SPI protocol [...]

@absalom-muc thanks for the info!

In this response you mention a Sept 2021 PCB update but it looks like the latest in the repo is Jan 2021?

absalom-muc, 2022-01-21:
Sept 2021 is the date when the new PCB was released in v2.2. Introduced was it in Jan 2021. If someone trusted the "draft" version in the repo the Jan is the deadline, for a release version the September.

johnneerdael, 2023-08-08:
Have you found any instructions ANYWHERE on how to access the PCB?

honkasalo, 2026-09-20:
> Have you found any instructions ANYWHERE on how to access the PCB?

Access to PCB is really good question eceryone seems to leave unanswered! Findind CNS connector on PCB can't be that difficult. Assembling ESP-board is easy, as well as flashing it. I have removed front cover from my heat pump wall unit, as an engineer I know how to assembly and disassembly things, bit in this case, my wall unit has black plastic box (obviously contains PCB) and no obvious way to remove it safely, without breaking anything. This is really the hard part, not electronics itself.