absalom-muc/MHI-AC-Ctrl #58: SRK-50-ZSW in relation to Home Assistant (use of capitals to set mode)

henkjanvanderkolk, 2021-04-28:
Many thanks for this great script. 

Plugged the unit in this afternoon. Setpoint, vane position and fan speed worked out of 'the box', but mode was returning 'invalid parameter on my MQTT listener. Changed the MQTT 'PAYLOAD_MODE' in MHI-AC-Ctrl.h:

- Auto -> auto
- Stop -> stop
- Dry -> dry
- Cool -> cool
- Fan -> fan
- Heat -> heat 

That did the trick on my SRK-50-ZSW.

absalom-muc, 2021-04-30:
Thanks for the info @henkjanvanderkolk 
Only for completeness: We had similar topics already [here](https://github.com/absalom-muc/MHI-AC-Ctrl/issues/14) and [here](https://github.com/absalom-muc/MHI-AC-Ctrl/issues/29)

henkjanvanderkolk, 2021-04-30:
My bad. Thank you.

closed 2021-08-14 by absalom-muc