absalom-muc/MHI-AC-Ctrl #59: IotMQTTPanel Dashboard export

DYLaKo, 2021-04-29:
On your first page of this project, there is an image of a created dashboard in IotMQTTPanel. 

Do you happen to have an export of that dashboard which can be used as a baseline for other users with the same app?

absalom-muc, 2021-04-30:
No problem, please have a look on [IoTMQTTPanel-210330_095559.txt](https://github.com/absalom-muc/MHI-AC-Ctrl/files/6404612/IoTMQTTPanel-210330_095559.txt)

Before usage you have to change the file extension to .json and in the json file you have to adapt the MQTT broker name. Name of my broker is ds218p. I don't use username/password for the broker.

DYLaKo, 2021-04-30:
Thank you. Just tested this for my setup.
My remarks, for future use for when someone else wants to use it too.

You need to remove the comma  `,`  after the  close brace `}` on line 293, otherwise the JSON file will give an error when importing. 

If you want to use username & password for the broker, you can add it between line 9 and 10. Just add the following lines:
`			"username":"your_username",`
`			"password":"your_password",`

If you want to create multiple dashboards (for example, when having 2 airco devices), copy all lines from line 16 until line 291, and paste them from line 292. Make sure that you add a comma `,` behind the close brace `}` on line 291.
In the pasted lines, change the following lines to match your setup:
- `dashboardID` (Must be the following dashboard in your config. When this is the second dashboard, this should be 1)
- `dashboardName` (Unique name to recognize the dashboard in the app)
- `dashboardPrefixTopic` (Should match your MQTT topic as configured in the controller).

When also using HomeAssistant, you probably changed the payloads in the configuration of the controller to be lower case. If that is the case, you should change the payloads in this dashboard to lower case also.

absalom-muc, 2021-09-18:
I believe your question is answered, so I close this topic. But feel free to re-open it in case of further questions.

(in general please close an issue once it is answered - thank you)

closed 2021-09-18 by absalom-muc

Bill-Dung, 2024-07-06:
The JSON import doesn't work anymore even with the changes DYLaKo suggested. Maybe the file has changed to much, the actual dataVersion is 152, this one here is 54. Would be nice if you could upload a newer backup. Thx in advance. 

Btw. great work, I'm using MHI-AC-Ctrl  for quite a while and am very happy with it. I use the ESPHome version of Ginkage. Now my brother in law will use the "original" via mqtt ;-)