absalom-muc/MHI-AC-Ctrl #60: Example: Node-Red integration with Google Assistant

pdufrenne, 2021-05-04:
I want to share my Node-Red integration for this Heath-pump module.

It provides a dashboard for controlling the most important airconditioning functions.
But the best part it, provides integration with Google Assistant.
For Google-Assistent only a small sub-set of functions are available
With your voice you can set the set-point temperature, heat, cool or heat-cool function, request the actual room temperature and set the airco “off”.

Google Assistant integration is build with the new Smart-Nora service, which works fast and accurate.

You have to register an account at: https://smart-nora.eu/

You can use the code as an example for building your own dashboard and integration.

Succes!

Peter D.

[Heath-pump.json.txt](https://github.com/absalom-muc/MHI-AC-Ctrl/files/6421948/Heath-pump.json.txt)

[image]
[image]
[image]

absalom-muc, 2021-05-15:
Thanks for sharing @pdufrenne !
I added a link to this issue in [SW-Configuration](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/SW-Configuration.md#integration-examples)

absalom-muc, 2021-09-18:
Since your integration is referenced in the documentation I close this issue. Thanks for sharing it.

closed 2021-09-18 by absalom-muc