absalom-muc/MHI-AC-Ctrl #62: SPI Timing

stumpjr, 2021-06-18:
I tried this Projekt on an ESP32. The problem is that the SPI frequency is mostly below 3000 Hz.
The reason seems to bee that there are large gaps between the clock cycles. See Oscilloscope picture.
So i have varying frequencies depending where the measurement starts. The Unit is a SRK25ZS-W.
Do other units have a continous clock? The part where the clock is not idle seems to have a frequence around 30 kHz
Any ideas how to accomplish this?

[image]

absalom-muc, 2021-06-19:
Your timing looks good. Have you compared with the [timing diagram](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/SPI.md#timing)?

absalom-muc, 2021-09-18:
I believe your question is answered, so I close this topic. But feel free to re-open it in case of further questions.

(in general please close an issue once it is answered - thank you)

closed 2021-09-18 by absalom-muc

oliv3r, 2024-06-10:
Just for completness sake, I wonder if the confusion is from the fact that the ESP32 is a slave, and the AC is the master? So the 'breaks' is just the master (the AC) deciding to do something else. The AC probably 'polls' the client, and what you see is the poll interval.

New link to the document which has since moved ;) https://github.com/absalom-muc/MHI-AC-Trace/blob/main/SPI.md