absalom-muc/MHI-AC-Ctrl #63: Room temperature

mark5612, 2021-06-22:
The original 'rjdekker' code (for MHI-AC-Ctrl.ino) has a note to check his derivation.
As I've tested over time, room temperature appears to actually represent "0.25 degree increments above -20C".
At the very least, the offset in the code should be a multiple of 4 !  (currently -61)
So, temp = (value - 80)/4

absalom-muc, 2021-08-29:
In MHI-AC-Ctrl the formula described [here ](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/SPI.md#room-temperature-read-only) is used or the room temperature. I compared in the past the values with commercial available remote controls. And the values matched. Therefore I assume the used formula temp = (value - 61)/4 is correct

absalom-muc, 2021-09-18:
I believe your question is answered, so I close this topic. But feel free to re-open it in case of further questions.

(in general please close an issue once it is answered - thank you)

closed 2021-09-18 by absalom-muc