absalom-muc/MHI-AC-Ctrl #64: Domoticz/fMISO

dreamer1976, 2021-06-25:
I installed software and can see MHI-AC-Ctrl in MQTT Explorer.

For my next step, I have two questions. I hope someone can help me.

1. I see these values in MQTT Explorer, listed below. I notice fMISO=0, is this correct?
connected = 1
RSSI = -62
fSCK = 3200
fMOSI = 560
fMISO = 0

2. I would like to connect to Domoticz. Both Domoticz and Mosquitto run on the same Raspberry Pi. Is this possible? Does anyone know of any tutorials on how to do this, or has someone done this already?

Thanks.

absalom-muc, 2021-06-27:
I can only answer the first question: Yes these values are o.k., please compare with [MHI-AC-Ctrl-core.cpp line 79](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/4531327ff2b474eb047317975211df0430bdf199/src/MHI-AC-Ctrl-core.cpp#L79)

DYLaKo, 2021-06-28:
For your second question: I do not use Domoticz, but Home Assistant, and run both HA and the MQTT broker on the same machine without any problems. I assume this should also be possible with Domoticz.

dreamer1976, 2021-06-28:
Thank you, absalom-muc and DYLaKo for your replies. This is helpful and reassuring to know. 

@DYLaKo do you use something like Node-RED for the connection between HA and MHI?

If anyone else has any experience with Domoticz, I would be interested to know. Domoticz uses an idx as an identifier and I am unable to figure out these are set. Whether this is something Domoticz assigns, or if the idx is assigned by the MQTT broker.

As an aside, is there an app that can connect directly to an MHI ESP module? E.g. something similar to Blynk?

DYLaKo, 2021-06-29:
> Thank you, absalom-muc and DYLaKo for your replies. This is helpful and reassuring to know.
>
> @DYLaKo do you use something like Node-RED for the connection between HA and MHI?
>
> If anyone else has any [...]

I am not using Node-RED. Home Assistant can run addons, I run the MQTT Broker as an addon within Home Assistant. Home Assistant itself is a MQTT client which listens to the topics entered in the HA configuration and publishes the sensors & controls as entities within Home Assistant. From there, I can use the airco as HVAC unit with all needed controls.

For your app question, if you have setup a MQTT broker (which is a requirement you cannot drop for this software to work), you can use a MQTT Client app on your mobile device to connect to the broker and control the airco.

An example is the android application IoTMQTTPanel, see the example of @absalom-muc below:
https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/images/IoT-MQTT-Panel.jpg

The dashboard export and some configuration hints for IoTMQTTPanel can be found in the following issue:
https://github.com/absalom-muc/MHI-AC-Ctrl/issues/59

DYLaKo, 2021-06-29:
@dreamer1976, did you add the MQTT broker already to Domoticz as gateway?
It is mentioned here:
https://www.domoticz.com/wiki/MQTT#Add_hardware_.22MQTT_Client_Gateway.22

The config also mentions the topics Domoticz listens for incoming and outgoing MQTT traffic (Topic In Prefix and Topic Out Prefix), make sure that those are set to any topic your controller is also publishing to.

dreamer1976, 2021-07-06:
Thanks for the responses. Domoticz works with an idx(number that a senser or switch needs). I now have a test working with node red and can read it in domoticz. I will test more thanx.

absalom-muc, 2022-06-16:
closed since there was no update for a long time. Please feel free to re-open in case of follow-up topics.

closed 2022-06-16 by absalom-muc