absalom-muc/MHI-AC-Ctrl #65: Level Shifter necessary?

SpatialJ, 2021-06-28:
Hi,
usually the esp8266 are 5v 'tolerant', has anyone tried to connect an esp without the extra levelshifter?

DYLaKo, 2021-06-28:
As far as I know, the ESP8266 as used (The Wemos D1) only supplies 5V, but does not accept 5V on the inputs. That is what the level shifter is used for.

SpatialJ, 2021-06-28:
I have a pzem-004t directly connected with 5v ttl on an esp, and it works.
That's why I was wondering if anybody tried it yet. It might of course decrease the lifespan of the esp.

absalom-muc, 2021-08-14:
According to the electrical characteristics the esp8266 is **not** 5V tolerant. I'm sure the vendor would release the devices for 5V if they could guarantee that it would work for all operating conditions.

closed 2021-08-14 by absalom-muc

SpatialJ, 2021-08-14:
Well, as desceibed in the link I provided, it is :-D

absalom-muc, 2021-08-14:
> 
> Well, as desceibed in the link I provided, it is :-D 

Sorry, but your conclusion is wrong. There might be some examples (maybe 10 or 100) where the user haven't observed a wrong behavior or damage when using 5V for the port inputs. But it is wrong to conclude that this is valid for all ESP8266 (maybe many millions).

dasrue, 2021-11-25:
Hi, I have set mine up with no level shifter today. I needed to put a pullup resistor (I used 12K) on the MISO line to stop it picking up a frequency on the line. It works. Beforehand I also tested putting 5V into the pins and found the draw was only a few uA so I don't think there will be any damage to the pins

nmg196, 2022-06-16:
The CEO "of Espressif has stated that the IO pins (only) are 5V tolerant.

[image]

....and the reason it doesn't mention this in the data sheet is apparently due to user confusion:

[image]

(Source: https://www.facebook.com/groups/1499045113679103/permalink/1731855033731442/?hc_location=ufi)

So it seems a level shifter is definitely not necessary.

absalom-muc, 2022-06-16:
Thanks for sharing it @nmg196. It sounds strange for me considering a Facebook statement,  but not the datsheet. This is for me more confusing, especially because I don't understand the statement 
[image]
Does he mean "open drain"? Is it a hint that a resistor should be used?
Maybe I'm being overly cautious but I will continue using the level shifter - just my 2 cents.