absalom-muc/MHI-AC-Ctrl #66: External temperature sensor

SpatialJ, 2021-06-28:
What is the motivation for this feature? Are there issues with the internal one ?

DYLaKo, 2021-06-28:
The board itself does not have a temperature sensor, it reads the temperature sensor from your airco unit. In case of heating a room with the airco unit, the internal temperature sensor will give a higher reading than the real temperature (because warm air rises).

If you add the external sensor, you can also measure on another height for example.

SpatialJ, 2021-06-28:
I see. One should be very careful though, leading wires out of the isolating  enclosure:-)

absalom-muc, 2021-09-18:
I believe your question is answered, so I close this topic. But feel free to re-open it in case of further questions.

(in general please close an issue once it is answered - thank you)

closed 2021-09-18 by absalom-muc