absalom-muc/MHI-AC-Ctrl #67: "SELF CLEAN operation" to prevent mold

al400k, 2021-07-11:
Hello,
is it possible to switch on Aircon in "SELF CLEAN operation" with MHI-AC-Ctrl?

SELF CLEAN operation should be run after AUTO, COOL and DRY operation to remove the moisture from inside of the indoor unit andcontrol the growth of mold and bacteria.
The indoor unit fan runs for about two hours in SELF CLEAN operation
(can be switched on using settings on remote control)

See mitsubishi heavy manual that explains that function:
https://www.mhi-mth.co.jp/en/products/pdf/rlf012a201_english.pdf

When this function is swithed on, i see this data in MQTT Explorer:

MHI-AC-Ctrl-1
connected = 1
RSSI = -79
fSCK = 3200
fMOSI = 500
fMISO = 0
Mode = fan_only
Power = on
Fan = 1
Vanes = ?
Troom = 30.75
Tsetpoint = 24
Errorcode = 0
OpData
	Mode = fan_only
	Tsetpoint = 24
	RETURN-AIR = 31.00
	THI-R1 = 27
	IU-FANSPEED = 1
	TOTAL-IU-RUN = 4600
	OUTDOOR = 34.50
	THI-R2 = 71
	COMP = 0.00
	THO-R1 = 19
	TD = 43
	THI-R3 = 27
	CT = 0.00
	PROTECTION-NO = 0
	OU-FANSPEED = 0
	DEFROST = off
	TOTAL-COMP-RUN = 3800
	OU-EEV1 = 250
	TDSH = 2
	cmd_received = o.k.

in this mode: 
	Vanes are closed,
	indoor fan speed- very slow (slower than minimum on cool mode)
	OUTDOOR unit -is swithed off

thank you.

absalom-muc, 2021-07-13:
Activating "SELF CLEAN operation" with MHI-AC-Ctrl is not supported because I don't know if the self-clean command is supported via the [SPI](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/SPI.md#spi-bus) bus and - if yes - which SPI command it is. Currently I have no plans to analyze it.

drdoener, 2021-07-26:
Hi,
for you info: I subsituted this function by a short script in node red.
Shut down with SELF CLEAN: Via MHI-AC-Ctrl immediately to FAN mode, after 2 hours via MHI-AC-Ctrol-> OFF.

al400k, 2021-07-26:
thanks,  i did the same with script, but "SELF CLEAN" works little different because vanes are closed. I thought there is the way to send command to MHI (fan=1(nomal min mode=2),vanes=0) but it is not so easy.

drdoener, 2021-08-06:
Hi,
that sounds interesting. I **assumed** that there is no difference between SELF CLEAN and the ordinary FAN mode.
How have you figuered it out? I can´t see the difference.
As an example, I dumped the values via MHI-Ctrl and copied those into the attached xls to compare the values.
[MHI_SELF_CLEAN.xlsx](https://github.com/absalom-muc/MHI-AC-Ctrl/files/6944299/MHI_SELF_CLEAN.xlsx)

And here are two pics:
The IU in FAN mode:
[image]

The IU in SELF CLEAN mode:
[image]

Thanks & greets,
drdoener

al400k, 2021-08-06:
Hello,very strange, your data shows mqtt.0.MHI-AC-Ctrl.OpData.IU-FANSPEED=2 like normal minimum fan speed.
 are you sure that you push "SELF CLEAN operation" and not the "ALLERGEN CLEAR operation"?
[image]
[image]

drdoener, 2021-08-06:
Hi,
really sure. This mode is set in my RC and the timer on RC display runs down when I press "ON/OFF)
I never used the "ALLERGEN CLEAR operation". Just tried it out. After pressing the "Alien Button" on the RC, the Unit went ON und shut down after approx. 1 min.

al400k, 2021-08-06:
Probably depends of model, mine is: SRK25ZS-W and SRK40HG-S both working with closed Vanes and Fan=1 in Self clean mode. May i know your model?

drdoener, 2021-08-06:
It´s MHI SRK 35 ZSX - W

drdoener, 2021-08-10:
Hi,
disregard everything I wrote above.
I had a huge bug in my scripts, resulting in triggering the normal FAN mode when I switched on SELF CLEAN via RC.
This means: Same at my device, if I switch on SELF CLEAN, the vanes become closed and **IU-FANSPEED = 2** for 2 hours.

Further info:
Prior MHI-AC-Ctrl, I used the original WLAN Adapter via intensis cloud.
As far as I remember (I`m very sure), as well the SELF CLEAN function didn´t work via the WLAN Adapter.
That leads me to expect, that the SELF CLEAN mode is only working via the RC.

Sorry for the misleading information, I´ll delete my posts above if you agree.

al400k, 2021-08-10:
thank you for sharing, now everything is clear:) We still don't have working solution for "SELF CLEAN" mode except Fan mode with script.

kARTechnology, 2021-10-31:
I

> 
> 
> thank you for sharing, now everything is clear:) We still don't have working solution for "SELF CLEAN" mode except Fan mode with script.

Is it possible to close the vanes fully with mhi-ac-ctrl? and also we see the power led blinking (3sec on/1sec off) is present which means there is some mode to activate it.

kARTechnology, 2022-01-20:
i would like to change the action on the off command to

1. set temp to 30deg
2. fan at 2 speed
3. run for 2 hrs on a timer and then turn off
4. and reset the temp to what value it was set before(either by remote too, so how to read the current value from the esp/ac before i change it to 30)

if the ac is turned on, the running timer to turn off after 2 hrs is cancelled.

this looks the workaround but without it my ac starts to stink. even though we tried analyzing the raw values thru the mhi-trace, unable to fund out. there might be some other setting bit to set but without a wired RC with self clean, cant explore

absalom-muc, 2022-06-16:
closed since there was no update for a long time. Please feel free to re-open in case of follow-up topics.

closed 2022-06-16 by absalom-muc