absalom-muc/MHI-AC-Ctrl #70: New PCB revision

Spud2233, 2021-09-07:
This is less of an issue and more of a hardware revision.

I have been working on a version of the PCB that is almost fully turn-key,
The fully assembled PCB can be ordered from JLCPCB, PCBway, etc all the end user needs to do is flash the firmware and solder on the header connector or lead.

The fully populated board measures 34.5 x 26.5 with a thickness of 5-6mm without the connectors. 
Instead of using the D1 mini as the basis of the module I have opted for an ESP12F. This may not be ideal for some users as it requires a separate USB to serial adaptor for programing, but I have included a programing header.

Any feedback that you can provide would be appreciated.

[image]
[image]
[image]
[image]
[image]
[image]

### **I HAVE NOT BUILT A PROTOTYPE OF THIS YET, USE THE BELOW FILES AT YOUR OWN RISK.**
[BOM_PCB_MHI-AC-Ctrl_v2.8_2021-09-08.csv](https://github.com/absalom-muc/MHI-AC-Ctrl/files/7122527/BOM_PCB_MHI-AC-Ctrl_v2.8_2021-09-08.csv)
[Gerber_PCB_MHI-AC-Ctrl_v2.8_2021-09-08.zip](https://github.com/absalom-muc/MHI-AC-Ctrl/files/7122585/Gerber_PCB_MHI-AC-Ctrl_v2.8_2021-09-08.zip)
[PickAndPlace_PCB_MHI-AC-Ctrl_v2.8_2021-09-08.csv](https://github.com/absalom-muc/MHI-AC-Ctrl/files/7122529/PickAndPlace_PCB_MHI-AC-Ctrl_v2.8_2021-09-08.csv)

damienmorrissey, 2021-09-07:
Nice!

I was just about to put an order in for some boards. I'll review it and get
back to you.

absalom-muc, 2021-09-08:
Thank you @Spud2233 for the layout. I had a rough look on the layout with an online Gerber viewer and there might be some problems, but I'm not sure since I have no scale with the Gerber viewer, so I can only estimate sizes and distances:
I have the impression that the wires are very small while you have sufficient area. In addition it seems that the distances between wires and pads are partly too small. example:
[image]
The small distances and wire sizes increase the risk for short and opens. In parallel the PCB production costs could be unnecessarily increased.
Not sure if your copper heat sink is sufficient for the heating by the approx. 1 Watt thermal power loss.
You could create a short between the programming connector pins and the ESP12F.

Spud2233, 2021-09-09:
Thanks for the feedback, I am fairly new to PCB design so all feedback is appreciated.

I have done a bit more research and made a few changes.

To be on the safe side I have Increased clearance between tracks to 0.254mm/10mil, this should be sufficient for up to 30V DC
The power traces  have been increased to 0.381mm/15mil, the calculations said this should be good for 1A, the signal traces are 0.254mm/10mil.
I have added via's and a 5V plane on the back of the PCB to help with heat dissipation from the AMS1117-5, do you think that is sufficient?
I have also added a ground plane to the top layer of the PCB as well as via fencing at the perimeter of the PCB nearest the ESP's antenna, I'm not sure if this is over the top though?

[image]
[image]
[image]
[image]

### I HAVE NOT BUILT A PROTOTYPE OF THIS YET, USE THE BELOW FILES AT YOUR OWN RISK.
[PickAndPlace_PCB_MHI-AC-Ctrl_v2.9_2021-09-10.csv](https://github.com/absalom-muc/MHI-AC-Ctrl/files/7137360/PickAndPlace_PCB_MHI-AC-Ctrl_v2.9_2021-09-10.csv)
[Gerber_PCB_MHI-AC-Ctrl_v2.9_2021-09-10.zip](https://github.com/absalom-muc/MHI-AC-Ctrl/files/7137361/Gerber_PCB_MHI-AC-Ctrl_v2.9_2021-09-10.zip)
[BOM_PCB_MHI-AC-Ctrl_v2.9_2021-09-10.csv](https://github.com/absalom-muc/MHI-AC-Ctrl/files/7137362/BOM_PCB_MHI-AC-Ctrl_v2.9_2021-09-10.csv)

dvisser, 2021-09-10:
Good work, however you should really avoid right-angles in your pcb traces and with regards to SMD components, you may get some strange "tombstoning" effects if you don't run tracks in certain directions into SMD pads.
This is the PCB a friend and I designed for this purpose if you want to take a look (still uses wemos module as we like how the usb interface is already there).  

https://github.com/dvisser/MHI-ESP-PCB

absalom-muc, 2021-09-10:
It is much improved compared to the first version and I believe the board will work.

However, my target is always to maximize the distances and have some buffer for the wire size. e.g.
[image]
There is no reason having the wire so close to the pin. Why don't you go from the pin to the right and then up?
Might be not a must, but gives you some buffer in case of PCB production issues.
btw: I assume you run a DRC with the ruleset from your PCB vendor?
The ground plane is nice to have, but I believe it is not needed in this case.
And I still see the risk of a short between the programming connector pins and the ESP12F.

damienmorrissey, 2021-09-11:
Agreed. So much better. I'm just getting into this stuff myself and feel
that extra space can not hurt provided it fits the board. What does it look
like if you increase the minimum distance by another 20%? In some respects
you could almost keep going until the board cannot be resolved within the
space constraints without any harm.

That said the rest of the board looks good so perhaps a manual adjustment
would be fine.

Regarding the additional cost, in my case I'm building 5 of them so
delivered and built it would be just under $50 for all 5. That is excellent
value imo. However, if it's a one off it is not as good of a deal.

fonske, 2021-09-16:
What are the differences between this board with ESP12F and the previous one with the Wemos mini?

absalom-muc, 2022-06-16:
closed since there was no update for a long time. Please feel free to re-open in case of follow-up topics.

closed 2022-06-16 by absalom-muc

ThomasHFWright, 2024-05-30:
@Spud2233 did you ever make an updated board or test these? I'm looking to get some fresh PCBs made and shipped and I'd prefer a turnkey solution if possible.