absalom-muc/MHI-AC-Ctrl #71: Adafruit Huzzah 8266

PeterPHanlon, 2021-09-29:
I had been working on the rjdekker model which was well documented. I will swap into your stuff as you seem equally talented, and are active.
[image]
.

Could you comment whether the Adafruit Huzzah Feather has any chance of running this program?

On the MHI AirCon, mine is a huge 3-phase split indoor/outdoor ducted 160VS but much of the reading I have done on it is favourable. I give it rough chance of working. The wired controller seems familiar in your project.

Sorry that English is my only language. 

I put a photo of what I have. Level-shifting would be necessary and understood.

Peter Hanlon
Canberra Australia

absalom-muc, 2021-09-29:
Since Adafruit Huzzah Feather is ESP8266 based I assume it will work.
[Here](https://github.com/absalom-muc/MHI-AC-Ctrl#prerequisites) you find a list of MHI models confirmed to work with MHI-AC-Ctrl.

PeterPHanlon, 2021-09-29:
Thanks Absalom Muc,
If I went with Huzzah, I would need to redo the PCB a little  16 HDR and 12 HDR.
It doesn't quite compile, with printf_p constructions unknown, but similar strcmp all OK. I will check versions of everything. The MHI model (2013) certainly has the JST connector, and it uses a wired controller very recognisable in the stuff I have seen. Because it is ducted, there is no Vane/Louvre function.
Thanks for your help,
Peter

PeterPHanlon, 2021-11-07:
Got the software compiling now. Waiting on a few hardware bits. Covid seems to be affecting deliveries. I think that I may have had an outdated 8266 platform. 
Thanks for your help,
Peter

closed 2021-11-07 by PeterPHanlon