absalom-muc/MHI-AC-Ctrl #73: MHI 18YXS2 Model

kARTechnology, 2021-10-31:
I have successfully connected to my Indian Version (high efficiency 5.7 ISEER unit) to mhi-ac-ctrl. 
1) how to control the Horizontal Left/Right Vanes(7 modes and 1 swing mode)
2) How to set 3D auto operation
3) RC has 5 Vertical Up/Down Swings where as MHI-Ac-Ctrl seems to support only 4
4) Silent Operation of outdoor unit
5) Eco Mode/Powerful Mode
Why am i getting err_msg_invalid_signature and err_msg_invalid_checksum
[image]

Following data is available
[image]

absalom-muc, 2021-11-02:
> 
> 
> I have successfully connected to my Indian Version (high efficiency 5.7 ISEER unit) to mhi-ac-ctrl 2.04. latest version with 2.2 has errors when ac isnt connected to the cns connector. crashes with exception in the mhi core loop(), found out that the changes from 2.04 to 2.2 have caused the issue.

Could be resolved by v2.3

> some enhancements to make:
> 
>     1. how to control the Horizontal Left/Right Vanes(7 modes and 1 swing mode)
> 
>     2. How to set 3D auto operation
> 
>     3. RC has 5 Vertical Up/Down Swings where as MHI-Ac-Ctrl seems to support only 4
> 
>     4. Silent Operation of outdoor unit
> 
>     5. Eco Mode/Powerful Mode

All functionality supported by SPI and known by me is supported. If you are aware that more functionality is supported on the SPI please provide the SPI code.
 
>     6. how to read Humidity sensor value in indoor unit

There is no humidity sensor in my indoor unit, so I don't how to read it out

>     7. whats the difference between troom & return-air values.

I don't know 

>        Why am i getting err_msg_invalid_signature and err_msg_invalid_checksum
>        [image]
> 
> 
> Following data is available [image]

Maybe the cable between AC and MHI-AC-Ctrl PCB is too long? If this is not the case I don't know the answer and I would not spend too much time in it when it is working most of the time

kARTechnology, 2021-11-03:
@absalom-muc 
Thanks for the reply

I've been using 2.3 only. looks like the "releases" section has only 2.2 and in 2.2 only i had got the exceptions
.
How do i help to provide the SPI code, is there any way to capture the data when i use RC and have this mhi-ac-ctrl connected?

cable length is like around 10 to 15cm. good quality one, tried re flashing with 160mhz and little better than before.

i see the refresh rate of data values is pretty slow, like the outdoor COMP frequency measurement is sent like every few minutes. is there any way to get Realtime data like every 5 seconds of all parameters...

absalom-muc, 2021-11-06:
> 
> 
> @absalom-muc Thanks for the reply
> 
> I've been using 2.3 only. looks like the "releases" section has only 2.2 and in 2.2 only i had got the exceptions . How do i help to provide the SPI code, is there any way to capture the data when i use RC and have this mhi-ac-ctrl connected?

I'm using a second MHI-AC-Ctrl PCB in Spy mode (listening only). The raw data are analyzed with a c# program. But currently I don't want to publish this SW.

> cable length is like around 10 to 15cm. good quality one, tried re flashing with 160mhz and little better than before.

maybe it is because SCK frequency is higher than expected (usually 3200Hz).

> i see the refresh rate of data values is pretty slow, like the outdoor COMP frequency measurement is sent like every few minutes. is there any way to get Realtime data like every 5 seconds of all parameters...

You can increase the read out frequency by reducing the number of selected data in [opdata](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/67f4a31c179f2699ce7451895948440d522ad264/src/MHI-AC-Ctrl-core.h#L7).
You could try to reduce the [NoFramesPerPacket](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/67f4a31c179f2699ce7451895948440d522ad264/src/MHI-AC-Ctrl-core.h#L30), you have to test the minimum number.
Finally the read out frequency could be limited by how often the AC provides updated data.

kARTechnology, 2021-11-09:
Fiddling with the noframes(currently at 8) and commenting out many parameters of the opdata, now having only 
returnair, outdoor temp, iu and ou fanspeed, comp freq, ct and protection,
 spi errors occurrence is greatly reduced. whats the min. no. frames/packet
 
I would like to help you get the raw data for the 3d auto, r/r vanes, silent outdoor/eco/powerful/selfclean mode, and the missing u/d vane.
 
I can also connect my laptop to the esp while its connected to the ac.

absalom-muc, 2021-11-13:
> Fiddling with the noframes(currently at 8) and commenting out many parameters of the opdata, now having only
> returnair, outdoor temp, iu and ou fanspeed, comp freq, ct and protection,
> spi errors occurrence is greatly reduced

That is interesting, I was not aware of this effect.

kARTechnology, 2021-11-20:
Leaving 3D auto on on the remote control, subsequent commands from mhi-ac-ctrl doesnt change the state.
Could you please help me provide u raw data to analyze?

kARTechnology, 2021-11-26:
After a SELF CLEAN operation by remote. mhi-ac-ctrl for turning on/off the ac doesnt work unless the unit is again turned on / off without self clean.

absalom-muc, 2021-12-05:
> Leaving 3D auto on on the remote control, subsequent commands from mhi-ac-ctrl doesnt change the state. Could you please help me provide u raw data to analyze?

I plan providing the program for recording the SPI traces in the next weeks. Based on these traces and [SPI.md](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/SPI.md) you could try to analyze it (maybe also related to the SELF CLEAN issue you mentioned before).

kARTechnology, 2021-12-05:
> I plan providing the program for recording the SPI traces in the next weeks. Based on these traces and [SPI.md](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/SPI.md) you could try to [...]

thanks, would be happy to help, i plan to keep this unit for the next 15 years
[image]
.

PietNL, 2021-12-08:
Hi! I suspect that at the eco/power swing and 3d auto modes should be possible if I look at the following 2 guides for the wifi module from MHI.
Installation manual: https://www.mhi-mth.co.jp/en/products/pdf/rkz012a105a_english.pdf
App manual: https://www.mhi-mth.co.jp/en/products/pdf/rkz012a107c_english.pdf

This module connects to the same CNS port on our devices so it should be possible to offer the same functionality or am I seeing this wrong? :)

damienmorrissey, 2021-12-08:
Makes sense. If we had a wifi module we would probably be better off. Do we
know of someone who has a wifi module?

PietNL, 2021-12-08:
Hmmh, just found this Repo: https://github.com/TheZeroBeast/ACWiFi, for me (not a developer) it looks like the 3dauto function is included in that code?

damienmorrissey, 2021-12-08:
Certainly looks like it...

kARTechnology, 2021-12-08:
> Hmmh, just found this Repo: https://github.com/TheZeroBeast/ACWiFi, for me (not a developer) it looks like the 3dauto function is included in that code?

looks like ir code for 3d auto and not by cns?

absalom-muc, 2021-12-18:
> 

> I plan providing the program for recording the SPI traces in the next weeks. Based on these traces and [SPI.md](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/SPI.md) you could try to [...]

Please have a look on [MHI-AC-Trace](https://github.com/absalom-muc/MHI-AC-Trace)

kARTechnology, 2021-12-18:
> Please have a look on [MHI-AC-Trace](https://github.com/absalom-muc/MHI-AC-Trace)

[image]

absalom-muc, 2021-12-18:
Just tried it, OTA works. But the PCB must be connected to the AC, at least SCK must be connected. One esp8266 is sufficient. I suggest you close the issue here, and open a new one in [MHI-AC-Trace](https://github.com/absalom-muc/MHI-AC-Trace) if needed.

absalom-muc, 2021-12-19:
Your AC is one of the few with a different MOSI signature. 6d 80 04 instead of 6c 80 04. I adapted the SW, it should work now. Please also consider the new option MQTT_CHAR in support.h.

closed 2022-01-20 by kARTechnology