absalom-muc/MHI-AC-Ctrl #74: set/Troom precision

kARTechnology, 2021-12-08:
How to get .5 precision for setting the target room temperature. the ac does support this as in eco mode by remote it operates 1.5 deg.. higher than set temperature.

absalom-muc, 2021-12-08:
By intention it is not supported. That would only counterfei an accuracy for the temperature measurement and control that the AC cannot achieve.

kARTechnology, 2021-12-09:
> By intention it is not supported. That would only counterfei an accuracy for the temperature measurement and control that the AC cannot achieve.

the ac does support upto 0.5 degree precision. i think then the only way to get that precision is to get the eco mode working
[image]

closed 2021-12-09 by kARTechnology

absalom-muc, 2021-12-10:
we have a misunderstanding, the AC supports the higher resolution. MHI-AC-Ctrl doesn't support it because see above