absalom-muc/MHI-AC-Ctrl #76: room temperature lower than set Temperature by 2 degree

kARTechnology, 2021-12-13:
I have been using the set/Troom from external sensor. sensor reports 24, ac reports the same. but the set temperature is 26.
any further tests to be done? target temp reporting is correct as ac remote corresponds to the one displayed by mqtt. but is the room temperature reading flawed? even if the external sensor is not calibrated, set temp must be equal to room temp.
[image]

absalom-muc, 2021-12-13:
The diagram is a little bit confusing for me.
What is the time slot in your scope? 1am-4:30pm?
Which mode do you use? COOL?
Have you made the identical test with the internal room temperatur sensor?

damienmorrissey, 2021-12-13:
If it helps this happens to me with the standard remote in cool mode. Set
to 23 and the room will be 21 as per other measurements. This is with an
old 2.5kW split system.

kARTechnology, 2021-12-13:
> 

this is with new inverter series.

closed 2022-01-20 by kARTechnology