absalom-muc/MHI-AC-Ctrl #77: 3d auto and vanes l/r seems possible over CNS

sauermouth, 2022-01-23:
MHI has a (new?) WiFi module called WF-RAC. This module is connected to the CNS port of the airco. Then you can use the 'Smart M-air' app to control your airco. I've read a confirmation that 3D auto and vanes left/right is supported by the app and also works. I don't know if the app has more extra features.

I see some sites advertising the module should work with models SRKxxZS(X), SRKxxZR, SRRxxZS and SRFxxZSX. 

The app has a demo, maybe it can be used to decompile the SPI commando's?

niclet, 2022-01-24:
Hi,
I'm afraid app will only send WiFi instructions to the WF-RAC, SPI signal will be only between the WF-RAC and the airco
But I like the idea of an external WiFi item, I intend to use a Wio Terminal beside the airco to be able to control it manually and remotely.
This is just an idea for now

absalom-muc, 2022-01-24:
> MHI has a (new?) WiFi module called WF-RAC. This module is connected to the CNS port of the airco. Then you can use the 'Smart M-air' app to control your airco. I've read a confirmation that 3D auto [...]

I've read [here (German)](https://www.dynoair24.de/detail/index/sArticle/7954) contrary information, **no** support of  3D auto and vanes left/right

sauermouth, 2022-01-24:
That is the old module. The new module is a wired module.
Here's is a manual: https://www.mhi-mth.co.jp/en/products/pdf/rkz012a105a_english.pdf

They can be bought in The Netherlands:
https://www.coolmark.nl/nl/artnr/7140199

@niclet You are correct that the SPI commands will (probably) not be in the app, but in the module. But I believe absalom-muc and another user were investigating tapping into the signals between the wifi-module and the CNS-port.

sauermouth, 2022-01-24:
Here are a few screenshots of the app, with buttons for 3D Auto and vanes L/R. I also see support for fan speed 'AUTO'. That would be really nice! I would like to be able to slow down the fan speed on dinner times and revert it back to 'auto' after dinner.
https://gathering.tweakers.net/forum/list_message/70335624#70335624

The same user confirms on that forum the buttons really work.

absalom-muc, 2022-01-24:
o.k. I see, thank you. That is the first time that I see that a wired module supports these commands.
Then it should be not too complicated to snif the related SPI commands with the MHI-AC-Ctrl PCB and the SW [here](https://github.com/absalom-muc/MHI-AC-Trace).

th3sniper86, 2022-02-04:
https://www.coolmark.nl/media/files/technical_manual_2021-wf-rac.pdf

picanl, 2022-04-28:
Any updates on this ?? would be nice feature to have

absalom-muc, 2022-05-16:
You need a  wired remote control supporting the interesting modes, where you can snif the SPI communication to implement it in MHI-AC-Ctrl.

hungyip, 2022-06-09:
I can help to see how to sniff the SPI, what does a wired remote control mean?

kARTechnology, 2022-06-15:
> You need a wired remote control supporting the interesting modes (i.e. the new WF-RAC mentioned above), where you can snif the SPI communication to implement it in MHI-AC-Ctrl.

Regarding self clean, it is actually a mode like fan/cool/heat.
As i told before the ac didn't respond to on/off from mqtt once it is in self clean by remote.
Today i tried to change mode from mqtt when it is in self clean more and the ac responded as usual.

So it is concerned with the mode bits and not the power bits.

My other ac didn't have cns support and i installed ir-esp8266 control for it. And even in that, self clean, 3d auto, turbo, econo are supported fully, turbo, econo are also connected to the fan bits. 

Is there a way for me to experiment by sending possible combinations of mode bits(cool dry auto) to find self clean?
Also fan bits for turbo and econo.

Silent mode in remote Is limiting compressor to 40%. With which im able to run ac all day long with reduced power, but higher efficiency (cop is high for partial load as per spec)

absalom-muc, 2022-06-15:
As you can see in the [frame summary](https://github.com/absalom-muc/MHI-AC-Trace/blob/main/SPI.md#summary) many bits of the frame are used by known functions (green), for some others (grey & white) I'm not aware if they are used or which function they have. Of course you could test different combinations - but as usual on your own risk.
But maybe it is easier to record the MOSI frame and check what happens when self clean is activated. Maybe so far unknown frame bits are used.

kARTechnology, 2022-06-16:
I had previously mosi bits, you were unable to find any changes in it..

KostyaSha, 2022-06-28:
Could somebody describe what is the difficulty to press interesting mode and snif in on SPI?

KostyaSha, 2022-06-28:
As i see on screenshots AM-MHI-01 has auto fan button and left-right setting, isn't it?

absalom-muc, 2022-06-29:
> Could somebody describe what is the difficulty to press interesting mode and snif it on SPI?

No difficulty, you need only someone who has the wired remote control supporting the interesting cases and is willing to check it.

> As i see on screenshots AM-MHI-01 has auto fan button and left-right setting, isn't it?

Not sure which sreenshots you mean

kARTechnology, 2022-06-29:
He meant the mobile app i guess for the wifi remote control given by mhi

DYLaKo, 2022-07-20:
@absalom-muc If I can get my hands on a MHI WF-RAC, are you willing to implement the new features like 3D auto and Vanes if I provide you with the needed information? If yes, what do I need to build / flash in order to give you the needed information?

KostyaSha, 2022-07-20:
@absalom-muc  is it possible to sniff and log communication? I think we can then send single commands and verify how a/c reacts to them (i can join with testing soon (waiting d1 mini from post)).

absalom-muc, 2022-07-21:
> @absalom-muc If I can get my hands on a MHI WF-RAC, are you willing to implement the new features like 3D auto and Vanes if I provide you with the needed information? If yes, what do I need to build / flash in order to give you the needed information?

yes, please check [MHI-AC-Trace](https://github.com/absalom-muc/MHI-AC-Trace)

DYLaKo, 2022-07-21:
> yes, please check [MHI-AC-Trace](https://github.com/absalom-muc/MHI-AC-Trace)

Ok, I might be able to get one. Not quite sure yet how to connect it, both the MHI-AC-Ctrl and the original controller need to be connected parallel? You mention the old version of the board, is there any image to explain it?

absalom-muc, 2022-07-22:
Conncetion has to be parallel according to the diagram (just added)  [here](https://github.com/absalom-muc/MHI-AC-Trace/blob/main/README.md#installing).

Aesculapius, 2022-08-06:
Just for information, I'm seeing a lot of development on the HomeAssistant forums with this new module - they seem to have decoded stuff. I placed a request to help out here...
https://community.home-assistant.io/t/mitsubishi-wifi-module-wf-rac-smart-m-air/411025/76

synegic, 2022-10-09:
Would be great if someone with the new module could sniff out the unknowns. If someone has the new WiFi module and is able to send it to me (Netherlands), I am willing to take a stab at sniffing the SPI communication. I promise to send it back afterwards. :-)

DYLaKo, 2022-10-09:
> Would be great if someone with the new module could sniff out the unknowns. If someone has the new WiFi module and is able to send it to me (Netherlands), I am willing to take a stab at sniffing the SPI communication. I promise to send it back afterwards. :-)

I have one, managed to get my hands on one this summer but haven't found the time yet to sniff the commands. Maybe we can work something out with the module. How can i contact you?

synegic, 2022-10-09:
> I have one, managed to get my hands on one this summer but haven't found the time yet to sniff the commands. Maybe we can work something out with the module. How can i contact you?

Sounds good! You can contact me via `jorrit<at>jorritpouw.com`.

DYLaKo, 2022-10-09:
> Sounds good! You can contact me via `jorrit<at>jorritpouw.com`.

Mail sent. :)

kARTechnology, 2022-10-11:
I can be the tester

synegic, 2022-10-11:
@kARTechnology I guess the more the merrier. DYLaKo is also sending his WF-RAC to me so I can do some SPI sniffing as well. If you'd do the same at least we have a very broad data set to reverse engineer upon. :-)

kARTechnology, 2022-10-11:
> merrier. DYLaKo is also lending his WF-RAC to me so I can do some SPI

Sure. i would be available after few days for testing

Pls check the outdoor silent option, which actually caps the compressor
power to max 40% irrespective of set temperature. Though The room will take
longer time to cool the efficiency is higher

synegic, 2022-11-08:
@absalom-muc So @DYLaKo has sent his MHI official WiFi remote to me to do some SPI Spy'ing for you. I've also bought some JST-XH connectors so I can hook it up with a MHI-AC-Spy in between without cutting up any cables. If I understand correctly, I can simply hook both the official WiFi remote and the MHI-AC-Ctrl loaded with MHI-AC-Trace software up in parallel by soldering two JST-XH 5p female connectors to a JST-XH 5p male connector?

Also, could you give me some instructions what would help best for your/our investigations while recording SPI traffic? Your documentation is pretty thorough so I could probably figure some things out myself, but it'll probably go faster with an extra pair of (trained) eyes.

absalom-muc, 2022-11-12:
> If I understand correctly, I can simply hook both the official WiFi remote and the MHI-AC-Ctrl loaded with MHI-AC-Trace software up in parallel by soldering two JST-XH 5p female connectors to a JST-XH 5p male connector?

Yes, that is correct.
> 
> Also, could you give me some instructions what would help best for your/our investigations while recording SPI traffic? Your documentation is pretty thorough so I could probably figure some things out myself, but it'll probably go faster with an extra pair of (trained) eyes.

A good starting point is [here](https://github.com/absalom-muc/MHI-AC-Trace/blob/main/SPI.md#summary). I suggest you try first to change something which is already known, e.g. temperature setpoint and check if you see the expected update of db2. If this works you can try a new function and identify the affected databytes. After that you could provide the log, then I can recheck it and if applicable add the new function to the program.

synegic, 2022-11-13:
OK, so here are some first results;

```
Baseline settings: FAN on full speed, U/D vanes UP, L/R vanes SWING, AUTO 3D OFF, SET TEMP 25c

16:42:40 AC OFF capture begin
16:43:03 AC ON
16:43:30 L/R VANES 1 1
16:44:00 L/R VANES 1 2
16:44:30 L/R VANES 2 2
16:45:00 L/R VANES 2 3
16:45:30 L/R VANES 3 3
16:46:00 L/R VANES 1 3
16:46:30 L/R VANES 3 1
16:47:00 L/R VANES SWING
16:47:28 L/R VANES 1 1 (sorry misclicked)
16:47:30 3D AUTO ON
16:48:00 3D AUTO OFF
16:48:30 3D AUTO ON
16:49:00 3D AUTO OFF
16:49:30 AC OFF
```
The app was quite sluggish in responding to commands, so the operations are probably not precise on the second.

Attached the timestamped log of the raw data as received by my MQTT broker, in CSV format. I really hope it contains the unicorns we were searching for. If you need anything else from me, just let me know.

[log000.csv](https://github.com/absalom-muc/MHI-AC-Ctrl/files/9997679/log000.csv)

kARTechnology, 2022-11-13:
I have ac with 3d auto,U/D and L/R vanes, silent mode and turbo/hi-power.
Happy to test.

absalom-muc, 2022-11-13:
@synegic: Not sure which console you use for the log, but the format of the file is not o.k.
`1668354158136,2022-11-13_16:42:38,24 09 6d 80 04 aa 8e b2 87 00 00 88 00 ff ff ff ff ff 00 00 08 e5 aa 00 07 aa 8e b2 ff 00 00 80 00 00 ff ff ff ff 0f 02 08 27 00 `
I don't know the number 1668354158136 and there should be no time stamp. [Here](https://github.com/absalom-muc/MHI-AC-Trace/blob/main/src/MHI-ErrOpData.log) is an example file how it should look like.

And I would appreciate when you would first check yourself the results, please see above:
> A good starting point is [here](https://github.com/absalom-muc/MHI-AC-Trace/blob/main/SPI.md#summary). I suggest you try first to change something which is already known, e.g. temperature setpoint [...]

synegic, 2022-11-13:
Hi @absalom-muc,I’m not using the serial console as ways to get the data but a Python script that listens to the MQTT published messages in the raw topic. The AC itself is not accessible with USB in my case.The timestamps are very handy actually cause you can directly see at which time the message was received with a clear date. Open it in excel or equivalent to see just the raw data.But, I will do my best. Byte stuff is not my strong suit but I can probably figure it out.To be continued.Verstuurd vanaf mijn iPhone

absalom-muc, 2022-11-13:
I believe you have only to remove first 34 chars, the rest should be fine.

synegic, 2022-11-14:
@absalom-muc just a quick question; are we looking for a 'consistent' change in frame data, or just a single line of changed frame data?

Just an example (nothing interesting probably), but to describe what I'm talking about; like the yellow marked messages or the ones around it?

[image]

absalom-muc, 2022-11-14:
Let us first focus on permanent change of the MOSI databytes db0-db9. When we don't find it, we have to extend the search. The rows you marked in yellow is a read of operating data.

synegic, 2022-11-15:
@absalom-muc So I actually hooked up my laptop through USB this time, and I noticed nothing really changes consistently in the MOSI part when I toggle the L/R vanes/3D Auto button.

The only thing that seems to happen in the MISO part is that one byte changes from 00 to 80, for the duration of the vane motors movement (maybe to communicate back their position in some timing encoded way?). Could it be that we are missing some bytes from the (currently captured) SPI frames/they extended the SPI bytes and added something we are not seeing now?

```
16:14:28.848 -> 34 e4 6d 80 04 af bf b2 88 00 00 88 00 ff ff ff ff ff 01 00 09 1d aa 00 07 af bf b2 ff 00 00 80 00 00 ff ff ff ff 0f 06 08 61 00
16:14:28.886 -> 34 e5 6d 80 04 af bf b2 88 00 00 88 00 ff ff ff ff ff 01 00 09 1d aa 00 07 af bf b2 ff 00 00 00 00 00 ff ff ff ff 0f 02 07 dd 00
```

kARTechnology, 2022-11-15:
> @absalom-muc So I actually hooked up my laptop through USB this time, and I noticed nothing really changes consistently in the MOSI part when I toggle the L/R vanes/3D Auto button.
> 
> The only thing that seems to happen in the MISO part is that one byte changes from 00 to 80, for the duration of the vane motors movement (maybe to somehow instruct the vane position to move/change in a timing encoded way?). Or could it be that we are missing some bytes from the (currently captured) SPI frames/they extended the SPI bytes and added something we are not seeing now? It is weird as the controller/app does seem pretty sure of the current position.
> 
> Maybe one of the unknown dataframes communicates back the vane position so the controller/app knows?
> 
> ```
> 16:14:28.848 -> 34 e4 6d 80 04 af bf b2 88 00 00 88 00 ff ff ff ff ff 01 00 09 1d aa 00 07 af bf b2 ff 00 00 80 00 00 ff ff ff ff 0f 06 08 61 00
> 16:14:28.886 -> 34 e5 6d 80 04 af bf b2 88 00 00 88 00 ff ff ff ff ff 01 00 09 1d aa 00 07 af bf b2 ff 00 00 00 00 00 ff ff ff ff 0f 02 07 dd 00
> ```

I would suggest you to use MQTTExplorer if you want to keep track of previous data..
If the app and MHI's official box supports 3d auto then the data should definitely contain the bytes for them for us to decode...

synegic, 2022-11-15:
@kARTechnology apparently it's not that easy. I guess it's also quite logical it's not returned within the (currently known) MOSI bytes, because if it would have been included we would already have discovered it when changing vanes L/R or toggling 3D auto functions by IR remote. The MISO bytes as now printed by the trace program simply do not contain any changes. I am not (completely) blind. ;-)

Re: MQTTExplorer; I am using Serial monitor in my latest attempts, so it's about as reliable as it will get. I noticed the ESP would sometimes just reset on its own, so I decided to remove all MQTT code from the MHI-AC-Trace program code and just leave the Serial.println's in. It is quite easy to spot the differences in the Serial monitor log when pressing buttons. You can see all the different bytes change, according to earlier identified operation data.

synegic, 2022-11-16:
> @synegic only to prevent misunderstandings. The commands should be send by the wired remote control, not by the infrared RC.
> 
> Exception: If you see that the status of the wired remote control changes by a command issued by the IR RC then a deep analysis is appropriate.

All commands were sent with WIFI-RAC remote. Didn't use the IR remote. I will create a full log again where I cycle through all L/R vane options and also check whether I see the change in L/R position being updated in the app when changing it through the IR remote. If that's the case, at least we know the data for position is there 'somewhere'.

KostyaSha, 2022-11-16:
You can make video where live log is and remote device is pressed. That may help understand by placing on pause what command and reply was sent.

absalom-muc, 2022-11-20:
> Maybe one of the unknown dataframes communicates back the vane position so the controller/app knows?
>
> ```
> 16:14:28.848 -> 34 e4 6d 80 04 af bf b2 88 00 00 88 00 ff ff ff ff ff 01 00 09 1d aa 00 07 [...]

The sequence "af bf" (db0=af db1=bf) indicate the following write commands
db0=af: vanes swing disable, mode=fan only, power=on
db1=bf: vanes up/down position set to 4, fan set to Auto

Currently I try to compress your log file of about 9600 rows because it is to big for an analysis.
The combination of your AC and WF-RAC behaves differently than with my AC and WF-RAC. With every frame there is in your log a toggle of MISO db14 between 0x02 and 0x06. This is the main reason for this huge log. I will eliminate the toggle and also the jitter of the room temperature.

Will come back to you once I have news.

absalom-muc, 2022-11-20:
I can see the following Baseline settings:  FAN level 4, U/D vanes UP,SET TEMP 25c, MODE COOL
The following operating data is read:
- RETURN AIR: 18.5°C
- Outddor temp: 12°C
- data which I can't identify, but since the response from the AC is always identical "94 10 00 00" for the complete log I assume it is not related to the functionality we are looking for

1668354184580,2022-11-13_16:43:04: AC ON
1668354580173,2022-11-13_16:49:40: AC OFF (but according your post it should be 16:49:30)

The log doesn't show L/R VANES or 3D AUTO. Currently I have no idea how to continue..

synegic, 2022-11-20:
@absalom-muc thanks for your effort. I will do another log including video from app interaction and result as suggested. But yeah, the currently logged bytes do not contain any changes as it seems.

Would it help if I could get my hands on a spectrum analyser and get you a reading of the SPI signal I see on the controller <-> AC communication lines?

absalom-muc, 2022-11-20:
> Would it help if I could get my hands on a spectrum analyser and get you a reading of the SPI signal I see on the controller <-> AC communication lines?

The protocol incl. timing is described [here](https://github.com/absalom-muc/MHI-AC-Trace/blob/main/SPI.md#protocol). If you can record the communication you could compare it. In case of questions I can support, but to be honest I don't plan to spend much more time for it.

rwijnhov, 2022-12-12:
Any updates on this? whenever you control the vanes now it will reset van postion and you have to use the remote to set l+r again.

synegic, 2022-12-16:
No updates yet. Still haven't made progress finding out which bytes are changed.

I have a friend who has a logic analyser/data logger device. He's gonna lend it to me so I can do some further investigation as to what's actually communicated between the new WiFI module and the unit. Maybe it reveals something.

glsf91, 2023-01-25:
@synegic Any progress on this?

glsf91, 2023-02-04:
I also took a look with a logic analyser when my WF-RAC module is connected.
Now every SPI frame has 33 bytes instead of 20. Also the signature of the MISO frame is different.
So a big difference.
I think this explains why 3D auto and vanes L/R is visible in the app and not when using MHI-trace.

dennistd4, 2023-02-04:
So the MHI-AC-Ctrl should identify itself somehow as WF-RAC to get access to these functions?

glsf91, 2023-02-04:
Yes

synegic, 2023-02-05:
> I also took a look with a logic analyser when my WF-RAC module is connected. Now every SPI frame has 33 bytes instead of 20. Also the signature of the MISO frame is different. So a big difference. I think this explains why 3D auto and vanes L/R is visible in the app and not when using MHI-trace.
> 
> Edit: I can confirm that L/R vanes and 3d auto is visible in the extra bytes of the MOSI frame

This is great stuff!! It prompted me to actually try something I was thinking about doing already; simply enlarge the byte range the MHI-AC-Trace code is taking out of the frames.

So I modified the MHI-AC-Trace sketch to broaden the byte range for the frames from 20 to 24 and low and behold, the extra data is indeed there. It seems they simply slapped on another 3 bytes to the end of the original response to add support for this as the checksums seem to be included at the same positions and are not failing, even not with the missing bytes (I conclude this as I'm not getting any invalid checksum warnings from MHI-AC-Trace). They probably did this for backwards compatibility reasons?

Now we have the extra bytes available in the log output, we can pinpoint the bytes that are used for these settings;

```
16:47:35.933 -> 01 33 6d 80 04 ab b8 a8 84 00 00 88 02 ff ff ff ff ff 05 00 09 0a 00 16 0a 00 aa 00 07 ab b8 a8 ff 00 00 80 00 00 ff ff ff ff 0f 02 08 48 00 16 0a 00 00 
16:47:35.971 -> 01 34 6d 80 04 ab b8 a8 84 00 00 88 02 ff ff ff ff ff 05 00 09 0a 00 16 0e 00 aa 00 07 ab b8 a8 ff 00 00 00 00 00 ff ff ff ff 0f 06 07 cc 00 16 0e 00 00 
```

Here we can find 3D Auto seems to be controlled by the 23rd byte. It is reflected back in both MOSI and MISO frames. Possible values seem to be 0a (3D Auto off) and 0e (3D AUTO on).

Horizontal vane control is inside byte 22;

```
17:05:04.696 -> 15 3f 6d 80 04 ab b8 a8 83 00 00 88 02 ff ff ff ff ff 05 00 09 09 00 10 0a aa 00 07 ab b8 a8 ff 00 00 80 00 00 ff ff ff ff 0f 02 08 48 00 10 0a 00 
17:05:04.768 -> 15 40 6d 80 04 ab b8 a8 83 00 00 88 02 ff ff ff ff ff 05 00 09 09 00 11 0a aa 00 07 ab b8 a8 ff 00 00 00 00 00 ff ff ff ff 0f 06 07 cc 00 11 0a 00 
```

In above excerpt 3D Auto is disabled (0a - mandatory for horizontal vane to be selected, of course) and horizontal vane position is changed from 10 to 11. (// to /|).

Walking through all different options, we seem to get the following mapping;

```
/ /  -> 10
/ | -> 11
| | -> 12
| \ -> 13
\ \ -> 14
/ \ -> 15
\ / -> 16
```

Seems like this is the data we need to build in support. I have not done any debugging yet to find out if the actual handshake is changed to 'enable' anything. I doubt it to be that sophisticated, but maybe it has something to do with the fact that MHI-AC-Trace sketch currently does not properly detect MOSI changes..

kARTechnology, 2023-02-05:
This is great news. I hope the silent and hi power could be possible too. I
have ac with at 3d auto and silent, eco, hi-power

synegic, 2023-02-05:
So I've enlarged the byte range to 35 bytes per frame. It seems there are more bytes that contain a value, as already found by @glsf92. Maybe another checksum for the new/larger payload format and/or used to tell the AC it is supported?

```
17:25:10.742 -> 00 1a 6d 80 04 aa b8 a8 84 00 00 88 00 ff ff ff ff ff 00 00 09 02 00 16 0a 00 00 00 00 00 ff ff ff ff 29 00 00 aa 00 07 aa b8 a8 ff 00 00 80 00 00 ff ff ff ff 0f 02 08 47 00 16 0a 00 00 00 00 00 ff ff ff ff b2 00 00 00 
17:25:10.742 -> 00 1b 6d 80 04 aa b8 a8 84 00 00 88 00 ff ff ff ff ff 00 00 09 02 00 16 0a 00 00 00 00 00 ff ff ff ff 29 00 00 aa 00 07 aa b8 a8 ff 00 00 80 00 00 ff ff ff ff 0f 06 08 4b 00 16 0a 00 00 00 00 00 ff ff ff ff ba 00 00 00 
17:25:10.742 -> 00 1c 6d 80 04 aa b8 a8 84 00 00 88 00 ff ff ff ff ff 00 00 09 02 00 16 0a 00 00 00 00 00 ff ff ff ff 29 00 00 aa 00 07 aa b8 a8 ff 00 00 80 00 00 ff ff ff ff 0f 02 08 47 00 16 0a 00 00 00 00 00 ff ff ff ff b2 00 00 00 
17:25:10.785 -> 00 1d 6d 80 04 aa b8 a8 84 00 00 88 00 ff ff ff ff ff 00 00 09 02 00 16 0a 00 00 00 00 00 ff ff ff ff 29 00 00 aa 00 07 aa b8 a8 ff 00 00 80 00 00 ff ff ff ff 0f 06 08 4b 00 16 0a 00 00 00 00 00 ff ff ff ff ba 00 00 00 
17:25:10.785 -> 00 1e 6d 80 04 aa b8 a8 84 00 00 88 00 ff ff ff ff ff 00 00 09 02 00 16 0a 00 00 00 00 00 ff ff ff ff 29 00 00 aa 00 07 aa b8 a8 ff 00 00 80 00 00 ff ff ff ff 0f 02 08 47 00 16 0a 00 00 00 00 00 ff ff ff ff b2 00 00 00 
17:25:10.785 -> 00 1f 6d 80 04 aa b8 a8 84 00 00 88 00 ff ff ff ff ff 00 00 09 02 00 16 0a 00 00 00 00 00 ff ff ff ff 29 00 00 aa 00 07 aa b8 a8 ff 00 00 80 00 00 ff ff ff ff 0f 06 08 4b 00 16 0a 00 00 00 00 00 ff ff ff ff ba 00 00 00 
17:25:10.786 -> 00 20 6d 80 04 aa b8 a8 84 00 00 88 00 ff ff ff ff ff 00 00 09 02 00 16 0a 00 00 00 00 00 ff ff ff ff 29 00 00 aa 00 07 aa b8 a8 ff 00 00 80 00 00 ff ff ff ff 0f 02 08 47 00 16 0a 00 00 00 00 00 ff ff ff ff b2 00 00 00 
17:25:10.831 -> 00 21 6d 80 04 aa b8 a8 84 00 00 88 00 ff ff ff ff ff 00 00 09 02 00 16 0a 00 00 00 00 00 ff ff ff ff 29 00 00 aa 00 07 aa b8 a8 ff 00 00 80 00 00 ff ff ff ff 0f 06 08 4b 00 16 0a 00 00 00 00 00 ff ff ff ff ba 00 00 00 
17:25:10.868 -> 00 22 6d 80 04 aa b8 a8 84 00 00 88 00 ff ff ff ff ff 00 00 09 02 00 16 0a 00 00 00 00 00 ff ff ff ff 29 00 00 aa 00 07 aa b8 a8 ff 00 00 80 00 00 ff ff ff ff 0f 02 08 47 00 16 0a 00 00 00 00 00 ff ff ff ff b2 00 00 00 
17:25:10.937 -> 00 23 6d 80 04 aa b8 a8 84 00 00 88 00 ff ff ff ff ff 00 00 09 02 00 16 0a 00 00 00 00 00 ff ff ff ff 29 00 00 aa 00 07 aa b8 a8 ff 00 00 80 00 00 ff ff ff ff 0f 06 08 4b 00 16 0a 00 00 00 00 00 ff ff ff ff ba 00 00 00 
17:25:10.975 -> 00 24 6d 80 04 aa b8 a8 84 00 00 88 00 ff ff ff ff ff 00 00 09 02 00 16 0a 00 00 00 00 00 ff ff ff ff 29 00 00 aa 00 07 aa b8 a8 ff 00 00 80 00 00 ff ff ff ff 0f 02 08 47 00 16 0a 00 00 00 00 00 ff ff ff ff b2 00 00 00 
```

@kARTechnology Even if the extended protocol would support setting those features, as long as the M-Air app does not support these settings we cannot really know which values the AC would expect for that to work.

And it looks like the app does indeed signal the AC it takes over 'control' with the app control, as L/R vane control cannot be set when the remote control is in use.. The app does however understand the status updates on L/R vane position once you change them through remote.

Example log output when switching from remote to app control, which seems to enable those extra fields;

```
 6d 80 04 08 30 28 82 00 00 88 00 ff ff ff ff ff 00 00 07 56 00 06 00 00 00 00 00 00 ff ff ff ff b5 aa 00 07 08 30 28 ff 00 00 80 00 00 ff ff ff ff 0f 02 06 9d 00 06 00 00 00 00 00 00 ff ff ff ff 42 00 
17:40:33.207 -> 0d 2f 6d 80 04 08 30 28 82 00 00 88 00 ff ff ff ff ff 00 00 07 56 00 06 00 00 00 00 00 00 ff ff ff ff b5 aa 00 07 08 30 28 ff 00 00 80 00 00 ff ff ff ff 0f 06 06 a1 00 06 00 00 00 00 00 00 ff ff ff ff 4a 00 
17:40:33.281 -> 0d 30 6d 80 04 08 30 28 82 00 00 88 00 ff ff ff ff ff 00 00 07 56 00 06 00 00 00 00 00 00 ff ff ff ff b5 aa 00 07 08 30 28 ff 00 00 80 00 00 ff ff ff ff 0f 02 06 9d 00 06 00 00 00 00 00 00 ff ff ff ff 42 00 
17:40:33.315 -> 0d 31 6d 80 04 08 30 28 82 00 00 88 00 ff ff ff ff ff 00 00 07 56 00 06 00 00 00 00 00 00 ff ff ff ff b5 aa 00 07 08 30 28 ff 00 00 80 00 00 ff ff ff ff 0f 06 06 a1 00 06 00 00 00 00 00 00 ff ff ff ff 4a 00 
17:40:33.387 -> 0d 32 6d 80 04 b2 b8 b2 82 00 00 88 00 ff ff ff ff ff 00 00 09 12 00 16 0a 00 00 00 00 00 ff ff ff ff 49 aa 00 07 b2 b8 b2 ff 00 00 00 00 00 ff ff ff ff 0f 02 07 d9 00 16 0a 00 00 00 00 00 ff ff ff ff d5 00 
17:40:33.425 -> 0d 33 6d 80 04 b2 b8 b2 82 00 00 88 00 ff ff ff ff ff 00 00 09 12 00 16 0a 00 00 00 00 00 ff ff ff ff 49 aa 00 07 b2 b8 b2 ff 00 00 00 00 00 ff ff ff ff 0f 06 07 dd 00 16 0a 00 00 00 00 00 ff ff ff ff dd 00 
17:40:33.463 -> 0d 34 6d 80 04 b2 b8 b2 82 00 00 88 00 ff ff ff ff ff 00 00 09 12 00 16 0a 00 00 00 00 00 ff ff ff ff 49 aa 00 07 b2 b8 b2 ff 00 00 00 00 00 ff ff ff ff 0f 02 07 d9 00 16 0a 00 00 00 00 00 ff ff ff ff d5 00 
17:40:33.501 -> 0d 35 6d 80 04 b2 b8 b2 82 00 00 88 00 ff ff ff ff ff 00 00 09 12 00 16 0a 00 00 00 00 00 ff ff ff ff 49 aa 00 07 b2 b8 b2 ff 00 00 00 00 00 ff ff ff ff 0f 06 07 dd 00 16 0a 00 00 00 00 00 ff ff ff ff dd 00 
17:40:33.577 -> 0d 36 6d 80 04 b2 b8 b2 82 00 00 88 00 ff ff ff ff ff 00 00 09 12 00 16 0a 00 00 00 00 00 ff ff ff ff 49 aa 00 07 b2 b8 b2 ff 00 00 00 00 00 ff ff ff ff 0f 02 07 d9 00 16 0a 00 00 00 00 00 ff ff ff ff d5 00 
17:40:33.615 -> 0d 37 6d 80 04 b2 b8 b2 82 00 00 88 00 ff ff ff ff ff 00 00 09 12 00 16 0a 00 00 00 00 00 ff ff ff ff 49 aa 00 07 b2 b8 b2 ff 00 00 00 00 00 ff ff ff ff 0f 06 07 dd 00 16 0a 00 00 00 00 00 ff ff ff ff dd 00
```

glsf91, 2023-02-05:
The last byte of the MISO and MOSI frame is indeed an extra checksum. Just sum all previous bytes 0-31 and use the lower part of the sum as the checksum for the last byte.

> Also interestingly, once you control the AC by remote, the extra fields are not visible in the bytes. Only when you 'take over control' by app, they show up all of the sudden.
For me they where also available when changing by RC.

synegic, 2023-02-05:
> > Also interestingly, once you control the AC by remote, the extra fields are not visible in the bytes. Only when you 'take over control' by app, they show up all of the sudden.
For me they where also available when changing by RC.

Did you also try turning the AC off by app, then turning it on by IR remote, changing vanes up/down values? Maybe it doesn't do the switching back to 'legacy' if the AC is already on?

glsf91, 2023-02-05:
No. I turned on the AC with the remote and then start experimenting with changing settings in app and remote.
With RC for example 02 was shown. With app 12.

synegic, 2023-02-05:
> No. I turned on the AC with the remote and then start experimenting with changing settings in app and remote. With RC for example 02 was shown. With app 12.

Ah, that makes sense! I see it now as well; the value 06 in my log excerpt changes to 16 which is also what you're seeing. The 3D auto feature is 00 though and changes to 0a when app is used (without actually changing the 3D Auto feature status). Logically that probably also has a different value vs. app one if it's enabled by IR remote.

[43 more replies]

closed 2023-10-02 by glsf91